HARPLab/DReyeVR #7: make: *** No rule to make target 'carla'.  Stop. - DReyeVR Installation

Shroculus, 2022-02-17:
Hello team,

**CARLA version:** 0.9.13

**Platform/OS:** Windows 10 (Version 21H2)

**Problem you have experienced:** When I try to install the DReyeVR in the WSL using the command make carla CARLA=../carla it throws an error as shown below in the screenshot.

[image]

**Steps to reproduce:**

1. I previously had a working 0.9.13 Carla with Unreal 4.26 installed and working successfully. (All three make PythonAPI, make launch and make check executed without any errors)
2. I have the SteamVR installed and use HTC Vive eye Pro for eye tracking. 
3. Enabled the SteamVR in UE4 as shown in the screenshot below:

[image]

4. Currently concentrating on only Logitech Steering Wheel (I have Logitech 920 which is installed and works fine) so SRanipal is not downloaded or setup. But the Logitech Wheel Plugin is cloned and moved to the Carla Plugin folder and the plugin folder looks as below.

[image]

5.  Installing DReyeVR  is the step causing the problem. I have Ubuntu installed and I am at the Carla root folder to execute the command (make carla CARLA=../carla) which throws the error "make: *** No rule to make target 'carla'.  Stop.".

6.  I have git installed with the necessary environmental variable path. 

[image]

[image]

Could you please help me with where I am going wrong and how to move forward with the installation? 

I very much appreciate any help. Thank you in advance.

GustavoSilvera, 2022-02-17:
Hi, you need to run the `make carla CARLA=../carla` script in the `DReyeVR` directory, not the `Carla` directory. 

So for example, I assume you cloned both `DReyeVR` and `carla` and have a file structure like this:
```
DReyeVR/
carla/
```

so if you go to the `DReyeVR` directory and run `make carla CARLA=../carla` will find the correct carla installation. 

Technically you could also simply run `make carla CARLA=/PATH/TO/CARLA` without being a relative path. 

(Also, you don't need to run `make` with `sudo`, as its just moving files around)

Shroculus, 2022-02-17:
@GustavoSilvera Thank you for your prompt response. I was able to successfully launch the DReyeVR. However, when execute the below command only the logitech steering wheel/pedals work but cannot view anything in my HTC Vive eye pro. I have the SteamVR plugin enabled and also HTC vive pro setup and detected through steam VR as shown below:

[image]

cd \PATH\TO\CARLA\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\
CarlaUE4.exe -vr

I have set the SRanipal to false, but think that should not impact playing the DreyeVR's VR mode capabilities since it is purely for the eye tracking.

Also, I launched the editor through make launch and I don't see the VR preview enabled. 

[image]

Thank you for your help in advance.

GustavoSilvera, 2022-02-17:
Ok, I sometimes also get the grayed out **VR Preview** selection when I start UE4 before opening SteamVR. Usually my workflow is to start SteamVR first, and leave it running, then it should pick up a new `CarlaUE4.exe -vr` or editor instance. Maybe try rebooting your pc and starting SteamVR first, then launch Carla+DReyeVR again with CarlaUE4.exe -vr once it is packaged. 

You can check if DReyeVR has detected the SteamVR device by checking the logs (in the Engine you can get to the logs in realtime via Window -> Developer Tools -> Output Log), when you press play, you should see a log entry for whether or not the HMD was detected by DReyeVR:

```
# is detected
LogTemp: HMD detected: SteamVR, version 4.26.2-0+++UE4+Release-4.26, Driver: lighthouse, Serial: XYZ, HMD Device: HTC VIVE_Pro MV, Driver version: TrackedProp_UnknownProperty

# not detected
LogTemp: Warning: No head mounted device detected!
```

Also to find these logs afterwards in text form, you'll want to look in PATH\TO\CARLA\Unreal\CarlaUE4\Saved\Logs\CarlaUE4.log

Shroculus, 2022-02-18:
@GustavoSilvera Thanks again, this has been very helpful. I was able to Setup the DReyeVr in VR mode and also drive using steeringwheel/pedals.

closed 2022-02-18 by Shroculus