HARPLab/DReyeVR #10: Hardware requirements / performance recommendations?

tensorturtle, 2022-03-26:
Hello, I'm interested in trying out DReyeVR, but I don't have any of the hardware yet.
In [SetupVR.md](https://github.com/HARPLab/DReyeVR/blob/main/Docs/SetupVR.md), you mention that performance may be an issue even with powerful GPUs. What GPU/CPU/RAM was used in development? What kind of frame rates were you able to achieve?
I prefer to use Linux wherever I can.

Thank you
Jason

GustavoSilvera, 2022-03-26:
Hi @tensorturtle. Yes, performance is not going to be great because 
a) Carla itself is not very optimized ([ex 1](https://github.com/carla-simulator/carla/issues/2617), [ex 2](https://github.com/carla-simulator/carla/issues/3891), you can find many more examples)
b) VR is intrinsically very expensive because you will have to render at least two viewpoints of the scene (one for each eye, more in case you want accurate planar reflections (ie. realistic rear/side view mirrors)) and at high resolutions to avoid aliasing. 

To answer your question directly, for development we are primarily using an **Intel i9-10900K 10c 20t CPU @ 3.7GHz, with 64G DDR4 system memory, and an RTX 3090 24G GPU**. This is on a Windows machine with DirectX support. With this setup (along with the Vive (2x 1440x1600 @90hz screens)) **we see typically 25-30fps** when going full-tilt (heavy background activity, planar reflection mirrors enabled, etc.) but without mirrors that number **increases to ~50fps**. This is *not great* especially for VR where participants occasionally get motion sickness, but there is little that we can do to combat this. We have also verified with [MSI afterburner](https://www.msi.com/Landing/afterburner/graphics-cards) that the GPU is the main bottleneck as the rasterization is very expensive for these dynamic CARLA environments in VR. There are also resolution scaling options you might be interested in trying to optimize framerate for your headset. 

(For more insights, we found that more CPU cores => better performance if you have high activity in your scenes (lots of background actors) and this scaled with an AMD 3900x 12c 24t CPU on Linux with 100's of actors. System memory is also not a huge concern, as long as you have 32G you should be fine (unless you have many things running on the computer). Video memory is a very big concern, our 3090 system is fine as it has 24G and we have seen as much as 17G used just for the VR rasterization, but typical use entails ~8-10G of VRAM usage.)

We have also (me personally) used several Linux (Ubuntu 20.04) machines with GPU's ranging from a GTX 1070/1080, to a 2080 SUPER, but what I can tell you is that VRAM (all those cards have 8G) quickly becomes an issue with the Vulkan RHI (which is what you'll need to use on Linux). 

We'd also prefer to primarily use Linux, though unfortunately we have to use Windows because the eye-tracking and steering wheel plugins have proprietary dependencies that are Windows-only. Also, the performance using DirectX on Windows is significantly better than Vulkan on Linux (we saw ~30% improved frametimes on Windows compared to Linux with the same hardware, just because the Windows build was using DirectX which is Microsoft-only).

However, not all is lost! We are actually in the middle of implementing and benchmarking [foveated rendering](https://mariosbikos.com/vive-unreal-foveated-rendering/) as a plugin into the Unreal Engine and DReyeVR, which should give us some free performance by using the eye tracker sensor we are already tracking and reducing graphical intensity in the peripheral regions. But this project is still in its early stages.

tensorturtle, 2022-03-26:
Hi @GustavoSilvera, thank you very much for your wonderfully detailed response. It is very helpful.

For what I'm interested in, I won't need mirrors or eye tracking, so hopefully that alleviates some pressure.

I'm sure you've looked into this before, but I have some questions about ways to improve performance

1. Are you aware of any effort upstream (Carla, Unreal) to get dual-GPU VR rendering (one per eye) working?
2. If VRAM is the issue, have you considered getting an even bigger card than the 3090, like Quadro A6000 or RTX8000? (I know that these are astronomically priced and that part of your mission is system affordability)

Thanks
Jason

GustavoSilvera, 2022-03-26:
Hi @tensorturtle, good points! Not needing mirrors is definitely a big boost to performance (or rather, enabling them is a performance killer) but they'll still be available as a toggle for you to use if you choose (see `DReyeVRConfig.ini`)

1. I am not aware of any effort to upstream dual-GPU VR rendering, especially not within CARLA. To the best of our knowledge, CARLA is not interested in VR as an autonomous vehicle simulator (which wouldn't need VR) whereas we are more interested in placing human drivers in the simulator and improving the immersion via VR. I'm not sure about any efforts by UE4 to enable dual GPU support but this is also something we have discussed internally as well, though we are slightly concerned about the cross-GPU communication overhead to send final data to the VR headset.
1a. There is definitely potentially for this work, especially since we are mostly bottlenecked by raw rasterization performance, so it is still worth looking into. We have looked into [NVidia VRWorks](https://developer.nvidia.com/vrworks/graphics/vrsli) and other [NVidia x Unreal plugins](https://developer.nvidia.com/game-engines/unreal-engine) and they definitely have promise (eg. NVidia DLSS/AMD FSR, VR SLI, etc.) but we haven't gotten around to implementing them yet. Any help would definitely be appreciated! :)

2. VRAM is *an* issue, but it depends on a lot of factors. First off, resolution (of the output VR headset) is a big one since enough data needs to be sent to it for each eye to avoid aliasing. Also, CARLA is very liberal with their materials/textures and lighting effects which all add to the experience of realism but at a big performance cost. We have investigated disabling unnecessary lights (ie. everything except the sun & sky-light & traffic lights) and tuning the LOD (Level Of Detail) settings for all the models to better suit a VR experience, but it is a lot of tuning and was not our top priority. In sum, to minimize VRAM usage, you will be best off using one of the smaller, cheaper maps (ie. not Town03, 04, 05, 10HD), disable all unnecessary lighting & postprocessing effects, and have as little activity in the scene as possible (though this is often not what we want). 
2a. With that said, VRAM is typically not a concern with the RTX 3090 (which has 24G) as we typically see ~8-10G used for normal experiments (plus or minus ~3G depending on the experiment scale). So we haven't had much interest in getting a compute focused card for increased VRAM. 
2b. According to our measurements (which were taken nearly a year ago, so things could be different now) we also found significantly more VRAM usage on Linux using Vulkan compared to Linux using OpenGL for VR rendering (nearly 40% more). But this was a [known issue](https://carla.readthedocs.io/en/0.9.7/configuring_the_simulation/#:~:text=Vulkan%20(if%20installed)%20is%20the,bugs%20when%20running%20the%20simulator.).

tensorturtle, 2022-03-26:
@GustavoSilvera Again, thank you so much for the in-depth reply.
I'll spend some time trying out all this. Best wishes to you and your team!

closed 2022-03-26 by tensorturtle

GustavoSilvera, 2022-03-26:
Awesome, thanks and good luck to you too!

Also a quick note I'd like to make since you mention you are more interested in using Linux, you should verify that the VR headset you are planning to get will work with SteamVR. Notably this does **not** include the Oculus/Meta devices which have a Windows-only compatibility software :(

shh1v, 2022-07-14:
Hey! We were looking for a PC which would be suited for our research, You mentioned that you got around 25-30 fps which is really really low, and can cause severe motion sickness. What pc specifications do you recommend to reach 75 fps if we are not using eye tracking. Would you say VR is not recommended for DReyeVR (or Carla in general)? Would you rather recommend using a monitor display for the simulation?

GustavoSilvera, 2022-07-14:
First off, eye tracking on its own does not affect performance much if at all, it is very cheap to do. The main performance killer is going into VR mode as the scene will have to be rendered twice (once per each eye) and any mirrors will also have to be rendered again (new viewpoints) in VR. Without mirrors we get ~50FPS which we found to be reasonable for a smooth driving experience, with mirrors enabled we see ~30FPS which is more stuttery as you mention, but generally still usable as long as the framerate is consistent and the participant is not making too jerky of motions.

Without VR, the performance should ~double  so if you really need smooth performance, going into flat-screen is the best bet. Since the GPU is the main bottleneck here we don't have any more hardware recommendations since we are using (at time of writing) one of the best off-the-shelf NVidia GPU's and still manage to bottleneck it. Maybe once the NVidia 4000 series cards come out those will be able to singlehandedly improve the performance of DReyeVR by a significant amount.

shh1v, 2022-07-14:
Thanks for the amazing explanation!