HARPLab/DReyeVR #11: DReyeVR for cycles

TR-Monash, 2022-04-06:
Hi,

Thanks for the awesome and interesting work. The video looks promising!
I am currently working with standard Carla to simulate near mis-events involving cyclists (i.e. the ego vehicle is a cycle). We are considering implementing VR capability on Carla for one of our upcoming research. However, I was wondering what sort of challenges are there to adapt DReyeVR for cycles instead of the standard vehicle.
Thanks.

GustavoSilvera, 2022-04-06:
Hi @TR-Monash,

Sounds interesting! Converting the logic to work with cycles should not be too much work (even implementation wise, both are [CarlaWheeledVehicle](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.h) actors). 

From initial thoughts, you would need to modify some VR-specific things such as where the camera (& camera root) is located on the bicycle. You can see how we did this initially in the editor from [this older documentation](https://github.com/gustavosilvera/vr-carla-docs) (slightly out of date, but still mostly accurate). 

At a high level, I think you can replace (reparent) the EgoVehicle's base blueprint (In your case, use one of the Carla cyclist blueprints instead of the TeslaM3 we used) and tweak some of the finer details (like head origin, mirror locations, steering wheel animations, etc.).

Happy to answer more questions :)

TR-Monash, 2022-04-06:
@GustavoSilvera, Thank you very much for the detailed answer. I'll look at the older documentation and try to get an understanding of what sort of changes are needed. At the moment, the project is in its planning stages and I might have some questions later.
Thanks again and I'll close this issue for now.

closed 2022-04-06 by TR-Monash