HARPLab/DReyeVR #12: Assertion failed: px != 0, file shared_ptr.hpp

rampillas, 2022-04-07:
Dear HARPLab team,

We are trying to get a functioning installation of DReyeVR withtout success so far. Our computer specs are:
Windows 10
Nvidia 3090
24Gb RAM
Intel(R) Core(TM) i5-10600K
500Gb SSD
Lenovo Explorer VR Headset
Logitech G29 steering wheel

[We have followed the instructions at Installing DReyeVR to a working Carla 0.9.13 build](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md) and manage to launch CARLA server from the Unreal Engine, and see the car from up above with the VR headset, and steer the car with the wheel, but we cannot change the point of view to the driver's seat.

We have tried launching the script "schematic_mode.py", but it fails with the following output:
`Welcome to CARLA No-Rendering Mode Visualizer

    TAB          : toggle hero mode
    Mouse Wheel  : zoom in / zoom out
    Mouse Drag   : move map (map mode only)

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down

    F1           : toggle HUD
    I            : toggle actor ids
    H/?          : toggle help
    ESC          : quit

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.13
WARNING: Simulator API version  = 0.9.13-dirty
Initializing DReyeVR ego vehicle as hero_actor
initialized DReyeVRSensor PythonAPI client
Assertion failed: px != 0, file C:\workspace\carla\Build\boost-1.72.0-install\include\boost/smart_ptr/shared_ptr.hpp, line 734`

Could you help us to figure out what we need to do? Thanks for your work and best regards.

GustavoSilvera, 2022-04-13:
Hi @rampillas do you have any updates regarding your situation?

inv-mary, 2022-09-26:
Hi
@rampillas   when i was trying to add a safe distance sensor in carla (as per the doc : https://carla.readthedocs.io/en/0.9.13/tuto_D_create_sensor/) im also facing the same issue

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.13
WARNING: Simulator API version  = 0.9.13-1-g8854804f4-dirty
INFO:  Found the required file in cache!  Carla/Maps/TM/Town10HD_Opt.bin 
INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin 
basic agent set destination 
ActorBlueprint(id=sensor.other.safe_distance,tags=[sensor, other, safe_distance])
Assertion failed: px != 0, file C:\workspace\carla\Build\boost-1.72.0-install\include\boost/smart_ptr/shared_ptr.hpp, line 734

and the server crashed

Could you help us to figure out what we need to do? Thanks for your work and best regards.

GustavoSilvera, 2022-09-26:
Hi @inv-mary have you tried rebuilding the `PythonAPI` with command `make PythonAPI` once DReyeVR is installed?

inv-mary, 2022-09-26:
@GustavoSilvera i tried make PythonAPI ,
Still have the same problem

GustavoSilvera, 2022-09-26:
Can you provide the logs for the `make PythonAPI` command, there might be something failing that isn't letting it rebuild properly.