HARPLab/DReyeVR #15: After adjusting camera, DReyeVR_AI.py returns the camera to the original position and is unable to be moved in the Z-plane while driving

aureliuszi, 2022-04-25:
Hello Gustavo (and DReyeVR team)!

This is most likely related to #14 , but I had to reposition the camera in VR view. Upon executing python DReyeVR_AI.py, the camera was returned to the original unadjusted position. Once the DReyeVR_AI.py code was running, I was able to use the arrow keys to turn the camera on the XY axis, but I could not use page up/ page down to reposition the camera on the Zaxis. This was problematic because my "default" position appears to be somewhere on the roof of the car. 

Thanks for all of your help!

See photo below for my view when driving with AI

[image]
!

GustavoSilvera, 2022-04-25:
Hi @aureliuszi good catch! Yes, you are right that this is likely related to #14 because what I think is happening is that the `FirstPersonCamera` that you are unable to reposition is not the default camera (that you already repositioned prior to running `DReyeVR_AI.py`) but instead either the spectator or AI camera that you switch to when using the autopilot handoff. 

Does this camera behaviour persist when you toggle back to the manual control mode? (ie. press `2`)

aureliuszi, 2022-04-25:
Hi again! 

When I press 2, I become detached from the car. It continues to drive without me. I am able to use the arrow keys to move around but unable to use the pageup/ pagedown buttons to adjust on the Z axis.

Thanks!

Zi

GustavoSilvera, 2022-04-25:
Interesting. This is not expected behaviour as pressing 2 should relink you to the vehicle and return control to the user. 

We are working on a fix now.

GustavoSilvera, 2022-04-26:
Wait, sorry I confused myself. Pressing `1` should relink you to the vehicle. `2` should take you to "spectator mode" and `3` is for AI handoff. Is this the same behaviour you are seeing?

aureliuszi, 2022-04-27:
Hi Gustavo! All of this behaves as expected then.

GustavoSilvera, 2022-04-27:
Thanks for the clarification, we are in the process of implementing a shared "camera" so we no longer need to discreetly possess and unpossess the vehicle, but both the AI pilot and human pilot can control the vehicle and cooperate simultaneously.

PR #25 (2022-05-17): CustomActors, Pawn, Python install, Bugfixes
## This is a big one
- Adding "DReyeVR custom actors" to the available options for simulation interaction (added opaque and translucent material blueprints as well, also fully recordable and replayable)
- Creating a new DReyeVR pawn that should fix issues (#14, #15) with the camera and autopilot handoff
- Adding weather to the recorder
- More strictly enforced input priority, keyboard > logitech > autopilot
- Fix ScenarioRunner issue #24
- Adding initial code for foveated rendering (not ready yet)
- Overall several refactors and quality-of-life impr [... cut]
closed 2022-05-17 by ajdroid