HARPLab/DReyeVR #16: Trying to run scenario_runner example script fails - ModuleNotFoundError: No module named 'DReyeVR_utils'

ethicallab, 2022-04-25:
Hello again Gustavo and HARPLab,

I am trying to run the following code/ here is my output:

python run_experiment.py --title "dreyevr_experiment" --route %SCENARIO_RUNNER_ROOT%\srunner\data\routes_custom.xml %SCENARIO_RUNNER_ROOT%\srunner\data\town05_scenarios.json 0
Traceback (most recent call last):
  File "run_experiment.py", line 27, in <module>
    from scenario_runner import ScenarioRunner
  File "C:\Users\kp\Documents\G\carla9.12\scenario_runner\scenario_runner.py", line 39, in <module>
    from srunner.scenarios.route_scenario import RouteScenario
  File "C:\Users\kp\Documents\G\carla9.12\scenario_runner\srunner\scenarios\route_scenario.py", line 52, in <module>
    from DReyeVR_utils import find_ego_vehicle
ModuleNotFoundError: No module named 'DReyeVR_utils'

I have executed the commands to make both DReyeVR and the scenario runner part of DReyeVR.

Any thoughts are appreciated! Thanks again!

GustavoSilvera, 2022-04-25:
Hi @ethicallab. To help diagnose this issue, I'd like to know how you are running `python`. Are you doing it through the native Windows 10 `py` command or a `conda` environment or did you install `python` to the system?

If you use a `conda` environment the dependencies are usually a non-issue as everything is handled as [described here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#optionalusing-conda-for-the-pythonapi). 

In either case, it looks like your Python cannot find the `DReyeVR_utils` module which is a subset of the `carla` module.  In order to give `scenario_runner` access to the `DReyeVR_utils` module you need to make sure `PythonAPI\examples` is located on the PythonPath somewhere. You can add this either to your system PYTHONPATH variable (if you are using the system python) or to the `carla.pth` file if you are using conda. 

Hope this helps!

aureliuszi, 2022-04-27:
Hi! We are running Python through the x64 Native Tools command prompt. Here is our PYTHONPATH (ignore the "carla9.12", it is actually 0.9.13 and I mis-named it)

```
echo %PYTHONPATH%

C:\Users\kp\Documents\G\carla9.12\carla\Build\UE4Carla\0.9.13-1-g8854804f4-dirty\WindowsNoEditor\PythonAPI\carla\dist\carla-0.9.13-py3.7-win-amd64.egg;

C:\Users\kp\Documents\G\carla9.12\carla\Build\UE4Carla\0.9.13-1-g8854804f4-dirty\WindowsNoEditor\PythonAPI\carla\agents;

C:\Users\kp\Documents\G\carla9.12\carla\Build\UE4Carla\0.9.13-1-g8854804f4-dirty\WindowsNoEditor\PythonAPI\carla;

C:\Users\kp\Documents\G\carla9.12\carla\Build\UE4Carla\0.9.13-1-g8854804f4-dirty\WindowsNoEditor\PythonAPI
```

Will add PythonAPI/examples

aureliuszi, 2022-04-27:
Hi! Here is the traceback after adding examples to the PYTHONPATH

```
python run_experiment.py --title "dreyevr_experiment" --route %SCENARIO_RUNNER_ROOT%\srunner\data\routes_custom.xml %SCENARIO_RUNNER_ROOT%\srunner\data\town05_scenarios.json 0
Starting scenario runner
Traceback (most recent call last):
  File "run_experiment.py", line 175, in <module>
    main()
  File "run_experiment.py", line 168, in main
    start_scenario_runner(scenario_runner_instance)
  File "run_experiment.py", line 35, in start_scenario_runner
    result = scenario_runner_instance.run()
  File "C:\Users\kp\Documents\G\carla9.12\scenario_runner\scenario_runner.py", line 509, in run
    result = self._run_route()
  File "C:\Users\kp\Documents\G\carla9.12\scenario_runner\scenario_runner.py", line 470, in _run_route
    route_configurations = RouteParser.parse_routes_file(routes, scenario_file, single_route)
  File "C:\Users\kp\Documents\G\carla9.12\scenario_runner\srunner\tools\route_parser.py", line 57, in parse_routes_file
    tree = ET.parse(route_filename)
  File "C:\Users\zhuang\AppData\Local\Programs\Python\Python37\lib\xml\etree\ElementTree.py", line 1197, in parse
    tree.parse(source, parser)
  File "C:\Users\zhuang\AppData\Local\Programs\Python\Python37\lib\xml\etree\ElementTree.py", line 587, in parse
    source = open(source, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\kp\\Documents\\G\\carla9.12\\scenario_runner\\srunner\\data\\routes_custom.xml'
```

I attempted to run the test given on the [scenario runner documentation](https://carla-scenariorunner.readthedocs.io/en/latest/getting_scenariorunner/) and here is the error:

```
python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Traceback (most recent call last):
  File "scenario_runner.py", line 607, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 511, in run
    result = self._run_scenarios()
  File "scenario_runner.py", line 451, in _run_scenarios
    result = self._load_and_run_scenario(config)
  File "scenario_runner.py", line 355, in _load_and_run_scenario
    if not self._load_and_wait_for_world(config.town, config.ego_vehicles):
  File "scenario_runner.py", line 307, in _load_and_wait_for_world
    self.world = self.client.load_world(town)
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000
```

aureliuszi, 2022-04-27:
Hi again, after restarting my computer and adding PythonAPI/Utils to the PYTHONPATH, I was able to run the example scenario runner test from the [documentation](https://carla-scenariorunner.readthedocs.io/en/latest/getting_scenariorunner/) and produce these errors. In this case, the leading vehicle never appeared.

```
>python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

C:\Users\kp\Documents\G\carla9.12\scenario_runner>python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
INFO:  Could not find the required file in cache, downloading...  Carla/Maps//OpenDrive/Town01.xodr
INFO:  Could not find the required file in cache, downloading...  Carla/Maps//Nav/Town01.bin
INFO:  Could not find the required file in cache, downloading...  Carla/Maps//TM/Town01.bin
INFO:  Found the required file in cache!  Carla/Maps/TM/Town01.bin
Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Traceback (most recent call last):
  File "scenario_runner.py", line 412, in _load_and_run_scenario
    self.manager.run_scenario()
  File "C:\Users\kparodi\Documents\G\carla9.12\scenario_runner\srunner\scenariomanager\scenario_manager.py", line 140, in run_scenario
    self.cleanup()
  File "C:\Users\kparodi\Documents\G\carla9.12\scenario_runner\srunner\scenariomanager\scenario_manager.py", line 96, in cleanup
    CarlaDataProvider.cleanup()
  File "C:\Users\kparodi\Documents\G\carla9.12\scenario_runner\srunner\scenariomanager\carla_data_provider.py", line 800, in cleanup
    if actor is not None and actor.is_alive and actor.type_id != CarlaDataProvider.ego_DReyeVR:
AttributeError: type object 'CarlaDataProvider' has no attribute 'ego_DReyeVR'
type object 'CarlaDataProvider' has no attribute 'ego_DReyeVR'
Traceback (most recent call last):
  File "scenario_runner.py", line 607, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 511, in run
    result = self._run_scenarios()
  File "scenario_runner.py", line 451, in _run_scenarios
    result = self._load_and_run_scenario(config)
  File "scenario_runner.py", line 430, in _load_and_run_scenario
    self._cleanup()
  File "scenario_runner.py", line 186, in _cleanup
    self.manager.cleanup()
  File "C:\Users\kparodi\Documents\G\carla9.12\scenario_runner\srunner\scenariomanager\scenario_manager.py", line 96, in cleanup
    CarlaDataProvider.cleanup()
  File "C:\Users\kparodi\Documents\G\carla9.12\scenario_runner\srunner\scenariomanager\carla_data_provider.py", line 800, in cleanup
    if actor is not None and actor.is_alive and actor.type_id != CarlaDataProvider.ego_DReyeVR:
AttributeError: type object 'CarlaDataProvider' has no attribute 'ego_DReyeVR'
```

GustavoSilvera, 2022-04-27:
Right, ok so for `scenario_runner` we currently only support the `--route` mode. Can you try one of those? 

PS: the `run_experiment.py` script is essentially a wrapper around the `start_recording.py` and `scenario_runner.py --route` scripts, so you should be able to run the initial command again. Just make sure the `routes_X.xml` file exists, in our case we created a new file called `routes_custom.xml` but by default the [CARLA `scenario_runner`](https://github.com/carla-simulator/scenario_runner/tree/master/srunner/data) repo only contains these `xml` files.

aureliuszi, 2022-04-27:
To clarify, should I have `routes_custom.xml` if I built the DReyeVR portion of scenario_runner? I currently only have the default files in my `srunner/data folder`...

GustavoSilvera, 2022-04-27:
No, if you cloned from CARLA then you only get the default. Yet unfortunately it appears we did not include this file in our scenario_runner fork which we should've. I'll upload the file here soon and include it in a push to our fork.

For now you should be able to use the default routes files though.

aureliuszi, 2022-04-27:
Thanks for the clarification! Based on your instructions I made the following command: 

`python run_experiment.py --title "dreyevr_experiment" --route %SCENARIO_RUNNER_ROOT%\srunner\data\routes_debug.xml %SCENARIO_RUNNER_ROOT%\srunner\data\all_towns_traffic_scenarios1_3_4.json 0`

I successfully handed off control to the AI by pressing `3`. The camera wasn't rotating with the car but the scenario was otherwise executing. 

I want to ask a couple of things: after I run the above command, there is no need to run `python manual_control.py` right? Doing that caused my scenario_runner to crash the first time. 

Secondly. is there any way to remove the big red target in the 2D mode? Ideally I'm looking to create identical recordings of 2D and VR scenarios for my experiment.

[image]

Finally, if the `--route` mode is the only supported mode, I should develop my scenarios as routes and test them with DReyeVR, correct?

GustavoSilvera, 2022-04-27:
Yes, everything is working as expected. You don't need to run `manual_control.py` since all the input handling is happening inside the server (UE4 instance) and you can use either the keyboard or logitech hardware to control the ego vehicle (assuming the AI isn't driving)

The camera not rotating is exactly issue #14 and we are working on a patch now.

The big red target is what we use as the eye reticle to showcase the location where a user is looking, if you don't want this then you can simply change the params in `Configs/DReyeVRConfig.ini` specifically `DrawFlatReticle` to `False` will stop the target from showing when you run `CarlaUE4.exe`. If you are running in VR and notice the reticle on the flat-screen UE4 window (This is called the ["Spectator screen"](https://docs.unrealengine.com/4.26/en-US/SharingAndReleasing/XRDevelopment/VR/DevelopVR/VRSpectatorScreen/)) then you can disable the "Spectator" related params such as `DrawSpectatorReticle`

ajdroid, 2022-04-27:
In `Unreal\CarlaUE4\Config\DReyeVRconfig.ini`, set the `DrawSpectatorReticle` to `False` to turn off the target while VR is running and `DrawFlatReticle` to `False` to turn it off during non-VR operation.

Yes, we support only the route paradigm for developing scenarios.

GustavoSilvera, 2022-04-28:
Hi @aureliuszi sure this sounds like a good fit for DReyeVR. 
for 1.) any recordings you make (either in VR or in 2D mode) can be replayed in the other format as well (though it is weird to replay in VR since the camera will match the recordings rather than the actual VR orientation, so it is recommended to replay in 2D mode.)
for 2. Sure, you should be able to program a CARLA-compatible AI driver that has super-human abilities to drive the car. We are currently working on a patch to make the AI handoff/takeover less clunky (which should fix #14) and that will be out soon (ideally in the next few weeks). 
for 3. This is also not a problem, you can choose where in the vehicle you want to place the camera by editing the `CameraInit` variable in the `DReyeVRConfig.ini` params file. I assume you'd want to record the autopilot driving from the passenger seat (while in VR), this sound totally feasible and I'm interested to see what you come up with!

I believe @ajdroid is more familiar with the route-creation aspect of scenario-runner than I am, so I will hold on answering this. 

Hope this helps.

ajdroid, 2022-05-02:
@aureliuszi creating routes is an interactive process where you need to pick waypoints on a map to construct a route by using their locations. This is best done in the editor.
An example route can be found [here ](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/data/routes_debug.xml) and example usage can be seen in the video [here](https://youtu.be/dcnnNJowqzM?t=311)
You can pass in the `--debug` flag when you use scenario runner to see the waypoints visually on the map.

GustavoSilvera, 2022-05-18:
Hi @ethicallab do you have any updates on this issue, we recently specified on the [FAQ page](https://github.com/HARPLab/DReyeVR/wiki/Frequently-Asked-Questions) some example questions that are related to these and should help.

GustavoSilvera, 2022-05-30:
Due to lack of activity, will close this. Feel free to reopen if issue arises again.

closed 2022-05-30 by GustavoSilvera

FBanani, 2022-07-13:
hi 

> own05_scenarios.json

hi Gustavo said I can ask you
[image]

I want to bring highway in CarlaDreyeVr because I want to use VR headset and G29 
I change [DefaultEngine](https://github.com/HARPLab/DReyeVR/blob/main/Configs/DefaultEngine.ini#L11-L18) but noting happened when I make launch or execute CarlaUE4. exe --vr all of them is in town 3 
also 
[image]
but there is no town4 in 
[image]
what should I do?
also I need to add some signs to highway after I bring it. how can I apply these changes and it is compatible with DReyeVr?

GustavoSilvera, 2022-07-14:
Hi. the Town04 you are looking for is actually in `Unreal/CarlaUE4/Content/Carla/Maps/Town04.uasset`. Editing the DefaultEngine.ini should work. It should look something like this

```ini
[/Script/EngineSettings.GameMapsSettings]
EditorStartupMap=/Game/Carla/Maps/Town04.Town04
GameDefaultMap=/Game/Carla/Maps/Town04.Town04
ServerDefaultMap=/Game/Carla/Maps/Town04.Town04
GlobalDefaultGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C
GameInstanceClass=/Script/Carla.CarlaGameInstance
TransitionMap=/Game/Carla/Maps/Town04.Town04
GlobalDefaultServerGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C
```

Once you've run `make package` you should find this file in the newly created `WindowsNoEditor/CarlaUE4/Config/DefaultEngine.ini`. Edit this file and then reopen the carla.exe executable and you should be in Town04 automatically.

FBanani, 2022-07-14:
> Hi. the Town04 you are looking for is actually in `Unreal/CarlaUE4/Content/Carla/Maps/Town04.uasset`. Editing the DefaultEngine.ini should work. It should look something like this
>
> ```ini [...]

I have done it but I need to some changes like adding signs to highway but I could not editing because it is in town3\
[image]

ajdroid, 2022-07-14:
In the editor, you can simply press Ctrl+O and then navigate to `Content/Carla/Maps` and select Town04 to load that map

FBanani, 2022-07-14:
> In the editor, you can simply press Ctrl+O and then navigate to `Content/Carla/Maps` and select Town04 to load that map

thank you so much. I have another question if I made some changes in the editor is it apply when I execute "CarlaUE4.exe -vr"?

ajdroid, 2022-07-14:
No, you'll have to `make package` again

FBanani, 2022-07-14:
> No, you'll have to `make package` again

I am new Carla user could you explain a little more?

FBanani, 2022-07-14:
> I have done it but I need to some changes like adding signs to highway but I could not editing because it is in town3 [image]

why it is not working for twon07
[image]
I have done changes in DefaultEngine.ini
[image]

GustavoSilvera, 2022-07-18:
I think Town07 issue might be related to [this Carla issue](https://github.com/carla-simulator/carla/issues/3684). Can you try the solution suggestion there?

FBanani, 2022-07-22:
hi I found the solution
It seems that Town06 and Town07 are not by default packaged. You may enable this in CarlaUE4Editor:

File -> Package Project -> Packaging Settings.
[image]
project settings: then in map & modes
in default maps change to the desire town 
[image]
then close unity and make package

Nitro60zeus, 2023-06-01:
> Hi! Here is the traceback after adding examples to the PYTHONPATH
>
> ```
> python run_experiment.py --title "dreyevr_experiment" --route %SCENARIO_RUNNER_ROOT%\srunner\data\routes_custom.xml [...]

Hi!
I am also getting this error. Everytime i restart the PC, it works once, but then again this error
SO its like everytime i need to run, i have to restart... Any guide?

Btw I am talking about the "runtime error: time-out of 1000ms"

GustavoSilvera, 2023-06-01:
You probably have a rogue (unkilled) python instance (PythonAPI client) still running on your machine, which is taking precedence over any new PythonAPI clients that you create. We had this a lot on Windows, try going into your task manager and killing any existing UE4Editor, CarlaUE4.exe, and python instances, then try again.