HARPLab/DReyeVR #17: Position calibration unstable.

sunchongren, 2022-04-26:
Hi, thanks for the VR implementation!

I encontered a problem that when entering VR mode, the driver's position calibration seems failed. Is there any way to re-calibrate it?
[image]

GustavoSilvera, 2022-04-26:
Hi @sunchongren this is a common issue, likely related to #14 and for now there is a quick fix you can try which is to use the arrow keys + pgup/pgdwn to manually reposition the camera position to the driver's seat. 

We are currently working on a fix for the next update.

sunchongren, 2022-04-26:
Thanks! It is solved. I encountered another question after I start the application from the path ``` E:\carla\Build\UE4Carla\0.9.13-1-g8854804f4-dirty\WindowsNoEditor```

      ./CarlaUE4.exe -vr

The error occurs when I want to run any python operation file such as DReyeVR_AI.py or dynamic_weather.py.
Any possible reason for this?

GustavoSilvera, 2022-04-26:
Hi @sunchongren feel free to create a new issue with the Python logs so we can get a better sense of what the problem is. Thanks!

closed 2022-04-26 by ajdroid