HARPLab/DReyeVR #21: Bug on view calibration

sunchongren, 2022-04-27:
For the view calibration error (driver's view not inside the car):
[image]

I find the view is always higher than the car.
I did some experiments on changing different room setup. I found that when you set your height (VR headset height) to 0 (which means your eye's height level is the floor), the bug is solved. And I also found the actual view height above the normal view point (inside the car) is exactly your height (headset height to floor).

GustavoSilvera, 2022-04-28:
Yes, this is a known issue that we and other users have been experiencing (see #15) and we are working on a fix. 

For now the temporary solutions are to use the arrow-keys + pgup/pgdown to manually readjust the camera position (or you can use some of the face buttons on the wheel if you are using the logitech hardware), see #17 

Another solution we take is to run a scenario using `run_experiment.py` this often correctly places the camera upon reloading the world, so we assume this is a problem only on the first run (in vr) and can be fixed with a world reload. 

Regardless, we are working on a patch and will keep you updated

GustavoSilvera, 2022-05-17:
Hey, @sunchongren please check out the [upgrading-DReyeVR](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#upgrading-dreyevr) section of the documentation to pull from `main` to get all the latest bugfixes. This issue should be fixed with those changes.

closed 2022-05-17 by GustavoSilvera

sunchongren, 2022-06-01:
@GustavoSilvera 
Hi Gustavo, thanks for the update! For the ```make clean CARLA=/PATH/TO/CARLA SR=/PATH/TO/SR```, does the SR mean scenario runner or SRanipalEye?
And also, a typo of "unreal" in the file.

sunchongren, 2022-06-01:
@GustavoSilvera 
The typo is in install.md. I just proposed a change in that file (in the upgrading DReyeVR section)