HARPLab/DReyeVR #22: Steering Wheel Support

mullenj, 2022-04-28:
Hello,

I am using the Logitech Momo Racing wheel which according to the LogitechWheel github is supported by the plugin. However, I am getting no input into the system in either the full package or in the editor. Please let me know the right procedure for setting my steering wheel up or debugging this issue. I may be able to map the controls manually in the editor if I knew which blueprints I would need to edit to replace the keyboard controls (which do work for me).

GustavoSilvera, 2022-04-28:
Hi @mullenj can you try debugging the logitech outputs as described in #8 

There is a flag in DReyeVRConfig.ini (at the bottom) that will enable verbose logitech logging so you should see if anything is being picked up from there.

mullenj, 2022-04-28:
Everything works now after installing the Logitech gaming software. Thanks for the help.

closed 2022-04-28 by mullenj