HARPLab/DReyeVR #23: How to disable the speed display?

aureliuszi, 2022-05-02:
[image]

Hello friends! 

I'm wondering how to disable the red speed display on the right side. 

I have managed to disable the big red target and the FPS count by setting these variables in `C:\...\...\
CARLA_0.9.13-1-g8854804f4-dirty\WindowsNoEditor\CarlaUE4\Config` to false. I can't seem to find the speed display though. 

```
DrawFPSCounter=False; draw's FPS (frames per second) in top right corner of flat screen
DrawFlatReticle=False; reticle in flat-screen mode
```

GustavoSilvera, 2022-05-02:
Hi @aureliuszi unfortunately you currently cannot disable them from the runtime params `DReyeVRConfig.ini`. 

However, you can disable them from modifying the source code and rebuilding. Specifically comment out the call to `ConstructDashText` in `EgoVehicle.cpp` (In the constructor) and the `UpdateDash` call in `EgoVehicle::Tick`.

This will ensure that the dash text (speedometer, gear indicator, and turn signals) never gets created and does not tick.

aureliuszi, 2022-05-02:
To confirm, this involves opening `EgoVehicle.cpp` in the `C:\..\..\carla9.12\carla\Unreal\CarlaUE4\Source\CarlaUE4\DReyeVR` and rebuilding via `make carla CARLA=../carla `? 

There is another `EgoVehicle.cpp` in `C:\..\..\carla9.12\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\CarlaUE4\Development\CarlaUE4`

GustavoSilvera, 2022-05-02:
Correct, the intermediate files are effectively cache files that will get rewritten. So you never need to touch them.

GustavoSilvera, 2022-05-02:
Also, you wouldn't need to `make Carla CARLA=../Carla`, you only need to `make launch` or `make package` once you edit the file within the Carla directory (as opposed to the DReyeVR directory)

GustavoSilvera, 2022-05-18:
We recently added some documentation in the [FAQ page](https://github.com/HARPLab/DReyeVR/wiki/Frequently-Asked-Questions) on how to modify DReyeVR source once installed, which should answer some similar questions. 

Do you have any more updates with this issue that we should take care of?

GustavoSilvera, 2022-05-30:
Due to lack of activity, we will close this. If there is another issue, feel free to reopen.

closed 2022-05-30 by GustavoSilvera