HARPLab/DReyeVR #31: UE4-CarlaUE4 Editor crash in VR mode. " Fatal Error! "

ohashi3399, 2022-06-23:
I really appreciate your work for developing such great tools "DReyeDR".
I have an issue about activation failure of VR mode.
It would be very helpful if you could provide a way for avoiding the issue or any information.

The error what I face is shown as below.
[image]

Environmental setup is listed below.
- OS: Windows 10 Pro notebook
- GPU: NVIDIA GeForce RTX 3080
- VR goggle: HTC VIVE Pro Eye
- Other Hardware: logicool G29 (Pedals & steering)
- Carla version: 0.9.11
- UE4 version: 4.24
- DReyeVR version: 0.9.11
- Steam VR version: 1.22.13

I followed your detailed instructions for building environment refering to 
- UE4.24 and carla0.9.11 installation: https://carla.readthedocs.io/en/0.9.11/build_windows/
- DReyeVR0.9.11: https://github.com/HARPLab/DReyeVR/blob/0.9.11/Docs/Install.md
- (error occured) Setting up VR mode: https://github.com/HARPLab/DReyeVR/blob/main/Docs/SetupVR.md

You really helped me by detailed instructions, especially some bugs in plugin.
I finished up the above environmental setups without no errors except above one.
The error was happened when after selectiong "VR preview" mode of Play button in CarlaUE4 Editor.

When VIVE Pro was disconneted to PC, I run in "Stand alone" mode without errors. 
When VIVE Pro was conneted to PC, I run in "Stand alone" mode with the above error. 
I think there's an error in part of VR setting.

Could you know how to deal with the error above?
Especially I want to clarify which parts made mistakes.
Thanks.

ohashi3399, 2022-06-24:
Hi Gustavo-san, thanks for your quick response. I understand your thought and it makes sense. I decided to version up from 0.9.11 to 0.9.13. It will take a couple of days due to my environment issue. If everything has done, I will share that. Again, thanks for your coorporation.

ohashi3399, 2022-06-30:
I apologize for waiting the answer Gustavo-san.
I removed previous version and I reinstalled all setups including Visual Studio2019, UE4.26, Carla(latest), DReyeVR(0.9.13), and I confirmed it properly worked!! 

[image]
*this image is not VR mode.

I really appreciate for your response and effort.

At the end, could you share your GPU device and fps at VR mode on your environment?
I used NVIDIA GeForce RTX 3080 Laptop GPU and around 10fps at VR mode. I read your great work you published(https://arxiv.org/pdf/2201.01931.pdf), it seems to be clear image shown in Fig1(b). If you have any tricks to improve fps, it would be very helpful if you could share.
Thanks in advance.

ajdroid, 2022-06-30:
Hi our GPU used was the RTX 3090 and we see anywhere between 35-60 fps depending on a few factors (amount of non-ego vehicles in the scene, whether or not the mirrors are turned on, the map used, and the part of the map we are rendering in)

GustavoSilvera, 2022-06-30:
Hi @ohashi3399 glad it worked!

To elaborate on @ajdroid's response, check out [this issue](https://github.com/HARPLab/DReyeVR/issues/10) on performance with DReyeVR. The first (& easiest) step for a big performance boost is to disable the realistic mirrors if you don't need them (See `DReyeVRConfig.ini` and set the `_MirrorEnabled=False`). Another step if you don't mind the graphical fidelity loss would be to run DReyeVR with the flag `-quality-level=Low` which should also significantly improve the framerate. 

Aside from that you'll probably want to run diagnostics on your machine to see what is the bottleneck (CPU or GPU... its probably the GPU) and tune other graphics settings like resolution scale (pixel density) as described [here](https://docs.unrealengine.com/4.26/en-US/SharingAndReleasing/XRDevelopment/VR/DevelopVR/ContentSetup/).

Hope this helps!

closed 2022-06-30 by GustavoSilvera