HARPLab/DReyeVR #34: Modify PythonAPI to manuipulate boolean variables inside EgoVehicle.h or EgoVehicle.cpp

shh1v, 2022-07-04:
Hello, It's me, again.
I initialized two variables in EgoVehicle.h, and the behavior of the Heads-Up display I implemented ([see this](#32)) depends on the value of these two boolean variables. How can I modify PythonAPI such that it would allow me to manipulate these boolean variables using python scripts? I know this is a very high-level description of my problem, but I would really appreciate it if you could guide me to the right resources, or just give me a basic idea of how I can approach the implementation.

GustavoSilvera, 2022-07-05:
Hi @shh1v this is definitely easier said than done, but I think it should be doable. I can describe the approach you'd need to take to add a bidirectional communication channel from the PythonAPI to the DReyeVR/EgoVehicle cpp:

This is generally how the pipeline flows currently (for logging EgoSensor data to the PythonAPI)
1. EgoVehicle.cpp contains logic such driver inputs: [EgoVehicle.h/cpp](https://github.com/HARPLab/DReyeVR/blob/51ea84a65df001801f582051dc3899d996d3f7b9/DReyeVR/EgoVehicle.h#L135-L140)
2. EgoSensor gets the data from the EgoVehicle: [EgoSensor.cpp](https://github.com/HARPLab/DReyeVR/blob/51ea84a65df001801f582051dc3899d996d3f7b9/DReyeVR/EgoSensor.cpp#L110)
3. EgoSensor (which inherits from an ADReyeVRSensor (which inherits from an ACarlaSensor)) has a tick that streams data: [DReyeVRSensor.cpp](https://github.com/HARPLab/DReyeVR/blob/main/Carla/Sensor/DReyeVRSensor.cpp#L103-L144)
4. The receiving end of the stream is the serializer/deserializer in LibCarla: [DReyeVRSerializer.h](https://github.com/HARPLab/DReyeVR/blob/main/LibCarla/Sensor/s11n/DReyeVRSerializer.h)
5. (Asynchronously, from the PythonAPI call) The event to receive this data is defined in LibCarla: [DReyeVREvent.h](https://github.com/HARPLab/DReyeVR/blob/main/LibCarla/Sensor/data/DReyeVREvent.h)
6. The PythonAPI sensor is defined to execute these events on any listen callbacks: [SensorData.cpp](https://github.com/HARPLab/DReyeVR/blob/main/LibCarla/SensorData.cpp#L577-L617)
7. (Bonus: when printing the DReyeVR sensor in Python, the `__str__` method is implemented [here](https://github.com/HARPLab/DReyeVR/blob/51ea84a65df001801f582051dc3899d996d3f7b9/LibCarla/SensorData.cpp#L170-L185))

So this is the pipeline we take for data to travel (one way) from the DReyeVR code (EgoSensor/EgoVehicle) to the PythonAPI. Since you'd also like communication in the other direction I would recommend trying to follow this pipeline backwards. This would look something like modifying step 6) to add a function definition for a `set_your_bool` function. Then you'd have to define this logic in the event handler (5) and figure out a way to communicate over the network (serializer/deserializer from LibCarla (Python) to CarlaUE4 (cpp) which might be tricky and I don't immediately know of a way to do). Once you have this communication done it should be straightforward to add the interface methods in DReyeVR. 

Again, I'd like to note that I'm not actually very sure how to go about implementing the communication from LibCarla -> CarlaUE4 (Since the current read-only implementation is CarlaUE4 -> LibCarla). If you can't find out how to do it then I'd actually suggest trying another approach if possible. For instance, add a button that activates the toggle between your boolean (which can then be triggered from the server itself) or if you only need to set the variables once per run I'd recommend using our DReyeVRConfig.ini config file and adding your new variables in the code. 

Hope this helps!

shh1v, 2022-07-05:
Thank you so much for your response. This seems way complicated to execute, and not worth implementing just to conduct research in my case. I think I would be able to manipulate the settings of my heads-up display without the help of Python API. I could create a local settings text file, and the game would read this file to determine the behavior of the Heads-Up Display.
Thank you for explaining the flow of your code. It helped me a lot in my implementation

closed 2022-07-05 by shh1v

GustavoSilvera, 2022-07-05:
Sure thing! I'd like to note that we already have implemented that local settings text file idea you are interested in and we use it all the time: See [DReyeVRConfig.ini](https://github.com/HARPLab/DReyeVR/blob/main/Configs/DReyeVRConfig.ini). Just add a variable there (for example [this one](https://github.com/HARPLab/DReyeVR/blob/51ea84a65df001801f582051dc3899d996d3f7b9/DReyeVR/EgoVehicle.cpp#L47)) and follow the typical flow as seen in our several other source files. 

For more information see [this section](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#using-our-custom-config-file) of the documentation