HARPLab/DReyeVR #35: unreal engine

FBanani, 2022-07-05:
After running make launch and loading the editor, when I dropdown menu I could not select V
[image]
R
could you help me what I do wrong?

GustavoSilvera, 2022-07-05:
Hi @FBanani, first off this looks like a vanilla Carla 0.9.13 install. So make sure to use DReyeVR content you run `make install` from the DReyeVR directory once (a bunch of files will copy over) and then go back to the Carla directory and run `make launch` there. (See [`install.md`](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md) for more specifics). Once DReyeVR is installed you'll be placed in Town03 by default (so you'll know that its not vanilla carla). 

Then you should ensure that your headset is connected and active (yours looks asleep) while the UE4 editor is loading and you have the SteamVR plugin installed as described in [`SetupVR.md`](https://github.com/HARPLab/DReyeVR/blob/main/Docs/SetupVR.md). Then you should be able to select VR Preview. Or if all else fails you should be able to open the standalone packaged game (after `make package`) and add the flags `-vr` when you run `CarlaUE4.exe`. 

Hope this helps!

FBanani, 2022-07-05:
thank you. I successfully update content
[image]
but I have another issue
[image]
when I copy make PythonAPI && make launch in "x64  Native Tools Command Prompt for VS 2019" it show me this error:
[image]
what should I do?

GustavoSilvera, 2022-07-05:
Try cleaning and rebuilding: `make clean && make PythonAPI`. And ensure the `zlib-source` directory is entirely removed before running `make PythonAPI` again. This is a Carla error, you haven't actually installed the DReyeVR content yet. 

Tip: if you don't need the PythonAPI yet, you can actually skip this step (as long as `make launch` works) and then continue with the instructions to install DReyeVR and then keep using the editor.

FBanani, 2022-07-05:
I fixed it it is because of the different version of zlib
we have to change carla/Util/InstallersWin/install_zlib.bat and changing line 51 to "set ZLIB_VERSION=1.2.11" instead of "set ZLIB_VERSION=1.2.12
Thank you

FBanani, 2022-07-05:
I have another issue 
[image]
I have the same issue as this one before and by adding "\" to environment in UE4_ROOT,  it was fixed. now after make PythonAPI && make launch I again find this issue, what should I do?

FBanani, 2022-07-05:
I found a problem when I make PythonAPI, although it said successfully installed but it did not make dis folder in  PythonAPI\carla. I tried make clean then make PythonAPI for 3 times, but it does not make dist folder anytime. I copy the dist folder to PythonAPI\carla, it was made before when I install carla in windows. It is work but hopefully does not make problem in future!

FBanani, 2022-07-05:
when I want to do "make install CARLA=../carla" in the "Windows x64 Native Tools Command Prompt for VS 2019":
[image]
I faced this issue, I do not know why?

GustavoSilvera, 2022-07-05:
Okay can you try closing all UE4 instances (or rebooting) to fix the permission error. I think this because another application (probably UE4) has a lock on this file and Windows won't let you edit it live. 

I'm not sure about the compilation problem. Can you upload the entire build log?

FBanani, 2022-07-06:
I restart my PC and it is work. thank you

FBanani, 2022-07-06:
I want to make launch:
[image]
the dist file is created:
[image]
but I have an error:
[image]

GustavoSilvera, 2022-07-06:
Hi, based on a [similar issue](https://github.com/carla-simulator/carla/issues/2776) I think you need to define the UE4_ROOT environment variable since it appears to be missing. I'd recommend doing this by [creating a new environment variable](https://superuser.com/questions/949560/how-do-i-set-system-environment-variables-in-windows-10) called UE4_ROOT and assigning it the value of the folder where you installed UE4 (4.26 Carla fork) to.

GustavoSilvera, 2022-07-06:
To elaborate further, you should have gone through [this installation page](https://carla.readthedocs.io/en/latest/build_windows/) to get Carla up and running. So make sure to follow the steps to get UE4 (Carla fork) installed properly.

GustavoSilvera, 2022-07-14:
I'm going to assume this is resolved for the time being. Feel free to reopen if more questions come up.

closed 2022-07-14 by GustavoSilvera