HARPLab/DReyeVR #40: Enable autonomous driving mode for ego vehicle using C++

shh1v, 2022-07-11:
In your documentation, it was indicated that currently, AI handover is only supported through PythonAPI. But unfortunately, it doesn't serve well for my study implementation. Why is AI handover not supported using C++? Is there a way I could implement it?
I really appreciate any help you can provide.

GustavoSilvera, 2022-07-11:
I think there might be some confusion here that I'd like to clear up. 

The current implementation of DReyeVR "Autopilot" uses the same `SetAutopilot` function that CARLA uses for their NPC agents. At its core, this uses CARLA's [`TrafficManager`](https://carla.readthedocs.io/en/latest/adv_traffic_manager/) which acts like a simple PID controller for throttle/steering that follows a predefined route. To our knowledge the most straightforward (and intended) way to enable and interact with the TrafficManager was through the PythonAPI as we do in [`DReyeVR_AI.py`](https://github.com/HARPLab/DReyeVR/blob/main/PythonAPI/DReyeVR_AI.py). There is no technical limitation as to why you can't implement the interaction to start and communicate with the TM in C++, we just haven't done so yet. 

On the other hand, the "handoff" is what we define as the action which switches who/what is in control of the EgoVehicle. By default the user (human) is in control and can drive with keyboard/logi-wheel, but as per our documentation you can also (through [C++ events](https://github.com/HARPLab/DReyeVR/blob/2282adc009a97cb9f7590527125b655b0dc8fb5e/DReyeVR/LevelScript.cpp#L169-L223)) execute this handoff at any time and pass over control to the Autopilot (assuming the TM is running).

shh1v, 2022-07-11:
Hi! Thank you so much for the quick response. I am quite new to Carla. Could you please direct me to the documentation for setting autopilot using Traffic Manager? I would really apprecaite it.

GustavoSilvera, 2022-07-11:
Well the TL;DR of it that we use is to first enable a [TM instance in the PythonAPI](https://github.com/HARPLab/DReyeVR/blob/2282adc009a97cb9f7590527125b655b0dc8fb5e/PythonAPI/DReyeVR_AI.py#L125-L128). Then with that running we can just call [`SetAutopilot(true)`](https://github.com/HARPLab/DReyeVR/blob/2282adc009a97cb9f7590527125b655b0dc8fb5e/DReyeVR/EgoVehicle.cpp#L221-L226) in the EgoVehicle. 

For more in-depth documentation here is [CARLA's docs](https://github.com/HARPLab/DReyeVR/blob/2282adc009a97cb9f7590527125b655b0dc8fb5e/PythonAPI/DReyeVR_AI.py#L125-L128).

GustavoSilvera, 2022-07-11:
If you are more interested in the innards of the wheeled-vehicle AI controller check out CARLA's [WheeledVehicleAIController](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.h) class for this. (Note that our EgoVehicle's default controller inherits from this class so we have all the same functionality.

shh1v, 2022-07-11:
This helps me a lot. Thank you. I will close this issue soon once I am able to implement it :)

closed 2022-07-12 by shh1v