HARPLab/DReyeVR #41: Autopilot not working as expected

shh1v, 2022-07-12:
I am running the DReyeVR_AI.py script while running [DReyeVR(OFR)](https://github.com/shh1v/DReyeVROFR). However, the autopilot is not working as expected on the ego vehicle. The car keeps on braking even though the road is completely empty. What are the possible reasons why this is happening? [Here is a screen capture of the simulator.](https://drive.google.com/file/d/1xkihBS9BaMd3zmSB2oxm1ODIeKr9bMNN/view?usp=sharing)

shh1v, 2022-07-12:
It seems like the car breaks as soon as the ego vehicle attains a speed of 20 mph.

GustavoSilvera, 2022-07-13:
Ah ok yes, this sounds familiar. The other vehicles' erradic behaviour where they crash into each other and so-forth is because the PythonAPI is trying to operate at a synchronous pace while the server (UE4 instance) is unpredictable with a variable refresh rate depending on performance. 

The easiest way to combat this is to run the controller for the PythonAPI in sync with the server. You can do this by including the `--sync` flag in scenario runner (or `run_experiment.py` if you're using that). More info [here](https://github.com/carla-simulator/scenario_runner/issues/328)

GustavoSilvera, 2022-07-13:
This might be related to the EgoVehicle AI braking, but I'm not sure since I haven't seen this before. Can you try reproducing the issue once all the non-ego vehicles are operating correctly (using `--sync` flag)

shh1v, 2022-07-13:
> Ah ok yes, this sounds familiar. The other vehicles' erradic behaviour where they crash into each other and so-forth is because the PythonAPI is trying to operate at some asynchronous pace while the server (UE4 instance) is unpredictable with its own variable refresh rate depending on performance.
> 
> The easiest way to combat this is to run the controller for the PythonAPI in sync with the server. You can do this by including the `--sync` flag in scenario runner (or `run_experiment.py` if you're using that). More info [here](https://github.com/carla-simulator/scenario_runner/issues/328)

How can I run scripts that use PythonAPI in sync with CARLA if I am not using scenario runner? Moreover, can you please example the use of run_experiment.py and where is it located? Sorry

shh1v, 2022-07-13:
Okay, so according to Carla's documentation, I can set the synchronous mode using:
```
settings.synchronous_mode = True # Enables synchronous mode
settings.fixed_delta_seconds = 0.01
```
Moreover, I also set the synchronous mode for the traffic manager:
```
traffic_manager.set_synchronous_mode(True)
```
However, as soon as I run the script the simulator is stuck. I have no clue why. Can you help me out?
Here is the [screen capture](https://drive.google.com/file/d/11VEdA9nogAhMV9GOZtbF3pVWvWYYbyqS/view?usp=sharing), if it helps
And, below is the `startup.py` I ran alongside when the simulator was running:
```
import carla
import argparse
import time
import random

def main():
  argparser = argparse.ArgumentParser(
        description=__doc__)
  argparser.add_argument(
      '--host',
      metavar='H',
      default='127.0.0.1',
      help='IP of the host server (default: 127.0.0.1)')
  argparser.add_argument(
      '-p', '--port',
      metavar='P',
      default=2000,
      type=int,
      help='TCP port to listen to (default: 2000)')
  argparser.add_argument(
      '-n', '--number-of-vehicles',
      metavar='N',
      default=150,
      type=int,
      help='Number of vehicles (default: 150)')
  argparser.add_argument(
      '-w', '--number-of-walkers',
      metavar='W',
      default=50,
      type=int,
      help='Number of walkers (default: 50)')
  argparser.add_argument(
      '--tm-port',
      metavar='P',
      default=8000,
      type=int,
      help='Port to communicate with TM (default: 8000)')
  args = argparser.parse_args()

  try:
    client = carla.Client(args.host, args.port)
    client.set_timeout(10.0) # seconds
    world = client.get_world()

    traffic_manager = client.get_trafficmanager(args.tm_port)
    traffic_manager.set_global_distance_to_leading_vehicle(2.5)
    traffic_manager.set_respawn_dormant_vehicles(True)

    settings = world.get_settings()
    settings.synchronous_mode = True # Enables synchronous mode
    settings.fixed_delta_seconds = 0.01
    world.apply_settings(settings)
    traffic_manager.set_synchronous_mode(True)

    print(f"Spawning {args.number_of_vehicles} in the world.")
    spawn_other_avs(client, args.number_of_vehicles, world, traffic_manager)
  finally:
    print("Programme finished running. Exiting")
    time.sleep(5)

def spawn_other_avs(client, max_vehicles, world, traffic_manager):
    spawn_points = world.get_map().get_spawn_points()

    blueprints = world.get_blueprint_library().filter("vehicle.*")
    blueprints = sorted(blueprints, key=lambda bp: bp.id)

    vehicle_list = []
    batch = []
    for n, transform in enumerate(spawn_points):
        if n >= max_vehicles:
            break
        blueprint = random.choice(blueprints)
        if blueprint.has_attribute("color"):
            color = random.choice(blueprint.get_attribute("color").recommended_values)
            blueprint.set_attribute("color", color)
        if blueprint.has_attribute("driver_id"):
            driver_id = random.choice(
                blueprint.get_attribute("driver_id").recommended_values
            )
            blueprint.set_attribute("driver_id", driver_id)
        try:
            blueprint.set_attribute("role_name", "autopilot")
        except IndexError:
            pass
        vehicle = world.try_spawn_actor(blueprint, transform)
        if vehicle is not None:
            vehicle.set_autopilot(True)
            vehicle_list.append(vehicle)
        else:
            n -= 1
    print(f"successfully spawned {len(vehicle_list)} vehicles")
    return vehicle_list

if __name__ == '__main__':
    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')
```

shh1v, 2022-07-13:
Update: I fixed the simulator being stuck. It was because I was not calling `world.tick()` after spawning the vehicles as recommended [here](https://github.com/carla-simulator/carla/issues/2809). I apologize for not doing my due diligence.

Now, the spawned vehicles do not crash into each other, however, the simulator is really slow now. Is there a way to optimize that? Moreover, #42 still remains a problem.

closed 2022-07-13 by shh1v

GustavoSilvera, 2022-07-14:
Hi, glad to see this is resolved. Make sure the `fixed_delta_seconds` equals the inverse of your desired framerate. In our case we usually leave it as `0.03` to get ~30FPS. 

Other than that, check Task manager to see what the bottleneck might be. If all your CPU cores are at 100% then its likely that you spawned too many vehicle's, similar situation with memory. In this case you might want to spawn fewer vehicles.