HARPLab/DReyeVR #42: Seeting autopilot for spawned vehicles leads to ego vehicle also being on autopilot

shh1v, 2022-07-12:
I have no clue why this is happening. I tried to spawn vehicles on the road and set them on autopilot, however, the ego vehicle is also being set to autopilot. I have no clue why this is happening. To test that there is no problem in my python code, I also tried to run generate_traffic.py, but the same problem is occurring. Here is my repo link: https://github.com/shh1v/DReyeVROFR

GustavoSilvera, 2022-07-13:
Can I ask how exactly you are setting the other vehicles' autopilot? 

We had a similar issue here that [got resolved](https://github.com/HARPLab/DReyeVR/issues/24#issuecomment-1121945221). This was using scenario runner agents. 

In this case you have to make sure to not call `SetAutopilot` to a vehicle that has `"dreyevr"` in the `blueprint.id`

shh1v, 2022-07-13:
While spawning the other vehicles. I made sure that the vehicle with blueprint id "dreyevr" is not spawned (see the `spawn_other_avs() `function). Hence, there are no chances of it being set to autopilot without explicitly setting autopilot to the ego vehicle as shown in `DReyeVR_AI.py".
This is the script I ran. I am not sure if this problem is just occurring for me, and if you would be able to reproduce the error or not.
```
import carla
import argparse
import time
import random

def main():
  argparser = argparse.ArgumentParser(
        description=__doc__)
  argparser.add_argument(
      '--host',
      metavar='H',
      default='127.0.0.1',
      help='IP of the host server (default: 127.0.0.1)')
  argparser.add_argument(
      '-p', '--port',
      metavar='P',
      default=2000,
      type=int,
      help='TCP port to listen to (default: 2000)')
  argparser.add_argument(
      '-n', '--number-of-vehicles',
      metavar='N',
      default=150,
      type=int,
      help='Number of vehicles (default: 150)')
  argparser.add_argument(
      '-w', '--number-of-walkers',
      metavar='W',
      default=50,
      type=int,
      help='Number of walkers (default: 50)')
  argparser.add_argument(
      '--tm-port',
      metavar='P',
      default=8000,
      type=int,
      help='Port to communicate with TM (default: 8000)')
  args = argparser.parse_args()

  try:
    client = carla.Client(args.host, args.port)
    client.set_timeout(10.0) # seconds
    world = client.get_world()

    traffic_manager = client.get_trafficmanager(args.tm_port)
    traffic_manager.set_global_distance_to_leading_vehicle(2.5)
    traffic_manager.set_respawn_dormant_vehicles(True)

    settings = world.get_settings()
    settings.synchronous_mode = True # Enables synchronous mode
    settings.fixed_delta_seconds = 0.01
    world.apply_settings(settings)
    traffic_manager.set_synchronous_mode(True)

    print(f"Spawning {args.number_of_vehicles} in the world.")
    spawn_other_avs(client, args.number_of_vehicles, world, traffic_manager)
  finally:
    print("Programme finished running. Exiting")
    time.sleep(5)

def spawn_other_avs(client, max_vehicles, world, traffic_manager):
    spawn_points = world.get_map().get_spawn_points()

    blueprints = world.get_blueprint_library().filter("vehicle.*")
    blueprints = sorted(blueprints, key=lambda bp: bp.id)

    vehicle_list = []
    batch = []
    for n, transform in enumerate(spawn_points):
        if n >= max_vehicles:
            break
        blueprint = random.choice(blueprints)
        while blueprint.id == 'dreyevr':
            blueprint = random.choice(blueprints)
        if blueprint.has_attribute("color"):
            color = random.choice(blueprint.get_attribute("color").recommended_values)
            blueprint.set_attribute("color", color)
        if blueprint.has_attribute("driver_id"):
            driver_id = random.choice(
                blueprint.get_attribute("driver_id").recommended_values
            )
            blueprint.set_attribute("driver_id", driver_id)
        try:
            blueprint.set_attribute("role_name", "autopilot")
        except IndexError:
            pass
        vehicle = world.try_spawn_actor(blueprint, transform)
        world.tick()
        if vehicle is not None:
            vehicle.set_autopilot(True)
            vehicle_list.append(vehicle)
        else:
            n -= 1
    print(f"successfully spawned {len(vehicle_list)} vehicles")
    return vehicle_list

if __name__ == '__main__':
    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')
```

GustavoSilvera, 2022-07-14:
Hi, try replacing this line: 
```python
if blueprint.has_attribute("driver_id"):
```
With this
```python
if blueprint.has_attribute("driver_id") and "dreyevr" not in blueprint.id:
```

Since when you call `world.get_blueprint_library().filter("vehicle.*")` this will include the singleton EgoVehicle blueprint (in `vehicle.dreyevr.egovehicle` since it is still a carla vehicle. But you don't want to set this vehicle's Autopilot so make sure not to with the check above.

shh1v, 2022-07-14:
Unfortunately, this still does not solve the problem. The ego vehicle is still running in autonomous mode when the script is run.

shh1v, 2022-07-14:
Also, note that the vehicle is running autonomously for some time. Then it stops and then does not move forward. Moreover, I am also not able to control the vehicle once the vehicle stops. So this definitely does not look like setting autopilot is the problem since the autopilot behavior does not continue.

shh1v, 2022-07-14:
Update, I removed the dreyevr blueprint when getting all the blueprints to simplify my debug process. So, I did:

```
blueprints = get_actor_blueprints(world, args.filterv, args.generationv)
blueprints = [blueprint for blueprint in blueprints if "dreyevr" not in blueprint.id]
```
Your initial argument about setting autopilot for "dreyevr" was accurate and that was the reason why the ego vehicle was running autonomously. But, the solution you sent still didn't stop the code from spawning the ego vehicle (at least, that's what I interpreted from the code). Anyways, my issue is fixed now. I really appreciate your help!

GustavoSilvera, 2022-07-14:
Ok, yeah this works to filter out the blueprints. Your solution ensures there is no EgoVehicle bp in the `blueprints` pool to begin with so it only deals with the non-ego vehicles. 

Glad it works!

closed 2022-07-14 by shh1v