HARPLab/DReyeVR #44: make package

FBanani, 2022-07-17:
I want to insert something to the unreal engine (Town04) in Carla, but I work with windows! which is compatible with DReyeVR in windows can I make package?

GustavoSilvera, 2022-07-18:
DReyeVR/Carla is compatible with Windows. I'm not sure what you want to insert and how you want to do it. The default towns (up to Town05) should already be installed when you run `make package`.

FBanani, 2022-07-18:
> DReyeVR/Carla is compatible with Windows. I'm not sure what you want to insert and how you want to do it. The default towns (up to Town05) should already be installed when you run `make package`.

I want to add some signs to town04 and activate this signs in ReyeVR/Carla! this is my issue

FBanani, 2022-07-18:
> DReyeVR/Carla is compatible with Windows. I'm not sure what you want to insert and how you want to do it. The default towns (up to Town05) should already be installed when you run `make package`.

for example: I want to add stop sign here
[image]
then I 
[image]
because I work with windows
[image]
then I select CarlaUE4 folder after that what should I do? that is my question

GustavoSilvera, 2022-07-18:
Also, in case you ever wanted to add custom signs at runtime through the PythonAPI. We have a [tutorial](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Signs.md) just for that!

FBanani, 2022-07-18:
> * add the stop sign (like you already have)
> * save the map (ctrl + s)
> * (sanity check) you should be able to switch to another map and then come back to Town04 and see the stopsign you added in 1)
> * Close down the editor and run `make package` in the `carla` directory (This is Carla's way of packaging the project from the terminal with all the added libraries/assets)

thank you so much

closed 2022-07-18 by GustavoSilvera