HARPLab/DReyeVR #45: -vulkan flag on running DReyeVR/Carla leads to an error

shh1v, 2022-07-18:
Hello! I am trying to change the weather to foggy weather. However, it turns the sky dark. Many previous discussions (like [#3545](https://github.com/carla-simulator/carla/issues/3545), [#5303](#5303)) have highlighted that Vulkan should be used instead on OpenGL. However, 0.9.12+ uses Vulkan by default. So, I was confused why the foggy weather was not rendering. On further investigation, there exists a flag -vulkan which could be used while running Carla. Upon using it, the following error occurred. I am not sure if this is a Carla problem. But, since @GustavoSilvera has been very responsive and helpful (which I really appreciate), I am more inclined to ask here.
Does this error indicate that the game is not using Vulkan? What is the issue here?

[image]

GustavoSilvera, 2022-07-18:
Yes, in order to build with Vulkan enabled, you'll need to edit [this line](https://github.com/HARPLab/DReyeVR/blob/51ea84a65df001801f582051dc3899d996d3f7b9/Configs/DefaultEngine.ini#L108) (on Windows, Linux/MacOS have different lines according to their sections) and change the following:

from this:
```
[/Script/WindowsTargetPlatform.WindowsTargetSettings]
DefaultGraphicsRHI=DefaultGraphicsRHI_DX11
```

to this
```
[/Script/WindowsTargetPlatform.WindowsTargetSettings]
DefaultGraphicsRHI= DefaultGraphicsRHI_Vulkan
```
Then you'll need to rebuild. 

However, I remember trying Vulkan on my local setup before and it demonstrated worse performance and other visual artifacts/glitches in VR so we stuck with DirectX 11 on Windows. Your mileage may vary here depending on GPU. 

I think this can also be done by changing the default graphics settings in the editor: Project Settings -> Platforms -> Windows -> Target RHI -> Default RHI (set it to Vulkan)

shh1v, 2022-07-18:
Making the necessary changes to the DReyeVR/Configs/DefaultEngine.ini file and rebuilding (`make clean && make PythonAPI && make launch && make package`) still throws the same error. Are there any steps I should have executed but missed?

GustavoSilvera, 2022-07-18:
Have you tried both the file change and the UE4 UI change? (Project Settings -> Platforms -> Windows -> Target RHI -> Default RHI (set it to Vulkan))

shh1v, 2022-07-18:
No, I have not tried the UE4 UI change. Let me try rebuilding now. Sorry!

GustavoSilvera, 2022-07-18:
Are you still running with the `-vulkan` flag?

shh1v, 2022-07-18:
With and without `-vulkan` gives the same error.

GustavoSilvera, 2022-07-18:
Yeah try it

shh1v, 2022-07-18:
Checking `Vulkan (SM5)` Worked. Thank you!

shh1v, 2022-07-19:
I know this thread was about being able to use the `-Vulkan flag. However, the main concern was to correctly render the fog which according to previous discussions was only possible using Vulkan. Have you previously tried to render fog in the simulator? If yes, did you have any similar problems as I am having?
This is what after running `python .\environment.py --fog 100` looks like for both `-vulkan` and  `-dx11`:
[image]

shh1v, 2022-07-19:
I am sorry for the bombardment of questions, but I really need your help 😥

GustavoSilvera, 2022-07-19:
As a sanity check, does [dynamic_weather.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/dynamic_weather.py) render fog correctly? I remember using this script to play around with weather (including rain and fog) and it worked without issue. 

If it doesn't then as a last resort I'll point you to the cpp where the [weather data](https://github.com/HARPLab/DReyeVR/blob/main/Carla/Weather/WeatherParameters.h) is held. You'll probably want to change the defaults of this struct so `FogDensity` is larger and the params get applied on `BeginPlay` similar to [this](https://github.com/HARPLab/DReyeVR/blob/2282adc009a97cb9f7590527125b655b0dc8fb5e/Carla/Recorder/CarlaReplayerHelper.cpp#L428-L432) (or you could just set it once in the constructor with hardcoded values, that'd be easier).

shh1v, 2022-07-19:
No, using [dynamic_weather.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/dynamic_weather.py) renders the same dark sky.

I think I will try the latter implementation tomorrow. I need some sleep.

shh1v, 2022-07-19:
Yes, if I press Play in UE4 and run `python .\environment.py --fog 100`, I still get a dark sky.

GustavoSilvera, 2022-07-19:
Sure but what if you change the sky blueprint elements in the editor (no pythonAPI). You should be able to find the Skylight BP in the inspector tab on the right.

GustavoSilvera, 2022-07-19:
Yes, this looks like a problem with UE4 then, because that definitely should work.

GustavoSilvera, 2022-07-19:
I also noticed you're on Windows 11, which is not something I've tested before or even knew DReyeVR/Carla worked on. That *may* be an/the issue but I'm not sure.

GustavoSilvera, 2022-07-19:
Are you still rendering in Vulkan mode or DirectX mode? On NVidia or AMD?

shh1v, 2022-07-19:
Yeah, I am running on Windows 11, Nvidia Geforce RTX 3060. I tried tweaking these parameters in UE4 in bost Vulkan and DirextX.
I think a more likely problem would be that I am using a laptop that runs on hybrid mode (AMD Radeon/RTX 3060). This [reddit post ](https://www.reddit.com/r/pop_os/comments/u7ye9w/vulkan_support_in_integrated_graphics_mode_on/) argues this, but I am not sure how relevant this post is to my problem.'

But, the bigger concern is that you aren't even using Vulkan, and the fog renders for you, which is very strange.

GustavoSilvera, 2022-07-19:
Oh I see the issue, in that same window make sure to enable volumetric fog!

I think this might have been disabled by accident (or for performance) but it prevents fog from being rendered

shh1v, 2022-07-19:
Yes! That fixes it! I can't thank you enough! It would have taken years to figure this out! Thank you!
[image]

closed 2022-07-19 by shh1v