HARPLab/DReyeVR #48: Route Scenario issues

fegonda, 2022-07-21:
Hi,

I recently started using DReyeVR for a behavioral research project and ran into some issues, so some help would really be appreciated.  Note that I have not modified the code, just running the simulator, the scenario runner python script: “python run_experiment.py --title "dreyevr_experiment" --route %SCENARIO_RUNNER_ROOT%\srunner\data\routes_training.xml  %SCENARIO_RUNNER_ROOT%\srunner\data\all_towns_traffic_scenarios1_3_4.json 2 --debug”, and the DReyeVR_AI script.

Issues:
(1) The RouteScenario class specifies 10 supported scenarios  (scenario1,...,scenario10).  Triggered scenarios, however, give inconsistent behavior.  scenario1, which is ControlLoss, does nothing currently.  Scenario3, which is DynamicObjectCrossing, seems to be the only one that consistently triggers correctly, the other scenarios do something inconsistent with the description.

(2) The FollowLeadingVehicleWithObstacle scenario, is currently not listed in the RouteScenario.  I tried to add it to the list of supported scenarios but it does not work.  I assume there are some changes that have to be made either in data or code to get this working.  Can you give me some feedback on what I need to modify to get this working?  

(3) DynamicObjectCrossing currently triggers a pedestrian crossing the street from a soda machine.  According to the code, we should be able to set the adversary_type flag to true to choose a cyclist.  However, the cyclist spawns at the Soda machine and does nothing.  The log shows that the behavior is waiting on some whiteboard variable.  Can you tell me what extra step is needed to switch to the cyclist since setting the adversary_type is not enough?

(4) Changes to DReyeVRConfig.ini are not being read when I relaunch CarlaUE4.exe  Is there something extra needed when changing values in the INI file?

Thanks in advance for your help.
Regards,
Felix

GustavoSilvera, 2022-07-21:
I'm not as familiar with ScenarioRunner as @ajdroid so I will refrain from answering (1)-(3). 

But with regards to (4) I have a couple follow up questions:
- What variables are you trying to change and see affected?
- It sounds like you are running in standalone executable (`make package`) mode (as opposed to editor `make launch` mode). Are you editing the `DReyeVRConfig.ini` file that is local to the executable? (ie. when you run `make package` the `DReyeVRConfig.ini` file is copied over to the executable destination (something like `CarlaUE4/Config/DReyeVRConfig.ini` in the `Build/` directory)
- Are you using the latest DReyeVR (`main`) branch?

fegonda, 2022-07-21:
> I'm not as familiar with ScenarioRunner as @ajdroid so I will refrain from answering (1)-(3).
>
> But with regards to (4) I have a couple follow up questions:
>
> * What variables are you trying to change [...]

- I tried turning off the reticle by setting DrawFlatReticle=False but doesn't work
- I am using the latest version
- I'm running CarlaUE4.exe after make package.  So the DReyeVRConfig.ini is already copied to the exe location as you said, so I assume that making changes there should take automatically take effect.  
- Yes, I have the latest main branch.   It is a powerful system, but the documentation is not complete, so have to look at code and data to figure out things.

GustavoSilvera, 2022-07-21:
Hmm. Yes, you seem to be doing the right thing. Are you still seeing the reticle while in VR mode or while in flat-screen more. The `DrawFlatReticle` flag is specifically for flat-screen (ie. not VR) rendering so it would have no effect on the VR rendering.

fegonda, 2022-07-22:
> Hmm. Yes, you seem to be doing the right thing. Are you still seeing the reticle while in VR mode or while in flat-screen more. The `DrawFlatReticle` flag is specifically for flat-screen (ie. not VR) rendering so it would have no effect on the VR rendering.

Yes, I am currently working on flat-screen to test the scenarios, and I plan to test the scenarios on VR as well.  I want to record a video so that's why I want to be able to turn off the hud elements.   I'll check again my DReyeVRConfig and relaunch

ajdroid, 2022-07-22:
@fegonda can you link the scenario runner commit you are using?

fegonda, 2022-07-22:
> Yes, I am currently working on flat-screen to test the scenarios, and I plan to test the scenarios on VR as well. I want to record a video so that's why I want to be able to turn off the hud [...]

I resolved the config issue.  I was previously changing the DReyeVRConfig under \carla9.12\carla\Unreal\CarlaUE4\Config, but now I am changing the config under WindowsNoEditor\CarlaUE4\Config

fegonda, 2022-07-22:
> 

The version of scenario_runner.py I'm using is "VERSION = '0.9.13'".  It's the Commits on Nov 25, 2021.  I also did a diff of the latest github scenario_runner.py and my local copy and they match.

ajdroid, 2022-07-22:
> (1) The RouteScenario class specifies 10 supported scenarios (scenario1,...,scenario10). Triggered scenarios, however, give inconsistent behavior. scenario1, which is ControlLoss, does nothing [...]

Scenario 1 (Control Loss) should spawn an obstacle or bump that looks like broken tiles in the middle of the road (might not be visible from the 1st person view). For the others, is your vehicle passing through trigger points with the AI navigation? Maybe a video showing this would be helpful.

> (2) The FollowLeadingVehicleWithObstacle scenario, is currently not listed in the RouteScenario. I tried to add it to the list of supported scenarios but it does not work. I assume there are some [...]

Some of the scenarios are deprecated (by the orginial Carla team) and no longer work -- this is usually why they are not listed in the RouteScenario class. Additionally, a scenario will not trigger if not listed in the scenario config `.json` file with appropriate trigger locations/other atributes or if the chosen route (specified with the `.xml`) doesn't pass through/close to the trigger point in the scenario config file.

> (3) DynamicObjectCrossing currently triggers a pedestrian crossing the street from a soda machine. According to the code, we should be able to set the adversary_type flag to true to choose a [...]

In the code I'm looking at in the latest commit, 
https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenarios/object_crash_vehicle.py#L154
 `adversary_type` is not a boolean, but a blueprint filter instead. For a bike, you want to pass in something like `vehicle.diamondback.century` rather than the current `walker.*`. You can find a list of actor blueprints [here](https://carla.readthedocs.io/en/0.9.13/bp_library/)

> I also did a diff of the latest github scenario_runner.py and my local copy and they match.

Can you check if the above `object_crash_vehicle.py` matches the code you're using? If not, please provide a link to the commit you're using.

In general, one thing I would recommend is trying the same scenarios with a vanilla Carla and Scenario Runner install and seeing if the same issues occur. If so, these are not issues specifically with DReyeVR but rather with Scenario Runner. Additionally, Scenario Runner is not a super stable API from the Carla folks and things change all the time. If you plan on using this system, just know that you might need to dive deep into code and tweak stuff to make it work in the way that you need for your research! Hope that was helpful.

fegonda, 2022-07-22:
Thanks for the info.  My object_crash_vehicle matches Commits on Sep 30, 2020.  That seems to be when the adversary_type was changed from a boolean to blueprint filter.  I will try getting the latest commits and test on Monday.  I have a general understanding of Unreal and is comfortable with python, but mainly I think having guidelines on changing a single scenario that touches the different parts of the system can be useful documentation for new people - especially for route-based scenarios.  There's a fair amount of back and forth between data and code to get a good sense of how things work.

fegonda, 2022-07-25:
> Scenario 1 (Control Loss) should spawn an obstacle or bump that looks like broken tiles in the middle of the road (might not be visible from the 1st person view). For the others, is your vehicle [...]

My object crash version is:
https://raw.githubusercontent.com/carla-simulator/scenario_runner/v0.9.13/srunner/scenarios/object_crash_vehicle.py

This follows the installation documentation from your DReyeVR to use -b 0.9.13

If I try the master version of scenario_runner, I get crashes.  So version 0.9.13 is currently the functioning released version, which has adversary_type as a boolean.

FBanani, 2022-08-03:
> Scenario 1 (Control Loss) should spawn an obstacle or bump that looks like broken tiles in the middle of the road (might not be visible from the 1st person view). For the others, is your vehicle [...]

hi I have some stationary objects in the middle of the road and the vehicles that have autopilot mode cross them the "StationaryObjectCrossing(BasicScenario)" class is only for ego vehicle but I need this option for all autopilot vehicles mode!
could you help me?

closed 2022-09-22 by GustavoSilvera