HARPLab/DReyeVR #49: adding vehicle to simulation and set autopilot

FBanani, 2022-07-25:
I do not know why in this code
[image]
most of the car are around the DReyeVr simulator generated
[image]
could you help me ?

GustavoSilvera, 2022-07-25:
Hi, I'm not sure what your question is asking. Could you elaborate?

Also you are definitely spawning a lot of vehicles, since it looks like you are spawning on in every available spawn point as per the line
```python
for i, transform in enumerate(spawn_points): # this loops over all spawn points
  ...
  vehicle = world.try_spawn_actor(vehicle_bp, transform)
```

Note that spawning many vehicles takes a toll on memory and often demands more parallel CPU usage, so expect better performance by using less AI vehicles.

FBanani, 2022-07-25:
thank you and also I have another question vehicles do not follow autopilot however I call world.tick()
what should I do? I have read the autopilot issue but just found to add world.tick() which I added!
could you help me?

GustavoSilvera, 2022-07-26:
What do the other vehicles do if they arent following autopilot? Do they instead just run into walls/curves? Or do they not move at all?

FBanani, 2022-07-26:
They go to walls or move but not in a straight direction something like circle! Therefore I think they do not follow autopilot!

GustavoSilvera, 2022-07-26:
Gotcha. Ok this is a known problem when the server and client framerates desynchronize. It has been discussed in #41. As demonstrated there, you might find some success by setting synchronous mode to true and enabling a fixed deltaT:
```python
settings.synchronous_mode = True # Enables synchronous mode
settings.fixed_delta_seconds = 0.05 # set to the inverse of your desired framerate
traffic_manager.set_synchronous_mode(True)
```

FBanani, 2022-07-26:
thank you
I add settings.synchronous_mode but the simulation stuck
therefore I change if vehicle_bp.has_attribute("driver_id"):
With this" if vehicle_bp.has_attribute("driver_id") and "dreyevr" not in bvehicle_bp.id:
and it is again stuck. this is my coad:
[image]
could you help me?

GustavoSilvera, 2022-07-26:
You'll also need to run the `world.tick()` function in a while loop in order to tick all the other vehicles.

See Shiv's comment in #41

FBanani, 2022-07-26:
thank you. I have done it
[image]
but it stuck in the while loop how can I stop simulation?

GustavoSilvera, 2022-07-26:
You should check out existing scripts like [this one](https://github.com/HARPLab/DReyeVR/blob/main/PythonAPI/DReyeVR_AI.py) which uses a ` try catch` loop to allow you to interrupt the loop without your script crashing.

FBanani, 2022-07-29:
Hi
I add try and finally but the simulation does not stop and finally line does not execute therefore I close Carla's window and execute lines again. in this time it cannot find simulator as you can see in the picture that I attached.
 my main problem is that when the simulator get out of while true loop and execute finally line?
[image]
Could you help me?

GustavoSilvera, 2022-07-29:
Hi, not sure what you mean by the the lines not executing. Do you mean you don't see the print `\ndestroying %d vehicles`? (Also since this is python not C your print should look like `print(f"\ndestroying {len(vehicle_list)} vehicles")`). 

As for the time-out error, this is probably because there is still a lingering instance of the port/socket connected from the simulator in the previous run. So you should make sure to kill it either with `ps -ax | grep carla` on Unix or with `Task Manager -> Details -> CarlaUE4.exe (End Task)` on Windows.

You should be able to check that there is nobody else listening/subscribed to that port (`2000`) before your script runs.

FBanani, 2022-07-29:
thank you so much

closed 2022-07-31 by GustavoSilvera