HARPLab/DReyeVR #50: How to get Logitech steering wheel readings/measurements?

shh1v, 2022-07-26:
I would like to measure the handover performance in SAE level 3 autonomous vehicles when Take-Over Request (TOR) is issued, and measuring this involves measuring the `Standard deviation in steering wheel rotation` and `Maximum Steering Angle velocity`. These metrics require to get the steering wheel readings. I tried to interpret the DReyeVR source code but couldn't get a clear picture of how to achieve this. I would love it if you could answer a few questions:
1. How can I record the steering wheel measurements? Could you please give a high-level idea of the implementation?
2. What would be the units and range of those measurements? (You could just redirect me to the documentation)
3. Can it be done through PythonAPI, or will have to modify the DReyeVR source code?

_Note: I just require the raw readings (like steering angle). I would be able to figure out how to compute the metrics.
Thanks!_

GustavoSilvera, 2022-07-26:
Good question! I assume you already have a Logitech wheel hardware (and the Build.cs flag enabled) so you can already drive the wheel.

The raw steering inputs are processed in cpp in the DReyeVRPawn implementation (see `ADReyeVRPawn::LogitechWheelUpdate()`) which then get sent to the vehicle in the form of high level commands like throttle/brake/turn.

You could record these raw measurements by adding them to the EgoSensor API and streaming then to PythonAPI, but we already have similar data for that in the form of [User inputs](https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/EgoSensor.cpp#L110) (which is the raw inputs after processing, which I think you can invert to get the original raw data back).

Going off memory (but you can verify this with closer inspection of the process function above) the throttle and brake have range of 0 (fully released) to 1 (fully depressed), and the steering wheel has a range from -1 to 1.

Also note there is a bug in the LogitechWheel plugin we use that all the raw values are 0.5 at the start until the input is activated.

Once you have the right data streaming from the EgoSensor to the Pythonapi then you should be able to do everything in python. Else you can add it from cpp.

shh1v, 2022-07-27:
First off, Thank you for the detailed reply.
I encountered a weird error while editing `DReyeVRPawn.h` and `DReyeVRPawn.cpp`. When I edit the `DReyeVRPawn.h` file, and add:
```c++
    ////////////////:DATA_LOGGING:////////////////
    bool bRecordData = false;                                   // To know if data should be recorded/updated
    bool bReactedToTOR = false;                                 // To know if the driver ha reacted to TOR or not
    bool bRecordingOver = false;                                // To know if the recording time is over. Then write data to csv file.
    bool bDataIsWritten = false;                                // We only want to write data once.
    float FutureRecordingTimeStamp = World->GetTimeSeconds();    // If current time >= this variable, then record data
    const float RecordingTimeInterval = 0.2f;                   // in seconds
    int32 ReadSignalFile();
    std::vector<float> SteeringWheelAngles;
    std::vector<float> SteeringWheelVelocity;
    std::vector<float> BrakingInput;
    float ReactionTime = 0.f;
    void WriteData();
```
and then hit compile in UE4, I found that it just throws a "Fatal Error" message and exits, then when I try to relaunch, UE4 throws the same error and exits. This is weird since I am not able to know if it's a compilation error or something else. Have you encountered this problem before?
I have attached the cpp files with full implementation. I would really appreciate it if you could have a look. I know the implementation is sloppy af, but it should work 😂.

[DReyeVRPawn.cpp](https://github.com/HARPLab/DReyeVR/files/9195744/DReyeVRPawn.cpp.txt)
[DReyeVRPawn.h](https://github.com/HARPLab/DReyeVR/files/9195749/DReyeVRPawn.h.txt)

shh1v, 2022-07-27:
Do you think I am getting this behavior because there is a problem with my build? Because I don't know what the problem is.

GustavoSilvera, 2022-07-28:
I will get a chance to look into this tomorrow.

shh1v, 2022-07-28:
Thank you so much!

GustavoSilvera, 2022-07-28:
Ok so without delving into the logic very much, I am getting a segfault as soon as the application opens and it is because of `DReyeVRPawn.h:128`. You are assigning 
```c++
FutureRecordingTimeStamp = World->GetTimeSeconds(); 
```
without a `nullptr` check. In fact I think the world `World` is not even initialized since this assignment code gets run during the construction of `ADReyeVRPawn` where `UWorld *world = nullptr;` is defined on line 63. 

So TL;DR: you are trying to reference a null pointer with `World->GetTimeSeconds();` which is immediately crashing your program. Windows is bad and doesn't give useful stack traces (or any at all!) but I was able to quickly discover this on Linux. 

You should probably just set `FutureRecordingTimeStamp` to `0` or some other constexpr since `GetTimeSeconds()` upon initialization is just 0 anyways ([documentation](https://docs.unrealengine.com/4.26/en-US/API/Runtime/Engine/Engine/UWorld/GetTimeSeconds/)).

shh1v, 2022-07-28:
Wow! Thank you so much.! That fixed it. I really appreciate your help :)

closed 2022-07-28 by shh1v