HARPLab/DReyeVR #51: Eye tracker floating around in the experimenter's view

Shroculus, 2022-07-28:
Hi team,

**CARLA version**: 0.9.13

**Platform/OS:** Windows 10 (Version 21H2)

**Problem you have experienced:** I have installed the CarlaDReyeVR on the computer with below specifications and face issue with the eye tracking. The tracker floats around even when the gaze is fixed.

**Steering Wheels & Pedals:** I have Logitech 920 which is installed and works fine

**SRanipal version:** VIVE_SRanipalInstaller_1.1.0.1

**Steps to reproduce:** In the X64 Native command prompt, path to CarlaDReyeVR\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor and execute CarlaUE4.exe -vr

Below steps have been tried and yet unsuccessful. 

1. I started the application I ran the SRianipal Runtime and it said the runtime is running in the system tray. 
2. The CarlaUE4.log shows it was " Unable to start SRanipal Framework ''. Attaching the log for your reference too. 
3. I have installed VIVE_SRanipalInstaller_1.1.0.1. 
4. USE_SRANIPAL is set to true in CarlaUE4.Build.cs.
5. When I started the DReyeVR application from Carla root folder through make launch and viewed in the VR mode, the eye tracker looked different and it was seen both in the experimenter & participant's view. Attached is a snippet of the same.
6. I downloaded the latest version of CarlaDReyeVR on 26th of June 2022. Were there any commits after that? 

https://user-images.githubusercontent.com/85901551/181648698-2bff0cfc-aa04-4ebb-8a98-d0abf6e7cdf3.mp4

[CarlaUE4.log](https://github.com/HARPLab/DReyeVR/files/9214622/CarlaUE4.log)

Thank you very much in advance.

Best Regards

GustavoSilvera, 2022-07-29:
Ok upon looking at the logs it looks like the SRanipal framework failed to start. 

```
LogSRanipalEye: Error: [SRanipal] Start Eye Framework failed: 1035
LogTemp: Error: Unable to start SRanipal framework!
```

Since it sounds like you installed everything correctly, did you also go through the first-time eye calibration setup?

GustavoSilvera, 2022-07-29:
Also can you post the commit to the version of `DReyeVR` you're using. At time of writing the latest version is `51ea84a65df001801f582051dc3899d996d3f7b9`

GustavoSilvera, 2022-08-25:
Do you have any updates on this? If not I'm going to assume it is resolved.

closed 2022-09-12 by GustavoSilvera