HARPLab/DReyeVR #52: Obtain eye tracking data directly from sensor and not recorded log file during replay

SteadyBits, 2022-07-29:
It seems to me that the eye-tracker information from a user using the VR headset during the replay of a recorded scene is ignored. The eye-tracking data from the original log file produced by the start_recording.py script overrides the eye-tracking info from the user in replay mode (with the start_replaying.py script).
Do you already have a procedure to log the eye-tracking data from a user observing a recorded scene during replay, please? If not, do you have any suggestions (like a hack) other than using a scenario runner?

ajdroid, 2022-07-29:
Hi, just so I understand this question correctly -- are you having someone observe a driving replay in VR? Replays were designed to be viewed by the experimenter on a flat screen rather than by a human subject in VR and so some things may break if you try to do this (but it should be possible by listening to SRanipal instead of going through the replay flags -- this behaviour is untested and we can provide some more details on how to do this shortly).

SteadyBits, 2022-07-29:
Yes, that's what I meant. However, this issue has now been resolved.

Fix:
1. Remove the check that disables the sensor update when replay is enabled. This is found here: (https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/EgoSensor.cpp#L97). In my case, I simply replaced "if (!bIsReplaying)" with "if (true)"
2. Comment these lines: https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/EgoVehicle.cpp#L259-L260
     this prevents the camera from being updated with the recorded head position.
Thanks to @GustavoSilvera and @ajdroid

closed 2022-08-01 by ajdroid