HARPLab/DReyeVR #53: autopilot cars collide with the stuff in a road

FBanani, 2022-08-02:
Hi Gustavo this is me again :)
when I set autopilot for the vehicles not DReyeVR vehicle and put some staff on the road vehicles collide to this staff. I do not find any comment about preventing collision for autopilot mode. could you guide me what should I do?

GustavoSilvera, 2022-08-02:
Hi, can you explain further what the non-ego autopilot behaviour looks like? Are they crashing into pedestrians who are jaywalking or standing in the middle of the road? Or are they crashing into walls/each other as seen in #41 ?

FBanani, 2022-08-02:
I used generate_traffic.py and change the code in some lines therefore vehicles not crash to each other or walls, and this is the code. 
[adding autopilot vehicles.txt](https://github.com/HARPLab/DReyeVR/files/9243930/adding.autopilot.vehicles.txt)
but I provided the obstacle in the road which is not a car and I do not know how improve the autopilot mode for non-ego vehicles to avoid this obstacle. could you guide me because every thing that I found is for def collision_and_car_avoid_manager(self, waypoint) not for a fixed obstacle?

GustavoSilvera, 2022-08-02:
This sounds like more of a Carla issue than a DReyeVR issue. 

I'm not sure the vehicle's in carla will automatically stop for anything in their path, since they are blind to all things that have not been registered in the [Carla actor registry](http://carla.org/Doxygen/html/df/db5/classFActorRegistry.html). We get around this by [registering our (DReyeVR) EgoVehicle with the CarlaRegistry](https://github.com/HARPLab/DReyeVR/blob/51ea84a65df001801f582051dc3899d996d3f7b9/DReyeVR/EgoVehicle.cpp#L622-L643). Which might be something you're interested in. 

Another simple option is to use the [line-trace functions provided in the AI controller to detect](https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp#L44-L72) when there is something in front of the vehicle and apply the brakes (but this is not "smart" and would not be able to re-route around the obstacle).

FBanani, 2022-08-02:
thank you so much

closed 2022-08-11 by GustavoSilvera

wx983, 2023-02-27:
@GustavoSilvera 
In addition, when I switch to the manual drving through pressing `1` in the keyboard, it can't back up in the `R` mode. Can you fix it?
[image]

GustavoSilvera, 2023-02-27:
@wx983 For your first point, I haven't seen this behavior but I also haven't tested the built-in traffic manager autopilot feature very heavily yet, so there could be some bugs in CARLA. Does this behavior also happen for other vehicles on autopilot that are not the EgoVehicle?

For your second point, this sounds like a DReyeVR bug and I'll look into it. But how are you switching gears to reverse? Are you using the keyboard input (`z` or `LAlt`) or using the logitech (face buttons) wheel?

wx983, 2023-02-27:
> @wx983 For your first point, I haven't seen this behavior but I also haven't tested the built-in traffic manager autopilot feature very heavily yet, so there could be some bugs in CARLA. Does this [...]

First point, other vehicles on autopilot that are not the EgoVehicle are always able to pass intersections normally. This issue(can not pass intersection) happens for EgoVehicle occassionally. As things should be, both EgoVehicle and other vehicles are `autopilot`. They should behave the same. That's really strange. Next I will also look into it.

Second point, I use Logitech wheel to control EgoVehicle and I use them to reverse gear.
[image]

GustavoSilvera, 2023-02-27:
Okay interesting, I will definitely look into the reverse bug. Thanks for letting me know!