HARPLab/DReyeVR #56: Autopilot not functioning after Logitech steering wheel input is given

shh1v, 2022-08-10:
When I launch the simulator and turn on autopilot for DReyeR_vehicle using PythonAPI, then:
1. When I give a WSAD input momentarily, it does get simulated and the vehicle moves, then when the keys are released, the autopilot repositions the car to the correct lane. This behavior is accurate and expected.
2. When the Logitech steering wheel and pedal are connected before the launch, and then the simulator is launched, either (1) autopilot does not work correctly (the vehicle has 0-1 km/her speed), or (2) The autopilot works correctly initially but when Logitech input is given, unusual autopilot behavior is observed.
I have no clue why this is happening. I would really appreciate any input (pun intended) you have. Thank you.
Note that, I did modify DReyeVRPawn.cpp/.h but, I also tried the original codes, and they also seem to have the same problem.
Please let me know if you need any further information. Thank you/

GustavoSilvera, 2022-08-10:
This sounds to me like a case of input conflict between the WASD/Logitech/Autopilot controls. The basic logic is as follows:
(Assuming the autopilot is on)
- If the user sends input (either through WASD keyboard or logi wheel) then the autopilot is temporarily disabled in favour of the new inputs
- Once the user inputs are released the autopilot regains control (this is a "handoff back to AI") as you witnessed
- The behaviour (1) you see might be a problem with the LogitechPluginLibrary we are using, which (as stated [here](https://github.com/drb1992/LogitechWheelPlugin)) sets all the analog inputs (eg. pedals) to 0.5 until something is pressed. So can you debug print these messages to verify what the logitech output is?
- I'm not understanding why behaviour (2) is incorrect? It should perform the same temporary takeover as WASD would which is what you are observing?

shh1v, 2022-08-10:
I think I didn't do a great job expressing the problem. Here is a [video presentation](https://youtu.be/yejnwaU92x0) of what is happening.
This is the python script I ran:
```python
#!/usr/bin/env python

###############################################
# Copyright (c) Shiv Patel, 2022
# BC, Canada.
# The University of British Columbia, Okanagan
###############################################

import threading
import time
import os
import sys
import glob

try:
    sys.path.append(glob.glob('..\\carla\\dist\\carla-0.9.13-py*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla
from numpy import random
from typing import Any, Dict, List, Optional

def wait(world, duration):
  target_time = time.time() + duration
  while time.time() < target_time:
    world.tick()

def find_ego_vehicle(world: carla.libcarla.World) -> Optional[carla.libcarla.Vehicle]:
    DReyeVR_vehicle = None
    ego_vehicles = world.get_actors().filter("vehicle.dreyevr.egovehicle")
    try:
        DReyeVR_vehicle = ego_vehicles[0]  # TODO: support for multiple ego vehicles?
    except IndexError:
        print("Unable to find DReyeVR ego vehicle in world!")
    return DReyeVR_vehicle

client = carla.Client('127.0.0.1', 2000)
client.set_timeout(10.0)
world = client.get_world()
world.set_weather(carla.WeatherParameters.MidRainyNoon)

traffic_manager = client.get_trafficmanager()
traffic_manager.set_global_distance_to_leading_vehicle(2)
settings = world.get_settings()
traffic_manager.set_synchronous_mode(True)
if not settings.synchronous_mode:   
    settings.synchronous_mode = True
    settings.fixed_delta_seconds = 1.0/80
world.tick()
DReyeVR_vehicle = find_ego_vehicle(world)
DReyeVR_vehicle.set_autopilot(True, traffic_manager.get_port())
wait(world, 30)
```
You can see in the video that when Logitech input is given (you can observe the tilt in the steering wheel - that is when the input is given), the autopilot is disabled. Then when the steering wheel is released and it comes back to its original position (i.e., no input), the autopilot still does not turn back on.

GustavoSilvera, 2022-08-10:
Ah ok I see, so you are manually moving the wheel back to its resting position?

IIRC there is not a big dead-zone for the wheel so it might still be giving output of ~0.01 or something small but non-zero. This still satisfies the "user is giving input" clause from above so the autopilot doesn't turn back on. 

This is technically working as expected but I agree we should probably introduce a better way to trigger a manual autopilot handoff where it will simply ignore the user commands until say a button is pressed that switches back and forth between autopilot & manual. 

To verify this is what is going on can you log out some of the logitech wheel inputs in the tick function?

shh1v, 2022-08-10:
When I release the steering wheel, it comes back to its normal/resting position (by the centering spring feature).  I was debugging and printing the steering wheel angles and you are right. The rotated steering wheel comes back to its original position, but not precisely, its ±1-2 degrees. So, the game thinks that the user is still giving input. Some of the temporary ideas I think is to assume that no input is given unless it is greater than ±2 degrees, OR implement something to make the steering wheel come back precisely to 0 degrees. What do you think?

shh1v, 2022-08-10:
Okay, so in intrest of my study, I changed the code implementation to such that, no logitech inputs should be taken when autopilot is on. So, I modified the `void ADReyeVRPawn::LogitechWheelUpdate()` method:
```cpp
        /// NOTE: directly calling the EgoVehicle functions
        bool isSignificantInput = FMath::IsNearlyEqual(WheelRotation, WheelRotationLast, LogiThresh) &&
            FMath::IsNearlyEqual(AccelerationPedal, AccelerationPedalLast, LogiThresh) &&
            FMath::IsNearlyEqual(BrakePedal, BrakePedalLast, LogiThresh);
        // Disabled input from Logitech when autopilot is on
        if (EgoVehicle->GetAutopilotStatus() /* && isSignificantInput*/)
        {
            // let the autopilot drive if the user if autopilot is on
        }
```
But, it is not working. I turned the autopilot on using python API and then titled the steering wheel, and the autopilot should NOT turn off, but that's not the case. The autopilot gets disabled.

I am not sure why. I really would appreciate your help

shh1v, 2022-08-10:
Upon further debugging, I noticed that `EgoVehicle->GetAutopilotStatus()` is false even when autopilot is on.
```cpp
void ADReyeVRPawn::LogitechWheelUpdate()
{
    check(EgoVehicle);
    ensure(bOverrideInputsWithKbd == false); // kbd inputs should be false

    // only execute this in Windows, the Logitech plugin is incompatible with Linux
    if (LogiUpdate() == false) // update the logitech wheel
        UE_LOG(LogTemp, Warning, TEXT("Logitech wheel %d failed to update!"), WheelDeviceIdx);
    DIJOYSTATE2* WheelState = LogiGetState(WheelDeviceIdx);
    if (bLogLogitechWheel)
        LogLogitechPluginStruct(WheelState);
    /// NOTE: obtained these from LogitechWheelInputDevice.cpp:~111
    // -32768 to 32767. -32768 = all the way to the left. 32767 = all the way to the right.
    const float WheelRotation = FMath::Clamp(float(WheelState->lX), -32767.0f, 32767.0f) / 32767.0f; // (-1, 1)
    const float WheelRotationDegree = (FMath::Clamp(float(WheelState->lX), -32767.0f, 32767.0f) / 32767.0f) * 900;
    // -32768 to 32767. 32767 = pedal not pressed. -32768 = pedal fully pressed.
    const float AccelerationPedal = fabs(((WheelState->lY - 32767.0f) / (65535.0f))); // (0, 1)
    // -32768 to 32767. Higher value = less pressure on brake pedal
    const float BrakePedal = fabs(((WheelState->lRz - 32767.0f) / (65535.0f))); // (0, 1)
    // -1 = not pressed. 0 = Top. 0.25 = Right. 0.5 = Bottom. 0.75 = Left.
    const float Dpad = fabs(((WheelState->rgdwPOV[0] - 32767.0f) / (65535.0f)));

    const float LogiThresh = 0.01f; // threshold for analog input "equality"

    // weird behaviour: "Pedals will output a value of 0.5 until the wheel/pedals receive any kind of input"
    // as per https://github.com/HARPLab/LogitechWheelPlugin
    if (bPedalsDefaulting)
    {
        // this bPedalsDefaulting flag is initially set to not send inputs when the pedals are "defaulting", once the
        // pedals/wheel is used (pressed/turned) once then this flag is ignored (false) for the remainder of the game
        if (!FMath::IsNearlyEqual(WheelRotation, 0.f, LogiThresh) ||      // wheel is not at 0 (rest)
            !FMath::IsNearlyEqual(AccelerationPedal, 0.5f, LogiThresh) || // accel pedal is pressed
            !FMath::IsNearlyEqual(BrakePedal, 0.5f, LogiThresh))          // brake pedal is pressed
        {
            bPedalsDefaulting = false;
        }
        // EDITED Debug the wheel rotation
        const FString DebugString = FString::Printf(TEXT("Pedals defaulting; %s"), EgoVehicle->GetAutopilotStatus() ? TEXT("true") : TEXT("false"));
        const bool PrintToLog = false; // kinda annoying when flooding the logs with warning messages
        const bool PrintToScreen = true;
        const float ScreenDurationSec = 5.f;
        const FLinearColor MsgColour = FLinearColor(1, 0, 0, 1); // RED
        UKismetSystemLibrary::PrintString(World, DebugString, PrintToScreen, PrintToLog, MsgColour, ScreenDurationSec);
    }
    else
    {
        /// NOTE: directly calling the EgoVehicle functions
        bool isSignificantInput = FMath::IsNearlyEqual(WheelRotation, WheelRotationLast, LogiThresh) &&
            FMath::IsNearlyEqual(AccelerationPedal, AccelerationPedalLast, LogiThresh) &&
            FMath::IsNearlyEqual(BrakePedal, BrakePedalLast, LogiThresh);
        // Disabled input from Logitech when autopilot is on
        if (EgoVehicle->GetAutopilotStatus() /* && isSignificantInput*/)
        {
            // let the autopilot drive if the user is not putting significant inputs
            // EDITED Debug the wheel rotation
            const FString DebugString = FString::Printf(TEXT("Wheel rotation: %f; autopilot ON %s"), WheelRotation, (EgoVehicle->GetAutopilotStatus() ? TEXT("true") : TEXT("false")));
            const bool PrintToLog = false; // kinda annoying when flooding the logs with warning messages
            const bool PrintToScreen = true;
            const float ScreenDurationSec = 5.f;
            const FLinearColor MsgColour = FLinearColor(1, 0, 0, 1); // RED
            UKismetSystemLibrary::PrintString(World, DebugString, PrintToScreen, PrintToLog, MsgColour, ScreenDurationSec);
        }
        else
        {
            // EDITED Debug the wheel rotation
            const FString DebugString = FString::Printf(TEXT("Wheel rotation: %f; autopilot OFF %s"), WheelRotation, (EgoVehicle->GetAutopilotStatus() ? TEXT("true") : TEXT("false")));
            const bool PrintToLog = false; // kinda annoying when flooding the logs with warning messages
            const bool PrintToScreen = true;
            const float ScreenDurationSec = 5.f;
            const FLinearColor MsgColour = FLinearColor(1, 0, 0, 1); // RED
            UKismetSystemLibrary::PrintString(World, DebugString, PrintToScreen, PrintToLog, MsgColour, ScreenDurationSec);

            // take over the vehicle control completely
            EgoVehicle->SetSteering(WheelRotation);
            EgoVehicle->SetThrottle(AccelerationPedal);
            EgoVehicle->SetBrake(BrakePedal);
            ......
        }
```
It prints "Pedals Defaulting; false" even when autopilot is turned on. That's why my code was not working.

shh1v, 2022-08-10:
TLDR; all I want is to ignore the inputs of Logitech when autopilot is on.

GustavoSilvera, 2022-08-10:
Ah ok your code does look good and I think I know the problem. 

The EgoVehicle->GetAutopilotStatus() is only getting the local (EgoVehicle) variable bIsAutopilotEnabled which does not get updated alongside the PythonAPI automatically (only when manually triggered through `EgoVehicle::SetAutopilot`). So you're really looking for [this flag](https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.h#L133-L136) in the `WheeledVehicleAIController::IsAutopilotEnabled()`. 

A robust way to fix this which I will add in the next release will be to update the local `EgoVehicle:: bAutopilotEnabled` on every tick as follows:

```c++
// in ego vehicle tick()
bAutopilotEnabled = AI_Player->IsAutopilotEnabled();
```

shh1v, 2022-08-10:
This fixed the problem. Thank you so much!

closed 2022-08-10 by shh1v