HARPLab/DReyeVR #57: error while running "make launch"

zoughi, 2022-08-18:
Hi,
I follow this link step by step (part: "**DReyeVR installation command summary**") to install DReyeVR :
https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md

I done all step correctly up to install carla. In order to install "carla" when I run "**make launch**" in "x64 Native Tools Command Prompt for VS 2019" I get the following error. I would be appreciate if you help me to solve this error:

Building 8 actions with 8 processes...
  [1/8] Module.Carla.gen.5_of_7.cpp
  D:\VR\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/TaggedComponent.h(83): error C2504: 'FHierarchicalStaticMeshSceneProxy': base class undefined
  D:\VR\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/TaggedComponent.h(87): error C3668: 'FTaggedHierarchicalStaticMeshSceneProxy::GetViewRelevance': method with override specifier 'override' did not override any base class methods
  [2/8] Module.Carla.3_of_3.cpp
  D:/VR/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(540): error C2039: 'StopTelemetry': is not a member of 'UWheeledVehicleMovementComponent'
  D:\VR\UE\UnrealEngine-4.27\Engine\Plugins\Runtime\PhysXVehicles\Source\PhysXVehicles\Public\PhysXVehicleManager.h(11): note: see declaration of 'UWheeledVehicleMovementComponent'
  D:/VR/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(574): error C2039: 'SetWheelRotYaw': is not a member of 'UVehicleAnimInstance'
  D:\VR\UE\UnrealEngine-4.27\Engine\Plugins\Runtime\PhysXVehicles\Source\PhysXVehicles\Public\VehicleAnimInstance.h(61): note: see declaration of 'UVehicleAnimInstance'
  D:/VR/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(596): error C2039: 'GetWheelRotAngle': is not a member of 'UVehicleAnimInstance'
  D:\VR\UE\UnrealEngine-4.27\Engine\Plugins\Runtime\PhysXVehicles\Source\PhysXVehicles\Public\VehicleAnimInstance.h(61): note: see declaration of 'UVehicleAnimInstance'
  D:/VR/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(614): warning C4458: declaration of 'RootComponent' hides class member
  D:\VR\UE\UnrealEngine-4.27\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(672): note: see declaration of 'AActor::RootComponent'
 ...

GustavoSilvera, 2022-08-18:
Hi there, by looking at these errors my best guess is that you are using UE4-vanilla (ie. installed unreal-engine from source via the Epic Games github). However, building Carla 0.9.12+ requires Carla's own fork of UE4 in order to build (they added several things). For example the `StopTelemetry` function was first introduced to `WheeledVehicleMovementComponent` in this [commit](https://github.com/CarlaUnreal/UnrealEngine/commit/08f1400f3c31d9c00ff57d14793566edebd00415) which is on the Carla fork. So you should try reinstalling UE4 via the `github.com/CarlaUnreal/UnrealEngine` fork and then `make launch` again. 

For more notes check out:
```
# Linux: https://carla.readthedocs.io/en/0.9.13/build_linux/#unreal-engine
# Windows: https://carla.readthedocs.io/en/0.9.13/build_windows/#unreal-engine
```

zoughi, 2022-08-19:
hi dear Gustavo,
thank you for your quick response
OK, I will do that and I will report the results here.

zoughi, 2022-09-05:
hi dear Gustavo,
this problem was solved by using Carla's own fork of UE4.

best regards

closed 2022-09-05 by zoughi