HARPLab/DReyeVR #58: make install CARLA=../carla SR=../scenario_runner" errors

zoughi, 2022-09-05:
Hi 
I follow the steps in Docs/Install.md and install all of them correctly up to "Installing DReyeVR". I have execute the following command which encounter error:
make install CARLA=../carla SR=../scenario_runner
```
  File "Scripts/install.py", line 22
    DOC_STRING: str = "Install DReyeVR atop a working Carla and/or ScenarioRunner directory"
              ^
SyntaxError: invalid syntax
make: *** [install] Error 1
```
I run this command with the following python versions but still the same error remain:
python2.7,3.6
python3.8,3.9
I would be appreciate if someone help me to solve this error.

GustavoSilvera, 2022-09-05:
Hi there, this is almost certainly a `python2` vs `python3` problem since that syntax is Python3-only. 

Curious, are you building on Windows? 

The makefile (`make`) command defaults to the system `python` which might still be `python2.7` since we actually assume you only have python3 installed (since python2 is deprecated). But you should be able to either make a symlink from python2 -> 3 or force it to use python3. 

To see whether this is the problem, in your terminal (or command prompt) type:
```bash
python
```

and see whether this prompt (no suffix after `python`) is python 2 or 3. We require that this command invokes python3.

zoughi, 2022-09-05:
thank you for your response
I am building on Windows 10 and I execute this command on "WSL"

GustavoSilvera, 2022-09-05:
Ok that should be fine, note that you'll still need to run `make launch` and `make PythonAPI` and `make package` on the x64 command prompt though. But in WSL you should be able to run `make install`. What does the `python` command say your default python instance is?

zoughi, 2022-09-06:
Dear Gustavo, Many thanks for your assistance.
I works by uninstalling python2.7 and link python to python3 using the following commands:
```
sudo apt purge -y python2.7-minimal
sudo ln -s /usr/bin/python3 /usr/bin/python
```

closed 2022-09-06 by zoughi