HARPLab/DReyeVR #59: error while running make package command

zoughi, 2022-09-12:
hi, dear Gustavo
I follow the steps in Docs/Install.md and install all of them correctly up to "build everything". 
carla: 0.913
UE: 4.27
windows: 10
vs: 2019
I have execute the following command which encounter error:
```
make package    # create an optimized package
```
********** BUILD COMMAND COMPLETED **********
********** COOK COMMAND STARTED **********
Running UE4Editor Cook for project D:\VR\carla\Unreal\CarlaUE4\CarlaUE4.uproject
Commandlet log file is D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Cook-2022.09.12-07.28.18.txt
Running: D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Binaries\Win64\UE4Editor-Cmd.exe D:\VR\carla\Unreal\CarlaUE4\CarlaUE4.uproject -run=Cook  -TargetPlatform=WindowsNoEditor -fileopenlog -unversioned -abslog=D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Cook-2022.09.12-07.28.18.txt -stdout -CrashForUAT -unattended -NoLogTimes
  LogInit: Display: Running engine for game: CarlaUE4
  LogWindows: Error: begin: stack for UAT
  LogWindows: Error: === Critical error: ===
  LogWindows: Error:
  LogWindows: Error: Fatal error: [File:D:/VR/UnrealEngine-carla/UnrealEngine-carla/Engine/Source/Runtime/Core/Private/Internationalization/ICUInternationalization.cpp] [Line: 153]
  LogWindows: Error: ICU data directory was not discovered:
  LogWindows: Error: ../../../../../carla/Unreal/CarlaUE4/Content/Internationalization
  LogWindows: Error: ../../../Engine/Content/Internationalization
  LogWindows: Error:
  LogWindows: Error:
  LogWindows: Error: [Callstack] 0x00007ffcde174fd9 KERNELBASE.dll!UnknownFunction []
  LogWindows: Error: [Callstack] 0x00007ffc99e9e966 UE4Editor-Core.dll!ReportAssert() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Core\Private\Windows\WindowsPlatformCrashContext.cpp:1616]
  LogWindows: Error: [Callstack] 0x00007ffc99ea1ee9 UE4Editor-Core.dll!FWindowsErrorOutputDevice::Serialize() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Core\Private\Windows\WindowsErrorOutputDevice.cpp:79]
  LogWindows: Error: [Callstack] 0x00007ffc99bc250d UE4Editor-Core.dll!FOutputDevice::LogfImpl() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Core\Private\Misc\OutputDevice.cpp:61]
  LogWindows: Error: [Callstack] 0x00007ffc99a8114a UE4Editor-Core.dll!FICUInternationalization::Initialize() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Core\Private\Internationalization\ICUInternationalization.cpp:153]
  LogWindows: Error: [Callstack] 0x00007ffc99a714ba UE4Editor-Core.dll!FInternationalization::Get() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Core\Private\Internationalization\Internationalization.cpp:24]
  LogWindows: Error: [Callstack] 0x00007ffc99abcc59 UE4Editor-Core.dll!BeginInitTextLocalization() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Core\Private\Internationalization\TextLocalizationManager.cpp:335]
  LogWindows: Error: [Callstack] 0x00007ff676467332 UE4Editor-Cmd.exe!FEngineLoop::AppInit() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:5283]
  LogWindows: Error: [Callstack] 0x00007ff67647e77f UE4Editor-Cmd.exe!FEngineLoop::PreInitPreStartupScreen() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:2180]
  LogWindows: Error: [Callstack] 0x00007ff676471757 UE4Editor-Cmd.exe!GuardedMain() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Launch\Private\Launch.cpp:127]
  LogWindows: Error: [Callstack] 0x00007ff676471aea UE4Editor-Cmd.exe!GuardedMainWrapper() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:137]
  LogWindows: Error: [Callstack] 0x00007ff676485a9d UE4Editor-Cmd.exe!WinMain() [D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:268]
  LogWindows: Error: [Callstack] 0x00007ff676487abe UE4Editor-Cmd.exe!__scrt_common_main_seh() [d:\agent\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
  LogWindows: Error: [Callstack] 0x00007ffce00b7034 KERNEL32.DLL!UnknownFunction []
  LogWindows: Error: [Callstack] 0x00007ffce0482651 ntdll.dll!UnknownFunction []
  LogWindows: Error:
  LogWindows: Error: end: stack for UAT
Took 135.635193s to run UE4Editor-Cmd.exe, ExitCode=3
ERROR: Cook failed.
       (see D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Logs\Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
BUILD FAILED

-[Package]: [ERROR] There was a problem while packaging Unreal project.
          [ERROR] Please read the screen log for more information.
make: *** [package] Error 1

I make "clean carla" and reinstall it for three times but the error still remains. I would be appreciate if you help me to solve this problem.

GustavoSilvera, 2022-09-12:
Hi,

So does `make PythonAPI` and `make carla` work? It's only `make package` that fails like this?

zoughi, 2022-09-12:
yes, make PythonAPI and make carla worked.  The command "make package" causes this error.

GustavoSilvera, 2022-09-12:
Hm I haven't seen this error before but it looks like a [relatively common problem](https://forums.unrealengine.com/t/unreal-4-16-1-debug-game-build-error-icu-data-directory-was-not-discovered/395338) in older UE versions. Have you tried looking at the solutions that other folks mentioned?

You should make sure that your `UE4_ROOT` matches with your installation of UnrealEngine-carla (carla fork).

One thing I might recommend (since you already cleaned carla with `make clean`) is to also force a rebuild of the engine by deleting the `Saved` and `Intermediate` folders in your `UnrealEngine-carla` directory.

GustavoSilvera, 2022-09-22:
Do you have any more updates to this issue?

zoughi, 2022-09-25:
Thank you I really appreciate your help.
that error was solved. but the other error appeared when running "make package" as follows:
```
********** BUILD COMMAND COMPLETED **********
********** COOK COMMAND STARTED **********
Running UE4Editor Cook for project D:\VR\carla\Unreal\CarlaUE4\CarlaUE4.uproject
Commandlet log file is D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Cook-2022.09.24-11.53.25.txt
Running: D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Binaries\Win64\UE4Editor-Cmd.exe D:\VR\carla\Unreal\CarlaUE4\CarlaUE4.uproject -run=Cook  -TargetPlatform=WindowsNoEditor -fileopenlog -unversioned -abslog=D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Cook-2022.09.24-11.53.25.txt -stdout -CrashForUAT -unattended -NoLogTimes
  LogInit: Display: Running engine for game: CarlaUE4
  RenderDocPlugin: Display: Unable to initialize the plugin because no RenderDoc libray has been located.
  LogLuminAPIImpl: Warning: MLSDK not found.  This likely means the MLSDK environment variable is not set.
  LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
  LogModuleManager: Warning: ModuleManager: Unable to load module 'ExampleDeviceProfileSelector'  - 0 instances of that module name found.
  LogHAL: Display: Platform has ~ 8 GB [8468381696 / 8589934592 / 8], which maps to Default [LargestMinGB=32, LargerMinGB=12, DefaultMinGB=8, SmallerMinGB=6, SmallestMinGB=0)
  LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'AllDesktop'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ASTC'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_DXT'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ETC2'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'AndroidClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ASTCClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_DXTClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ETC2Client'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_Multi'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_MultiClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'IOSClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'IOS'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Lumin'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'LuminClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'MacNoEditor'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Mac'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'MacClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'MacServer'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'TVOSClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'TVOS'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'WindowsNoEditor'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'Windows'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'WindowsClient'
  LogTargetPlatformManager: Display: Loaded TargetPlatform 'WindowsServer'
  LogTargetPlatformManager: Display: Building Assets For WindowsNoEditor
  LogAudioDebug: Display: Lib vorbis DLL was dynamically loaded.
  LogShaderCompilers: Display: Using Local Shader Compiler.
  LogDerivedDataCache: Display: Max Cache Size: 512 MB
  LogDerivedDataCache: Display: Loaded Boot cache: ../../../../../carla/Unreal/CarlaUE4/DerivedDataCache/Boot.ddc
  LogDerivedDataCache: Display: Performance to ../../../Engine/DerivedDataCache: Latency=3.19ms. RandomReadSpeed=3.48MBs, RandomWriteSpeed=214.11MBs. Assigned SpeedClass 'Fast'
  LogModuleManager: Warning: ModuleManager: Unable to load module 'ExampleDeviceProfileSelector'  - 0 instances of that module name found.
  LogMaterial: Display: Missing cached shader map for material Widget3DPassThrough, compiling.
  LogShaderCompilers: Display: Worker (1/5): shaders left to compile 10
  LogShaderCompilers: Display: Worker (2/5): shaders left to compile 7
  LogShaderCompilers: Display: Worker (3/5): shaders left to compile 4
  LogShaderCompilers: Display: Worker (4/5): shaders left to compile 1
  LogMaterial: Display: Missing cached shader map for material Widget3DPassThrough, compiling.
  LogMaterial: Display: Missing cached shader map for material Widget3DPassThrough, compiling.
  LogShaderCompilers: Display: Worker (5/5): shaders left to compile 12
  LogMaterial: Display: Missing cached shader map for material Widget3DPassThrough, compiling.
  LogMaterial: Display: Missing cached shader map for material Widget3DPassThrough, compiling.
  LogMaterial: Display: Missing cached shader map for material Widget3DPassThrough, compiling.
  LogShaderCompilers: Display: Worker (4/5): shaders left to compile 43
  LogShaderCompilers: Display: Worker (2/5): shaders left to compile 34
  LogShaderCompilers: Display: Worker (3/5): shaders left to compile 27
[... cut ...]
Took 186.5346604s to run UE4Editor-Cmd.exe, ExitCode=3
ERROR: Cook failed.
       (see D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Logs\Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
BUILD FAILED

-[Package]: [ERROR] There was a problem while packaging Unreal project.
          [ERROR] Please read the screen log for more information.
make: *** [package] Error 1
```

but i I don't know where is the problem?

GustavoSilvera, 2022-09-25:
Huh. This is a strange problem that I haven't seen before. Looks like there is something wrong with your UE4 shader compilation. Can you send the full log that the `Cook failed` mentions?

zoughi, 2022-09-26:
yes of course. I put the log files in the following links:

https://drive.google.com/file/d/15oWkmzOAcbN2csdDoqQstPyIb3huBk2V/view?usp=sharing
https://drive.google.com/file/d/1AKALOc02H-zQ3xDdNEFC3JHCRUVfd_Jl/view?usp=sharing

GustavoSilvera, 2022-09-26:
Oh interesting, yeah we only officially support [CarlaUnreal 4.26](https://github.com/carlaunreal/unrealengine) since that is required by them for the 0.9.13 fork. I'm very surprised you were able to build Carla/DReyeVR with a vanilla 4.27 UnrealEngine since they made some code changes to the engine that are required to build LibCarla. 

Can you instead use [Carla's fork for UnrealEngine](https://github.com/carlaunreal/unrealengine) on branch `carla`

zoughi, 2022-09-27:
thank you, I will do that.
I have another question, I have an "HTC Vive" headset which is not pro ("HTC Vive pro"), is it compatible with [DReyeVR]?

GustavoSilvera, 2022-09-27:
Hi, I think that UnrealEngine version is OK, but did you download it from CARLA's fork github repo or from the UnrealEngine github organization? (You need to download it from CARLA's org)

Yes any SteamVR-compatible headset should work, but you'll want to disable the eye tracking (which requires special hardware/firmware) as mentioned [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#building-dreyevr).

zoughi, 2022-09-27:
I downloaded it from CARLA's fork.

GustavoSilvera, 2022-09-27:
Okay, then that should work. 

Taking a closer look at your log.txt file I found that the first errors occur here:
```
ProcessResult.StdOut:   LogBlueprint: Error: [AssetLog] D:\VR\carla\Unreal\CarlaUE4\Content\Carla\Static\TrafficSign\PostSigns\Round\SpeedLimiters\BP_SpeedLimit.uasset: [Compiler] Can't connect pins  Object  and  Object : This cast has an invalid target type (was the class deleted without a redirect?). from Source: /Game/Carla/Static/TrafficSign/PostSigns/Round/SpeedLimiters/BP_SpeedLimit.BP_SpeedLimit
ProcessResult.StdOut:   LogBlueprint: Error: [AssetLog] D:\VR\carla\Unreal\CarlaUE4\Content\Carla\Static\TrafficSign\PostSigns\Round\SpeedLimiters\BP_SpeedLimit.uasset: [Compiler] Can't connect pins  Object  and  Object : This cast has an invalid target type (was the class deleted without a redirect?). from Source: /Game/Carla/Static/TrafficSign/PostSigns/Round/SpeedLimiters/BP_SpeedLimit.BP_SpeedLimit
ProcessResult.StdOut:   LogBlueprint: Error: [AssetLog] D:\VR\carla\Unreal\CarlaUE4\Content\Carla\Static\TrafficSign\PostSigns\Round\SpeedLimiters\BP_SpeedLimit.uasset: [Compiler] In use pin  AsBase Vehicle Pawn  no longer exists on node  Bad cast node . Please refresh node or break links to remove pin. from Source: /Game/Carla/Static/TrafficSign/PostSigns/Round/SpeedLimiters/BP_SpeedLimit.BP_SpeedLimit
```
This looks like a question similar to [this issue](https://forums.unrealengine.com/t/custom-widget-generates-infine-loading-crash/374336/4). You might try refreshing/redownloading/fixing those blueprint files (in the editor) or just delete them and see if it works then.

zoughi, 2022-10-04:
hi dear Gustavo
I totally delete my "Content" directory and reinstall it
But still the following error remain. I attached the log file:
[Log.txt](https://github.com/HARPLab/DReyeVR/files/9702754/Log.txt)

GustavoSilvera, 2022-10-04:
Looks like there might be more logs in `D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Logs\Cook-2022.10.03-17.51.41.txt`. What do those say?

Also, I'm curious if you've tried building (vanilla) Carla before installing DReyeVR?

zoughi, 2022-10-04:
I installed carla from the following command:
git clone https://github.com/carla-simulator/carla -b 0.9.13
[Cook-2022.10.04-21.42.59.txt](https://github.com/HARPLab/DReyeVR/files/9709615/Cook-2022.10.04-21.42.59.txt)

GustavoSilvera, 2022-10-05:
This is presenting a curiously non-descriptive error message. Have you gotten Carla to build (`make package`) successfully yet? I'm wondering if this is a problem with your Carla installation instead of DReyeVR

zoughi, 2022-10-05:
"make package" command is not terminate successfully. I sent you the log file and whats appear in the "x64 command prompt" is as follow : 
````
  LogAudioCaptureCore: Display: No Audio Capture implementations found. Audio input will be silent.
  LogShaderCompilers: Display: Worker (4/5): shaders left to compile 249
  LogShaderCompilers: Display: Worker (2/5): shaders left to compile 239
  LogShaderCompilers: Display: Worker (1/5): shaders left to compile 229
  LogShaderCompilers: Display: Worker (3/5): shaders left to compile 219
  LogShaderCompilers: Display: Worker (5/5): shaders left to compile 209
  LogShaderCompilers: Display: Worker (4/5): shaders left to compile 199
  LogShaderCompilers: Display: Worker (2/5): shaders left to compile 189
  LogShaderCompilers: Display: Worker (1/5): shaders left to compile 179
  LogShaderCompilers: Display: Worker (5/5): shaders left to compile 169
  LogShaderCompilers: Display: Worker (3/5): shaders left to compile 159
  LogShaderCompilers: Display: Worker (4/5): shaders left to compile 149
  LogShaderCompilers: Display: Worker (2/5): shaders left to compile 139
  LogShaderCompilers: Display: Worker (1/5): shaders left to compile 129
  LogWindows: Error: begin: stack for UAT
  LogWindows: Error: === Critical error: ===
  LogWindows: Error:
  LogWindows: Error: Fatal error!
  LogWindows: Error:
  LogWindows: Error:
  LogWindows: Error: end: stack for UAT
  LogShaderCompilers: Display: Worker (5/5): shaders left to compile 119
  LogShaderCompilers: Display: Worker (2/5): shaders left to compile 109
  LogShaderCompilers: Display: Worker (3/5): shaders left to compile 99
  LogShaderCompilers: Display: Worker (4/5): shaders left to compile 89
Took 113.2335094s to run UE4Editor-Cmd.exe, ExitCode=3
ERROR: Cook failed.
       (see D:\VR\UnrealEngine-carla\UnrealEngine-carla\Engine\Programs\AutomationTool\Saved\Logs\Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
BUILD FAILED

-[Package]: [ERROR] There was a problem while packaging Unreal project.
          [ERROR] Please read the screen log for more information.
make: *** [package] Error 1

````

which is mentioned before in the "log.txt" file.

GustavoSilvera, 2022-10-05:
Right but I mean have you tried `make package` on a clean (vanilla) installation of Carla without DReyeVR. I'm just wondering if this is a problem with your CARLA installation or DReyeVR. If we remove DReyeVR from the equation then you first need to make sure standalone CARLA works and then DReyeVR can be installed.

unrealtz, 2022-11-11:
hi dear Silvera
I have two questions,
1. visual studio 2019 does not suport Windows 8.1 SDK so I install Windows 10 SDK, is it compatible to build Carla?

2. I have a NVIDIA GeForce GTX 1050 which has 4 GB GPU is it adequate to install carla?

CARLA version: 0.9.13
Platform/OS: windows 10

I would be appreciate if you could help me

GustavoSilvera, 2022-11-11:
VS12019 + Windows 10 SDK should be fine to install Carla. 

A GTX 1050 will be somewhat underpowered for vanilla Carla, especially so for DReyeVR (Carla in VR). But it will still install properly, though you'll have to crank down the graphics settings to get acceptable performance (`--quality-level=Low`). To learn more I wrote a [bit here about performance](https://github.com/HARPLab/DReyeVR/issues/10) and here is from [Carla's official documentation](https://carla.readthedocs.io/en/latest/adv_rendering_options/).

closed 2022-12-22 by GustavoSilvera