HARPLab/DReyeVR #61: Is building Unreal from source necessary?

laurayuzheng, 2022-09-19:
First of all, thank you to the authors for being this code available.
Do authors know if building Unreal 4.26 from source necessary? Or, can we use the Unreal 4.26 which is installed by Epic Games launcher?

GustavoSilvera, 2022-09-19:
Hi @laurayuzheng 

Unfortunately yes, it is a **hard** requirement in order to build DReyeVR at its current state. This is not a requirement set by us, but rather by the Carla team when [they upgraded from 0.9.12 to 0.9.13](https://github.com/carla-simulator/carla/blob/master/Docs/build_linux.md#unreal-engine). They do have some Carla-specific patches in [their fork](https://github.com/carlaunreal/unrealengine) that require their fork. 

Hope this helps!

laurayuzheng, 2022-09-19:
Thanks for such a fast response! I will close this issue now.

closed 2022-09-19 by laurayuzheng