HARPLab/DReyeVR #62: Error reading DReyeVRConfig.ini file

shh1v, 2022-09-23:
When I am packaging DReyeVR, it throws an error. It shows that the Cook Failed. I am not sure why this is happening, can you please help me out?
This is the part of the error in the log:
```
  LogInit: Display: LogTemp: Error: No variable matching Camera/FieldOfView found
  LogInit: Display: LogTemp: Error: No variable matching Camera/ScreenPercentage found
  LogInit: Display: LogTemp: Error: No variable matching Camera/VignetteIntensity found
  LogInit: Display: LogTemp: Error: No variable matching Camera/BloomIntensity found
  LogInit: Display: LogTemp: Error: No variable matching Camera/SceneFringeIntensity found
  LogInit: Display: LogTemp: Error: No variable matching Camera/LensFlareIntensity found
  LogInit: Display: LogTemp: Error: No variable matching Camera/GrainIntensity found
  LogInit: Display: LogTemp: Error: No variable matching
 Camera/MotionBlurIntensity found
  LogInit: Display: LogTemp: Error: No variable matching EgoVehicle/CameraInit found
```
I am not sure why this error is showing. So, I don't know what other relevant information you need to resolve this problem. Please let me know if you need any additional information. I have uploaded the full Unreal Engine log file. Thank you!
[Log.txt](https://github.com/HARPLab/DReyeVR/files/9631112/Log.txt)

GustavoSilvera, 2022-09-23:
This error is because you must have updated your `DReyeVRConfig.ini` file without updating the corresponding variables in EgoVehicle src. These camera variables were recent additions as part of the #26 pr which included parametrizable camera effects.

Your `DReyeVRConfig.ini` should have a section like this:
```ini
[CameraParams]
FieldOfView=90.0; horizontal field of view
ScreenPercentage=100; 100% is native resolution, increase for supersampling, decrease for subsampling
; all the intensities range from [0 (off) to 1 (maximum)]
MotionBlurIntensity=0; how much motion blur in the camera
VignetteIntensity=0; how intense the vignetting is (darkened corners)
BloomIntensity=0; how intense the bloom is
SceneFringeIntensity=0; how intense the SceneFringe is
LensFlareIntensity=0; how intense the lens flares are
GrainIntensity=0; how intense the grain is
```

But I suspect it is expecting a config like [this](https://github.com/HARPLab/DReyeVR/blob/5199d55f565a85b72cbe9af5c8f6d65867d770d8/Configs/DReyeVRConfig.ini) (from an older commit). 

You should ensure all the config value params should follow [this](https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/DReyeVRUtils.h#L403-L444) template:
```c++
ReadConfigValue("CameraParams", NameOfVariable, VariableToWriteTo);
```

shh1v, 2022-09-23:
That worked! Thank you so much! I really appreciate it.
We are very close to running the study. Thank you so much for your contribution to our project!

closed 2022-09-23 by shh1v