HARPLab/DReyeVR #63: Hi

inv-samuel, 2022-09-26:
Hi
@rampillas   when i was trying to add a safe distance sensor in carla (as per the doc : https://carla.readthedocs.io/en/0.9.13/tuto_D_create_sensor/) im also facing the same issue

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.13
WARNING: Simulator API version  = 0.9.13-1-g8854804f4-dirty
INFO:  Found the required file in cache!  Carla/Maps/TM/Town10HD_Opt.bin 
INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin 
basic agent set destination 
ActorBlueprint(id=sensor.other.safe_distance,tags=[sensor, other, safe_distance])
Assertion failed: px != 0, file C:\workspace\carla\Build\boost-1.72.0-install\include\boost/smart_ptr/shared_ptr.hpp, line 734

and the server crashed

Could you help us to figure out what we need to do? Thanks for your work and best regards.

_Originally posted by @inv-mary in https://github.com/HARPLab/DReyeVR/issues/12#issuecomment-1257859480_

GustavoSilvera, 2022-09-26:
Hi @inv-samuel  have you tried rebuilding the PythonAPI with command make PythonAPI once DReyeVR is installed?

inv-samuel, 2022-09-27:
@GustavoSilvera  I believe the DReyeVR is one of your publications. Can you please tell me, by installing this, how can this issue be solved?

GustavoSilvera, 2022-09-27:
Hi @inv-samuel this GitHub repository is for hosting DReyeVR content including source code, configs, and community issues. DReyeVR is a layer over CARLA that adds several human-immersion features for research including VR, audio, haptics, eye tracking, mirrors etc. 

I thought (by posting a question here) you were interested in installing DReyeVR and were running into this problem afterwards. If you are not interested in installing DReyeVR then this is a question for the CARLA team over in [their GitHub](https://github.com/carla-simulator/carla/issues).

closed 2022-12-22 by GustavoSilvera