HARPLab/DReyeVR #64: Fatal error! upon CarlaUE4.exe

canmanie-swov, 2022-09-27:
Hello! I'm very excited about getting DReyeVR running but have hit an issue upon the final loading (make check shows everything is fine). When I execute 'CarlaEU4.exe -vr' from WindowsNoEditor, I get the following error log:

Fatal error!

Unhandled Exception: 0xe06d7363

0x00007ff902164fd9 KERNELBASE.dll!UnknownFunction []
0x00007ff8e8256480 VCRUNTIME140.dll!UnknownFunction []
0x00007ff8e2f3293f SRWorks_Log.dll!UnknownFunction []
0x00007ff8e2f33dbd SRWorks_Log.dll!UnknownFunction []
0x00007ff8e01b3e95 ViveSR_Client.dll!UnknownFunction []
0x00007ff8dbf272ef SRanipal.dll!UnknownFunction []
0x00007ff6e3cad1b0 CarlaUE4-Win64-Shipping.exe!SRanipalEye_Framework::StartFramework() [D:\CarlaDReyeVR\carla\Unreal\CarlaUE4\Plugins\SRanipal\Source\SRanipalEye\Private\SRanipalEye_Framework.cpp:49]
0x00007ff6e422282a CarlaUE4-Win64-Shipping.exe!AEgoSensor::BeginPlay() [D:\CarlaDReyeVR\carla\Unreal\CarlaUE4\Source\CarlaUE4\DReyeVR\EgoSensor.cpp:74]
0x00007ff6e59ba9ed CarlaUE4-Win64-Shipping.exe!AActor::DispatchBeginPlay() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\Actor.cpp:3518]
0x00007ff6e59cb07e CarlaUE4-Win64-Shipping.exe!AActor::PostActorConstruction() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\Actor.cpp:3348]
0x00007ff6e59bf5da CarlaUE4-Win64-Shipping.exe!AActor::FinishSpawning() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\Actor.cpp:3247]
0x00007ff6e59ccac2 CarlaUE4-Win64-Shipping.exe!AActor::PostSpawnInitialize() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\Actor.cpp:3186]
0x00007ff6e5d8f579 CarlaUE4-Win64-Shipping.exe!UWorld::SpawnActor() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\LevelActor.cpp:572]
0x00007ff6e5d8f6b7 CarlaUE4-Win64-Shipping.exe!UWorld::SpawnActor() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\LevelActor.cpp:304]
0x00007ff6e4228866 CarlaUE4-Win64-Shipping.exe!AEgoVehicle::BeginPlay() [D:\CarlaDReyeVR\carla\Unreal\CarlaUE4\Source\CarlaUE4\DReyeVR\EgoVehicle.cpp:97]
0x00007ff6e59ba9ed CarlaUE4-Win64-Shipping.exe!AActor::DispatchBeginPlay() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\Actor.cpp:3518]
0x00007ff6e61e679a CarlaUE4-Win64-Shipping.exe!AWorldSettings::NotifyBeginPlay() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\WorldSettings.cpp:249]
0x00007ff6e5caa671 CarlaUE4-Win64-Shipping.exe!AGameStateBase::HandleBeginPlay() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\GameStateBase.cpp:205]
0x00007ff6e61be5fe CarlaUE4-Win64-Shipping.exe!UWorld::BeginPlay() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\World.cpp:4349]
0x00007ff6e615f32a CarlaUE4-Win64-Shipping.exe!UEngine::LoadMap() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\UnrealEngine.cpp:13133]
0x00007ff6e61456f8 CarlaUE4-Win64-Shipping.exe!UEngine::Browse() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\UnrealEngine.cpp:12338]
0x00007ff6e5cbf709 CarlaUE4-Win64-Shipping.exe!UGameInstance::StartGameInstance() [D:\UnrealEngineCarla\Engine\Source\Runtime\Engine\Private\GameInstance.cpp:584]
0x00007ff6e3a45620 CarlaUE4-Win64-Shipping.exe!FEngineLoop::Init() [D:\UnrealEngineCarla\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:3990]
0x00007ff6e3a541e1 CarlaUE4-Win64-Shipping.exe!GuardedMain() [D:\UnrealEngineCarla\Engine\Source\Runtime\Launch\Private\Launch.cpp:155]
0x00007ff6e3a5430a CarlaUE4-Win64-Shipping.exe!GuardedMainWrapper() [D:\UnrealEngineCarla\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:137]
0x00007ff6e3a5f1ff CarlaUE4-Win64-Shipping.exe!WinMain() [D:\UnrealEngineCarla\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:268]
0x00007ff6e6b311fe CarlaUE4-Win64-Shipping.exe!__scrt_common_main_seh() [D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
0x00007ff904067034 KERNEL32.dll!UnknownFunction []
0x00007ff9048a26a1 ntdll.dll!UnknownFunction []

Any tips?

Thanks and again I'm super excited about conducting research with your project!

GustavoSilvera, 2022-09-27:
Hi there, this exception occurred because SRanipal is unable to start. A few questions:
1. Are you interested in using the eye-tracking functionality of DReyeVR?
2. if YES to 1) then do you have the appropriate hardware (HTC Vive Pro Eye) and software installed (SRanipal UE4 plugin)
3. if NO to 1) then did you disable the USE_SRANIPAL flag in the build file?

canmanie-swov, 2022-09-27:
Hello! Yes I am interested and have also an HTC Vive Pro Eye set up, the software installed, and the Plugin moved to the correct folder (according to the DReyeVR instructions). Thus, the flag reads `true`.

GustavoSilvera, 2022-09-27:
Okay great, do you have the SRanipal runtime running in the background on your machine? (Similar to [this](https://mariosbikos.com/wp-content/uploads/2020/02/image-28.png))

GustavoSilvera, 2022-09-27:
Also which version of SRanipal do you have installed?

canmanie-swov, 2022-09-28:
Hi again. I ran `make package` and `make launch` and they both executed successfully. But I am still having the same issue when running `CarlaUE4.exe -vr`. By the way, is it normal that the WindowsNoEditor folder has both a `0.9.13-dirty` and a `CARLA_0.9.13-dirty` directory?

canmanie-swov, 2022-09-28:
Just an update that I found out there is no `dist` folder in my carla build, so something is wrong behind DReyeVR. I will try to get all those issues addressed before coming back here. Thanks for now!

canmanie-swov, 2022-09-30:
Hi again, sorry for the delay. I was having [this](https://github.com/carla-simulator/carla/discussions/4248) issue (also mentioned in the [Carla FAQ](https://carla.readthedocs.io/en/0.9.13/build_faq/#there-is-no-dist-folder-in-pythonapicarla-despite-a-successful-output-message)) and fixed that, which broke something along the way so I reinstalled/rebuilt Unreal and Carla and now those seem to be good again. But still when I run CarlaUE4.exe, it opens and closes almost immediately. 

The latest log file is attached.
[CarlaUE4_2.log](https://github.com/HARPLab/DReyeVR/files/9682669/CarlaUE4_2.log)

canmanie-swov, 2022-09-30:
If I press Play in the editor, it also crashes. Perhaps an issue with my machine except it's new and fits the requirements.

canmanie-swov, 2022-09-30:
Technically this issue is closed. The issue is now that the packaged binary crashes.

closed 2022-09-30 by canmanie-swov