HARPLab/DReyeVR #65: Change egovehicle with Python API

tanzilkm, 2022-09-27:
Hello,
I have custom AI which drives the CAR. We select random Cars. 
Is it possible to Change the egovehicle with Python API ?

Thanks 
Tanzil

GustavoSilvera, 2022-09-27:
Currently we only have one ego vehicle model but it is possible to change the vehicles chassis (static mesh) and dynamics to match any arbitrary car.

tanzilkm, 2022-09-27:
Hi @GustavoSilvera , 
Thank you so much for quick response. 

vehicles chassis would not be problem in my case. 

We need to Drive class by AI. 
When we  use apply_control ( carla.Vehicle apply_control() ) Car still in manual mode. Do you know how can I disable it ?
https://carla.readthedocs.io/en/latest/python_api/#carlavehicle

Thank you 
Tanzil

GustavoSilvera, 2022-09-27:
We currently allow the EgoVehicle to be driveable by manual control (human input) or CARLA Autopilot ([TrafficManager](https://carla.readthedocs.io/en/latest/adv_traffic_manager/)) using our `DReyeVR_AI.py` script. 

However, since the ego vehicle still inherits from a Carla vehicle it should still support the `apply_control()` methods that you might use in your custom driving AI. What do you mean by "manual mode"?

GustavoSilvera, 2022-09-27:
To be more concrete, if you open DReyeVR using one of CARLA's PythonAPI scripts (such as manual_control.py) and override the "`player`" with the ego vehicle as [follows](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#control-handoff-to-ai):
```python
        from DReyeVR_utils import find_ego_vehicle
        self.player = find_ego_vehicle(self.world)
```
then the egovehicle (`self.player` in this case) still supports methods that regular CarlaVehicle's would use (such as `apply_control()`)

tanzilkm, 2022-09-28:
Hi @GustavoSilvera , 
Yes, it worked. Thank you so much. 

In DReyeVR is there any way to take input from VR controls ?

Thanks 
Tanzil

GustavoSilvera, 2022-09-28:
What do you mean by VR controls? 

The default behaviour of the EgoVehicle is to take controls from the server (UnrealEngine's built CarlaUE4.exe executable) either from keyboard inputs or Logitech driving hardware. This also tracks the camera with the VR headset. This is all done natively in UE4 (no PythonAPI client for manual control) to minimize latency and leverage UE4's SteamVR plugin

tanzilkm, 2022-09-28:
VR controls/controller ( [htc vive controller](https://www.vive.com/eu/accessory/controller/)

GustavoSilvera, 2022-09-28:
Ah, currently we don't support VR controllers but they are easy to incorporate into [UE4's SteamVR Input system](https://docs.unrealengine.com/5.0/en-US/set-up-the-steamvr-input-system-in-unreal-engine/) and you can remap other controls (see [`DefaultInput.ini`](https://github.com/HARPLab/DReyeVR/blob/main/Configs/DefaultInput.ini) -- all the DReyeVR controls are suffixed with "`_DReyeVR`")

For instance, I think you might be able to change some of the `AxisName` values to the corresponding Vive Controller Keys.

tanzilkm, 2022-09-28:
> Ah, currently we don't support VR controllers but they are easy to incorporate into [UE4's SteamVR Input [...]

Tha

closed 2022-09-28 by tanzilkm