HARPLab/DReyeVR #66: Broken binary

canmanie-swov, 2022-09-30:
Hello,

After running make package, trying to execute the binary produces the following errors (from `carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\CarlaUE4\Saved`):

```
LogInit: Warning: Incompatible or missing module: CarlaUE4
LogInit: Warning: Incompatible or missing module: Carla
LogInit: Warning: Incompatible or missing module: SRanipal
LogInit: Warning: Incompatible or missing module: SRanipalEye
LogInit: Warning: Incompatible or missing module: SRanipalEyeTracker
LogInit: Warning: Incompatible or missing module: SRanipalEditor
LogInit: Warning: Incompatible or missing module: SRanipalLip
Running D:/CarlaUnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Win64 -Project="D:/CarlaDReyeVR/carla/Build/UE4Carla/0.9.13-dirty/WindowsNoEditor/CarlaUE4/CarlaUE4.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE
Using 'git status' to determine working set for adaptive non-unity build (D:\CarlaUnrealEngine).
Creating makefile for UE4Editor (no existing makefile)
ERROR: Unable to find plugin 'Carla' (referenced via CarlaUE4.uproject). Install it and try again, or remove it from the required plugin list.
LogInit: Warning: Still incompatible or missing module: CarlaUE4
LogInit: Warning: Still incompatible or missing module: Carla
LogInit: Warning: Still incompatible or missing module: SRanipal
LogInit: Warning: Still incompatible or missing module: SRanipalEye
LogInit: Warning: Still incompatible or missing module: SRanipalEyeTracker
LogInit: Warning: Still incompatible or missing module: SRanipalEditor
LogInit: Warning: Still incompatible or missing module: SRanipalLip
LogCore: Engine exit requested (reason: EngineExit() was called)
LogExit: Preparing to exit.
```

However, if I `make launch` and play from the editor, things (including DReyeVR, the VR headset, the eye tracking) seem to be fine. Any idea what could be happening?

GustavoSilvera, 2022-09-30:
Hi, so from your previous thread (#64) it looks like you are:
- Using a relatively new Windows 10 Machine -- Specs?
- Using the HTC Vive Pro Eye
- Having success in the editor (`make launch`)
- Produced a logfile (CarlaUE4.log) that shows no errors, but silently dies after SRanipal starts

Okay, I'll ask a few more questions:
- Are you sure SRanipal is running in the background when you are opening the CarlaUE4.exe binary?
- Which version of SRanipal are you using? We have been testing with 1.3.1.1. 
- Did you patch SRanipal like we suggested to [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#installing-dreyevr)?
- When you are in the editor, do you see debug gaze rays that follow your eye gaze? Or a reticle on the 2D screen overlaying where you are looking at?
- Did you `make clean` before rebuilding with `make package`?
- What does the log from `make package` look like? It appears that the final build binary is corrupted somehow.

GustavoSilvera, 2022-10-03:
Okay, everything seems alright. 

Yes the behaviour of the target circling around is expected when there is no measured EyeTracker data (ex. when SRanipal is not running or disabled). 

I find it interesting that the log errors from the executable can't seem to find the CarlaUE4/Carla targets either. Have you tried simply building (`make package`) vanilla Carla 0.9.13 and checked if that had the same errors?

canmanie-swov, 2022-10-05:
Hi Gustavo, sorry for the delay. I started again from scratch and can say confidently that vanilla Carla works as expected -- I can run the packaged executable. After adding DReyeVR to the mix, the package no longer works. Do you have any other ideas?

canmanie-swov, 2022-10-05:
Is it normal that if I try to open the `CarlaUE4.uproject` file in `0.9.13-dirty\WindowsNoEditor\CarlaUE4` in Visual Studio I get:
This project does not have any source code. You need to add C++ source files to the project from the Editor before you can generate project files.

And if I try to launch it, I see:
[image]

GustavoSilvera, 2022-10-05:
Gotcha, okay thanks for validation. This seems to be an issue with DReyeVR, specifically with SRanipal. Can you try setting the SRanipal flag to false (which will prevent building any SRanipal plugin components) to see if SRanipal is causing the problem?

canmanie-swov, 2022-10-05:
Hi, yes doing that makes a package that works! I will try reinstalling all the SRanipal elements and seeing if it changes anything, although it was working fine in the editor.

canmanie-swov, 2022-10-05:
Hi yes, I did that and the packaged binary works.

canmanie-swov, 2022-10-17:
Hi Gustavo, I tried installing and building everything again from scratch and can confirm now that the problem with SRanipal is not only in package mode, sorry for not noticing this. In the editor, if I try to Play it in VR, it crashes. This makes more sense at least, but still means I can't use DReyeVR properly. I'm currently trying to find a log with meaningful insight, let me know if you have an idea where to look.

canmanie-swov, 2022-10-17:
When it crashes, the last line in the log is:

`Logtemp: Warning: Attempting to use SRanipal eye tracking`

GustavoSilvera, 2022-10-17:
Okay, interesting. Which version of SRanipal are you using? 

There might be more interesting logs in the CarlaUE4.log file which might be available [here](https://forums.unrealengine.com/t/where-are-ue4-crash-log-files-stored/280008).

canmanie-swov, 2022-10-19:
Hey Gustavo, everything works with the version you've sent! I was able to find the 1.3.1.1 version of the installer online (but not officially through VIVE), but I couldn't find the old version of the plugin, so it really is only because you sent it that I could get things working. This is fixed but others may have the same issue. Thanks for all your help and patience!

closed 2022-10-19 by canmanie-swov