HARPLab/DReyeVR #68: LeapMotion plugin

canmanie-swov, 2022-11-03:
This is not an issue, but I am experimenting with the Leap Motion Controller with the goal of having hands visible in the ego vehicle (to increase the realism). I was able to get the [plugin to build successfully and to follow the tutorial](https://developer.leapmotion.com/unreal) to place the hands in the CARLA Unreal editor, but when I click play, they disappear. I was wondering if you could point me toward the files I might need to alter to fix this, or give me any tips.

Thanks for all your support so far!

GustavoSilvera, 2022-11-03:
Wow, this is very cool. We've had similar requests to this (#18 ) before and it is awesome that you've gotten some progress. 

I can try to point you in the right direction but I'm not sure how this API works and have never used it before. First off, when you say "CARLA Unreal editor" do you mean a vanilla (unmodified CARLA) installation or have you installed DReyeVR on it?

I have a feeling that instead of disappearing, the hands are missing an attachment root/actor so they might be still visible in the world but instantly transported to (0, 0, 0). In Town03 this is inside the fountain statue in the middle of the map. 

Can you point to some code you've added to make the change, or any blueprints if they were used. 

Have you tried building in package mode?

canmanie-swov, 2022-11-03:
Hi Gustavo,

Just some background on how I got it running with the DReyeVR/CARLA build: since we are using Unreal 4.26, I more or less followed what can be found here: https://github.com/ultraleap/UnrealPlugin/issues/70. The key parts were updating my VS installation to include Game Development with C++ and the Unreal Engine installer, and installing the plugin version 3.7.11 from [here](https://github.com/ultraleap/UnrealPlugin/releases?page=2). I start the engine with `make launch`.

The Leap plugin is working in this DReyeVR + CARLA installation. We have been making changes to the HMI in the vehicle and this has been working well within DReyeVR,. Now I'm adding the Leap hands to inside the ego vehicle and when I press play, they disappear. I have been attaching them to the ego vehicle (as I did with our HMI elements).  Even though my Leap plugin is enabled, something could be wrong with the toolkit, but if there's a chance it's DReyeVR-related, I'd love to start there :) 

I also tried adding the Leap actors to the DReyeVR sign blueprints, as this is what I did with our HMI elements and it worked within the PythonAPI, but they still didn't appear. Tomorrow (time zones :) ) I'll try building in package mode and keep working on figuring it out. I really am happy with the existence of DReyeVR! The Leap is not a must-have for the experiment we are working on, but more something I'd like to get working anyway!

GustavoSilvera, 2022-11-03:
Okay, nice! I'm happy to help you try to get it working. 

To see if this is DReyeVR related have you tried opening/using the hands plugin in:
- a) an empty UE4 scene (answers the "is this a UE4 problem" question?)
- b) a vanilla (unmodified) carla scene (answers the "is this a Carla problem" question?)
- c) a DReyeVR scene (answers the "is this a DReyeVR problem" question?)

I'm still not sure how the hands interface (Is that what HMI stands for?) is supposed to be rendered in the scene. Usually these things are rendered as actors, so have you tried: Pressing play -> Pressing eject (now you can no-clip around and access the map directory) -> Search for your Hands plugin in the top right Actor directory system. If they are actors they should be listed there and then you can visualize them. 

Code-wise I'm not sure why/how DReyeVR might have an effect on these hands interface. One theory is that these elements might get immediately [garbage collected](https://unrealcommunity.wiki/garbage-collection-36d1da) if they are not accessible from any main scene objects, but this is kind of a stretch.

canmanie-swov, 2022-11-04:
Hi Gustavo,

(Sorry, by HMI elements I meant the human-machine interface design we have been incorporating in the vehicle)

All good! Well there is a lot to figure out such as correctly mirroring the hands (screenshot below) and callibrating it with our steering wheel setup so they align when our participants hold the steering wheel but I'm super happy that it's working this far. In the end, the trickiest part is finding the right versions, and I had to install [3.2.1](3.2.1 from https://developer.leapmotion.com/releases?offset=1611192787115) of the Leap software in addition to the plugin version mentioned above and now it works. This is a pretty cheap device so I think it integrates well with your project goals!

[image]

closed 2022-11-04 by canmanie-swov

GustavoSilvera, 2022-11-04:
Wow yeah this looks great! Nice work!