HARPLab/DReyeVR #69: connecting created map in roadrunner to Dreyevr

FBanani, 2022-11-04:
Hi 
I created the map in roadrunner, and I want to use the amazing simulation you made not only in just Carla's maps but also in other maps which can connect to Carla. Could you help me to bring this vehicle to other maps?

GustavoSilvera, 2022-11-04:
Sure! There are two critical steps that you'll need to do:
1. Replace the level blueprint with our own DReyeVRLevel code. 
2. Drag-and-drop the EgoVehicle blueprint into the world

You can do 1.) by "reparenting" the existing UE4 level blueprint with the [`ADReyeVRLevel`](https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/LevelScript.h) blueprint that should be available. 
1. In the main editor, click `"Blueprints"` (top bar near the "Play" button)
2. Click `"Open Level Blueprint"` to open a new BP editor window
3. In the top bar of the new BP editor window click `"Class Settings"`
4. In `Class Options` (to the right in "details" tab) click the `Parent Class` and select `"DReyeVRLevel"`. Accept all the prompts and you should now be good to go!

You can then do 2.) by finding the EgoVehicle in the UE4 Content Browser (located at `Content/Carla/Blueprints/Vehicles/DReyeVR/BP_EgoVehicle_DReyeVR.uasset`) click this blueprint and drag+drop it to wherever you'd like in your new map. 

Once you've reparented the level script and added the EgoVehicle then you should be able to just press `Play` and have everything work. 

There are also useful logging messages for initialization you might want to look at in the resulting CarlaUE4.log file in case you run into issues.

FBanani, 2022-11-04:
Thank you so much.

closed 2022-12-28 by GustavoSilvera