HARPLab/DReyeVR #71: export from Roadrunner

FBanani, 2022-12-21:
Hi
I have 2 main issues:
**first one export and import**
When I want to export the map to DReyeVR should I export to Carla or unreal engine?
From your last guide, I understand I have to export to Unreal engine is it true?
if yes, I saw the following warning! 
[image]
and the export files is like as follows:
[image]
The steps that I do is as follows:
then I import the unreal engine in this way:
1:
[image]
2:
[image]
In this step should I select all of them?
After that, I press the "import all" button as follows:
[image]
Therefore, it shows me this message and does not add the new map to the unreal engine!
Could you help me? 
**Second one from your last guide:**
[image]
1: My question is should I do anything before the following 4 items under this sentence "You can do 1.) by "reparenting" the existing UE4 level blueprint with the [ADReyeVRLevel](https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/LevelScript.h) blueprint that should be available."?

2: You can then do 2.) by finding the EgoVehicle in the UE4 Content Browser (located at Content/Carla/Blueprints/Vehicles/DReyeVR/BP_EgoVehicle_DReyeVR.uasset) click this blueprint and drag+drop it to wherever you'd like in your new map."
I found it as follows but I do not understand where I should drop it.
[image]
Regards

GustavoSilvera, 2022-12-22:
Hi @FBanani, 

First I'd like to preface that I've never used the RoadRunner map customization tool before, so I'm not super familiar with the tech. 

For 1. how do you know the new map was not added? Have you tried opening a new map with `CTRL+O` to select your new map?

For 2. Sorry if this was confusing, but those 4 steps (1, 2, 3, 4) after the sentence are how you "reparent" the level blueprint. So you don't need to do anything beforehand. Just follow 1-4 and then you can add the EgoVehicle. Once you've opened the EgoVehicle in the content menu (as you have) and you have the map view open, you should be able to click and drag from the BP_EgoVehicle_DReyeVR icon to hover over your (new or old) map. Then you can reposition however you like.

FBanani, 2022-12-22:
Thank you. 
Actually, with CTRL+O it shows me maps were provided in the Carla but there is nothing new there. when I added it brings all the assets but does not show the new map which is provided in RoudeRunner as you can see in the following:
[image]

GustavoSilvera, 2022-12-22:
Hm, I'm not sure since we've never worked with RoadRunner before. My best guess would be to look for new files in your project directory (and/or look at the logs in the developer console to see if there are any log messages indicating the location). 

You'll also want to make sure to follow the instructions [here](https://carla.readthedocs.io/en/latest/tuto_M_generate_map/) and [here](https://carla.readthedocs.io/en/latest/tuto_M_add_map_source/) on RoadRunner import.

However you'll probably have better luck forwarding these questions to the Carla/RoadRunner team.

FBanani, 2023-01-04:
Hi
Happy new year.
Actually, I exported the RouadRunner map, but I have issues:
1) I see a purple arow, how can I get rid of it? 
[image]
2) the steering well does not work, what should I do?
3)when I want to execute CarlaUE4.exe -vr in CarlaDReyeVR\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor it does not show the new map, what should I do?
Thanks

GustavoSilvera, 2023-01-04:
Hi, the purple ray is visualizing the eye gaze in Editor mode, when you build in package mode (`make package`) it is not present. You can manually disable it (in editor) by commenting out the `DebugLines()` function in `AEgoVehicle::Tick` in `EgoVehicle.cpp`. 

If the steering wheel doesn't work, you should make sure to have the Logitech G Hub installed and working, the logitech plugin installed (and enabled in the Build.cs file), and the debug prints enabled to see if we can detect it or not. 

in package mode it may have not exported your new map, this required making sure the map is [correctly being cooked](https://forums.unrealengine.com/t/list-of-maps-to-cook/55462) and is present in the WindowsNoEditor folder within the Maps directory.

FBanani, 2023-01-04:
Thanks. Actually, the steering wheel works in the other Carla's map but does not work in the RoudRunner!
I do not understand your solution could you explain it a little more?
Regards

FBanani, 2023-01-04:
I try to use the steering wheel in the UE4 editor after pressing the Play button. The steering wheel works in the environment showing after executing the CarlaUE4.exe -vr which is not a customized map that I made but a Carla map. I tried to execute "make package" when the UE4 window was open but it showed me an error. Could you guide me on what should I do?
Regards

GustavoSilvera, 2023-01-04:
What is the error you are getting?

FBanani, 2023-01-05:
I think I found where my issue rise:
I want to setup my new map. Therefore, I have to change
[image]
Town3 which is in the "CarlaDReyeVR\carla\Unreal\CarlaUE4\Content\Carla\Maps" directory to my new map which is in the different directory "CarlaDReyeVR\carla\Unreal\CarlaUE4\Content\map_package\Maps\New_Map". How should I change the "DefaultEngine.ini". because I could not find a Game folder to change the directory.

FBanani, 2023-01-05:
I closed the unreal engine and executed "make package" in this time I did not see the previous error.

FBanani, 2023-01-05:
I checked and understood that in the unreal Engine when I press the play button. The environment has been shown in the VR headset, but the steering wheel does not work. On the other hand, the steering wheel just works in the environment in the CalaUE4 windows. Therefore, I think I found where my issue rise: I have to cook my new map. So, I have to change
[image]
Town3 which is in the "CarlaDReyeVR\carla\Unreal\CarlaUE4\Content\Carla\Maps" directory to my new map which is in the different directory "CarlaDReyeVR\carla\Unreal\CarlaUE4\Content\map_package\Maps\New_Map".  Because when I wanted to change Town3 to Town4 I just changed it here and executed "make package". But in this case, how should I change the "DefaultEngine.ini". Due to the different directory. Also, I could not find a Game folder (Shows in the attached picture) to change the directory to find my map.
the other thing is my DefaultEditor.ini is empty how it can cook maps.
Regards

GustavoSilvera, 2023-01-05:
Not sure since this is untested waters, but have you tried `/Game/map_package/Maps/New_Map.New_Map`?

FBanani, 2023-01-05:
No, it is not working.

FBanani, 2023-01-05:
I think the answer is this solution:
[image]
but I could not understand how I can see in the CarlaUE4 window after executing " CarlaUE4.exe -vr" because my steering wheel works in the unreal Engine when I press the Play button and when I make launch my new map has been shown as a default but when I execute " CarlaUE4.exe -vr" it does not show my new map in the CarlaUE4 window. Could you guide me how to change my config files in the directory "CarlaDReyeVR\carla\Unreal\CarlaUE4\Config"?
this is the map when I make launch:
[image]
this is the map when I press Play Button in the unreal Engine:
[image]
this is the map when I execute " CarlaUE4.exe -vr" (Town4 Carla):
[image]
I need to change this map to my new map as a default

GustavoSilvera, 2023-01-05:
Have you looked at this documentation for [packaging a large (new) map](https://carla.readthedocs.io/en/latest/large_map_import/)?

GustavoSilvera, 2023-01-05:
I think the addition of `ARGS="--packages=<mapPackages>" (along with the `make import ARGS="--package=<package_name>") should work?

But unfortunately this seems like more of a Carla issue than DReyeVR, have you gotten the map to show up in the package for a vanilla carla executable (no DReyeVR installed), this will tell us if DReyeVR has anything to do with the issue.

FBanani, 2023-01-05:
well after doing "ARGS="--packages=<mapPackages>"" my situation become worst (after press the Play button) the unreal engine shows the Fatal error and closed:
[image]
I erased all the things and just execute the make import but it crashes.
what should I do to fix it.

GustavoSilvera, 2023-01-05:
Does this still work in the editor? Try to find logs to see what is happening before the `Fatal error`

GustavoSilvera, 2023-01-05:
Specifically, these logs should be located in `CarlaUE4.log`, see if there are any errors mentioned there

FBanani, 2023-01-05:
The unreal engine works but when I press the play button it shows me this error. Do you think I have to execute "make clean"?
Sorry to bother you a lot I am a transportation engineer from Civil and I am not a computer. engineer. So, I face issues I am not pretty well to solve or recognize issues related to DReyeVR or Carla. I like to extend DreyeVR in any customized maps. If I could find the solution, your simulation is the most amazing simulation that I have ever known in any aspects.

GustavoSilvera, 2023-01-05:
Executing a `make clean` might fix the problem. I'd still recommend looking for the debug trace (crash log file) with the CarlaUE4.log file either [live in the editor](https://forums.unrealengine.com/t/how-do-i-view-logging-output/283700) (until it crashes) or once it is [saved to a file](https://forums.unrealengine.com/t/where-are-ue4-crash-log-files-stored/280008) (more reliable)

FBanani, 2023-01-13:
Hi, thanks for your help.
I fixed my issue.

closed 2023-01-13 by GustavoSilvera