HARPLab/DReyeVR #72: I can not use LogitechWheel in DReyeVR  project

unrealtz, 2022-12-27:
I have install DReyeVR from "https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md" instructor.
I use "Logitech G923 Racing Wheel & Pedals" and add it to the project by running the following lines:
````
# (optional) install LogitechWheelPlugin
git clone https://github.com/HARPLab/LogitechWheelPlugin
mv LogitechWheelPlugin/LogitechWheelPlugin Unreal/CarlaUE4/Plugins/ # install to carla

cd carla
make PythonAPI  # build the PythonAPI (and LibCarla) again
make launch     # launch in editor
make package    # create an optimized package
make check      # run Carla unit tests
````
I set the following flag to true
//////////////////////////////////////////////////////////////
// Edit these variables to enable/disable features of DReyeVR
bool UseSRanipalPlugin = false;
bool UseLogitechPlugin = true;
...
//////////////////////////////////////////////////////////////
but when running the following command LogitechWheel is not activate and I can run the car just by keyboard:
````
# on Windows x64 Visual C++ Toolset
cd \PATH\TO\CARLA\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\
CarlaUE4.exe
````

GustavoSilvera, 2022-12-27:
Did you set the `UseLogitechPlugin` flag to `true` before you `make package` again? You'll need to rebuild in order to add this functionality.

unrealtz, 2022-12-28:
thank you,
I have solved the problem by installing "Logitech G HUB".
I have another question, can I add "manual gear shift" to the car or it should be "Automatic gear"?

best regards

GustavoSilvera, 2022-12-28:
We have only programmed it to be in automatic gear.

closed 2022-12-28 by GustavoSilvera

zoughi, 2023-01-19:
hi, dear
I have a question, I would be appreciate if you answer me.
we run your DReyeVR project correctly using "Logitech G HUB". currently, we use the button on the wheel  for driving (D) and Reversing (R).
how can we use "Logitech G Driving Force Shifter" for driving (D) and Reversing (R)?

best regards

GustavoSilvera, 2023-01-20:
You'll first need to figure out what the shifter inputs correspond to in the [UE4 Logitech plugin](https://github.com/HARPLab/LogitechWheelPlugin), I currently don't know of a great way to do this than enabling (in [configs](https://github.com/HARPLab/DReyeVR/blob/main/Configs/DReyeVRConfig.ini#L120)) the [debugging/logging system](https://github.com/HARPLab/LogitechWheelPlugin) and brute-force trying all the inputs until finding a match. 

Then you'll want to map them to [these inputs.](https://github.com/HARPLab/DReyeVR/blob/8b4a31b78aa6b4f3107f906ef309b11698919143/DReyeVR/DReyeVRPawn.cpp#L497-L501) but rather than [PressReverse and ReleaseReverse](https://github.com/HARPLab/DReyeVR/blob/8b4a31b78aa6b4f3107f906ef309b11698919143/DReyeVR/EgoInputs.cpp#L143-L171) you'll probably want a function that directly sets the target gear to some setting (D, N, P, R, etc.) using [SetTargetGear](https://docs.unrealengine.com/4.26/en-US/API/Plugins/PhysXVehicles/UWheeledVehicleMovementComponent/SetTargetGear/) like in EgoInputs.cpp. 
```c++
// example
EgoVehicle->GetVehicleMovementComponent()->SetTargetGear(NewGear, true);
```