HARPLab/DReyeVR #75: Question about cropping of view on desktop vs headset view

canmanie-swov, 2023-01-17:
Hello! We are close to running our study now which uses eye-tracking feedback in real time, so the accuracy of the eye-tracking is particularly important to us. Even after successful calibration, it appears that the cross-hair which overlays the gaze-position on the Carla simulator window is always off in the same specific way. It's not hugely off but since we want to see gaze behaviour to objects in the vehicle which are quite near, this becomes more of a problem. 

We noticed that the Carla window shows a cropped view compared to the view of the VR headset (which you can view side by side via SteamVR). If you were to overlay the cross-hair over the VR view, it looks like it would be accurate. Do you think it is possible that the calculated gaze position is not taking into account the cropping of the screen?

canmanie-swov, 2023-01-17:
And a follow up question, which do you think is accurate, if the position were overlaid on to the Carla window or the VR headset view?

GustavoSilvera, 2023-01-17:
Hi, the cross-hair overlay "reticle" has its primary logic [here](https://github.com/HARPLab/DReyeVR/blob/8b4a31b78aa6b4f3107f906ef309b11698919143/DReyeVR/DReyeVRPawn.cpp#L251-L285). It is primarily designed as an assistive feature for the experimenter to get a close approximation of the gaze rays produced by the participant. 

There are a few reasons why the reticle might look slightly offset, but I think the primary one is because the rendering is done by projecting the 3D point on the [central gaze ray 10 meters away](https://github.com/HARPLab/DReyeVR/blob/8b4a31b78aa6b4f3107f906ef309b11698919143/DReyeVR/DReyeVRPawn.cpp#L258) from the eye origin to the screen (and this 10m was just an arbitrary quantity for visualization purposes) so it might not necessarily match the vanishing point or focus point of the gaze. We found it was very difficult to get the actual focus point since computing the [vergence](https://en.wikipedia.org/wiki/Vergence) length of the rays was highly noisy and unreliable with our HMD sensor. 

Another good reason (which you pointed out) why the crosshair might look off is because the "[spectator view](https://docs.unrealengine.com/4.26/en-US/SharingAndReleasing/XRDevelopment/VR/DevelopVR/VRSpectatorScreen/)" you are seeing is a camera dump from the left eye by default (specifically using UE4's [TexturePlusEye](https://github.com/HARPLab/DReyeVR/blob/8b4a31b78aa6b4f3107f906ef309b11698919143/DReyeVR/DReyeVRPawn.cpp#L179-L180) with the reticle as the overlaid texture). This is not centered as you've noticed and looks different from the SteamVR combined view. 

The most accurate rendering of the eye gaze for visualization has to be the 3D debug gaze rays that emit from each eye origin to the scene. These can be enabled by enabling this [code](https://github.com/HARPLab/DReyeVR/blob/8b4a31b78aa6b4f3107f906ef309b11698919143/DReyeVR/EgoVehicle.cpp#L754-L776). 

Of course you can feel free to tweak the screen-space reticle position to best match whatever 3D point you'd like from the gaze rays.

canmanie-swov, 2023-01-18:
Hi Gustavo. Thanks for your response! This clears a lot up.

closed 2023-01-18 by canmanie-swov