HARPLab/DReyeVR #77: creating bicycle simulator

FBanani, 2023-01-19:
Dear GustavoSilvera I come with a new issue. Is there any way to have two driving simulators in one environment? 
I mean in 2 different PC two different people drive their own DReyeVR simulator, but they can see each other's car and have interaction with each other?

GustavoSilvera, 2023-01-19:
Having multiple EgoVehicles within the same world in Carla/DReyeVR is difficult because it has to go through a network, so both of the worlds are communicated. While not impossible (since we could create a peer-to-peer mirror of the other person's EgoVehicle and sensor characteristics) this is certainly non-trivial in Carla and we haven't even really thought about this or designed DReyeVR to support this in any way. 

Since the EgoVehicles in DReyeVR are run on the "server" (UE4 binary) directly (to make use of UE4's VR capabilities and improved performance) rather than through a pygame client (which could then be operated over a network) I don't think this can be done in the same way Carla handles multiple ego-vehicles. 

What could be done is spawning a non-VR-DReyeVR-ego-vehicle using something like `manual_control.py` while there is a participant in the main "server", but then only one of the two participants is in VR while the other is controlling another vehicle through the flat Carla pygame instance. 

TL;DR: having two VR players on the same DReyeVR instance is very difficult because of synchronization/communication, but having one VR and one pygame player should be doable. Does this work for your situation?

FBanani, 2023-01-19:
You mean that it is possible to have 2 different people one of them can see Carla through VR headset one of them can see through the monitor, yes? If yes, could both of them drive with different Logitech wheels and pedals?

GustavoSilvera, 2023-01-19:
Yes, I believe this should work since it is effectively just running DReyeVR **plus** using this [`manual_control.py`](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control.py) script (provided by CARLA) which also supports a [logitech wheel plugin](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control_steeringwheel.py). However I'm not sure if these two players can run on the same computer (since the focus will be either on the server (player 1) or the pygame (player 2), so it might be simpler to have two computers where only one of them (with the better GPU) running the Carla server (with the VR participant: player1) and the other one (player2) just runs the pygame and connects to the fast computer.

FBanani, 2023-01-19:
Thank you so much, I want to find the combination of two different drivers in the driving simulator, therefore I ask this question. but I did not understand how connect player 2 to player 1. I have 2 computers that DReyeVR has been installed on them.
what should I do? Could you elaborate a little more?

GustavoSilvera, 2023-01-19:
I can elaborate but know that I haven't been able to test this at all and I only think it *should* work:

(I suspect there might be some python bugs in creating a new "hero" vehicle while the DReyeVR EgoVehicle already exists. But let me know if you run into those and I'll take a closer look)

These are the steps I think would work:
1. Install Carla+DReyeVR on both computer A (the faster one) and B (can be slower/cheaper than computer A)
2. Get computer A's ip information: `ipconfig /all` (in Windows cmd) or `ip a` (in Linux), lets say its 192.168.44.15. Carla's [default port is 2000](https://github.com/carla-simulator/carla/blob/1f18d3b911d2deb98999dc1f428bc0e888ac254d/PythonAPI/examples/manual_control.py#L1320-L1325) so you can leave this unchanged. 
3. Player 1 enters DReyeVR in the VR setting on computer A (the fast computer) and can drive manually with the logitech hardware
4. Player 2 (with DReyeVR installed on Computer B) runs `python PythonAPI/examples/manual_control.py --host 192.168.44.15 --port 2000` so this connects to Computer A's running Carla instance and starts a `pygame` instance that transmits the new "Player 2" vehicle to Computer B.

I'm also not sure how good the quality will be on Computer B since Computer A needs to transmit the rendered scene from its local Carla instance over the network to Computer B (so it might be choppy/laggy). But YMMV.

FBanani, 2023-01-23:
Hi
I see this error:
Traceback (most recent call last):
  File "PythonAPI/examples/manual_control.py", line 1325, in <module>
    main()
  File "PythonAPI/examples/manual_control.py", line 1317, in main
    game_loop(args)
  File "PythonAPI/examples/manual_control.py", line 1191, in game_loop
    sim_world = client.get_world()
RuntimeError: time-out of 20000ms while waiting for the simulator, make sure the simulator is ready and connected to 10.104.90.300:2000

but I executed CarlaUE4.exe -vr in the other computer and I see the environment is ready on that computer. then I execute the python PythonAPI/examples/manual_control.py --host 10.104.90.300 --port 2000 on the other computer
why this happened? because I think this error related to the not founding an environment in the carla!

FBanani, 2023-01-25:
in this time have this error. I know that because it spawned randomly. but how can I change it to know spawn in the location of DreyeVr vehicle?
INFO:  Could not find the required file in cache, downloading...  Carla/Maps//OpenDrive/map3.xodr
WARNING: cannot parse georeference: ''. Using default values.
WARNING: Traffic sign 6 overlaps a driving lane. Moving out of the road...
   Warning! Actor Generation is not valid. No actor will be spawned.
Traceback (most recent call last):
  File "PythonAPI/examples/manual_control.py", line 1325, in <module>
    main()
  File "PythonAPI/examples/manual_control.py", line 1317, in main
    game_loop(args)
  File "PythonAPI/examples/manual_control.py", line 1214, in game_loop
    world = World(sim_world, hud, args)
  File "PythonAPI/examples/manual_control.py", line 217, in __init__
    self.restart()
  File "PythonAPI/examples/manual_control.py", line 246, in restart
    blueprint = random.choice(get_actor_blueprints(self.world, self._actor_filter, self._actor_generation))
  File "C:\Users\elnaz\anaconda3\envs\carla\lib\random.py", line 261, in choice
    raise IndexError('Cannot choose from an empty sequence') from None
IndexError: Cannot choose from an empty sequence

GustavoSilvera, 2023-01-25:
Have you tried just running `manual_control.py` on the host machine (A) while the DReyeVR simulator was occupied and running? I was able to run `manual_control.py` on the host machine (A) without issue and both the server (DReyeVR instance) and client (pygame instance) were working nicely. Your error message is regarding missing blueprint data rather than spawn location information so I'm not sure what you are asking. 

If you'd like to know where the DReyeVR vehicle is located, you should follow [these steps](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#using-our-new-dreyevr-pythonapi) to get a reference to the EgoVehicle and then just use it like a normal carla.LibCarla.vehicle actor (which tracks orientation for you). For instance, once you have the reference in a variable you can use Carla's `get_location()` method as described [here](https://github.com/carla-simulator/carla/blob/master/Docs/python_api.md#getters).

FBanani, 2023-01-26:
I found out I could not open pygame window. I have this error:
   Warning! Actor Generation is not valid. No actor will be spawned.
Traceback (most recent call last):
  File "manual_control.py", line 1325, in <module>
    main()
  File "manual_control.py", line 1317, in main
    game_loop(args)
  File "manual_control.py", line 1214, in game_loop
    world = World(sim_world, hud, args)
  File "manual_control.py", line 217, in __init__
    self.restart()
  File "manual_control.py", line 246, in restart
    blueprint = random.choice(get_actor_blueprints(self.world, self._actor_filter, self._actor_generation))
  File "C:\Users\elnaz\AppData\Local\Programs\Python\Python37\lib\random.py", line 261, in choice
    raise IndexError('Cannot choose from an empty sequence') from None
IndexError: Cannot choose from an empty sequence
I could not find any solution for this.
I could execute generate_traffic.py without problem.

GustavoSilvera, 2023-01-26:
Hmm. Can you install DReyeVR on `gamemode` branch to a clean version of Carla? I verified this should work but not sure about `main` yet.

FBanani, 2023-01-26:
sure, but how? could you elaborate more? what should I do?

GustavoSilvera, 2023-01-26:
run `git pull` in the DReyeVR repo, then `git checkout gamemode`, then run `make install` on a clean version of CARLA (with no DReyeVR)

GustavoSilvera, 2023-01-26:
(This is a WIP branch that should be merged into `main` soon)

GustavoSilvera, 2023-01-26:
(You should `git pull` to get to the latest commit (currently e19925a2409924039106e63d52413548eb1dfd17) on DReyeVR)

FBanani, 2023-01-27:
thanks. should I install DReyeVR with the branch of game mode on both PCs?

GustavoSilvera, 2023-01-27:
Yes, and you'll need to rebuild the python API for both as well

wx983, 2023-03-18:
> 您是否尝试过`manual_control.py`在 DReyeVR 模拟器被占用并运行时仅在主机 (A) 上运行？我能够`manual_control.py`在主机 (A) 上毫无问题地运行，并且服务器（DREyeVR 实例）和客户端（pygame 实例）都运行良好。您的错误消息是关于缺少蓝图数据而不是生成位置信息，所以我不确定您在问什么。
> 
> 如果您想知道 DREyeVR 车辆的位置，您应该按照[以下步骤](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#using-our-new-dreyevr-pythonapi)获取对 EgoVehicle 的引用，然后像普通的 carla.LibCarla.vehicle actor（为您跟踪方向）一样使用它。例如，一旦您在变量中有了引用，您就可以按照[此处](https://github.com/carla-simulator/carla/blob/master/Docs/python_api.md#getters)`get_location()`所述使用 Carla 的方法。[](https://github.com/carla-simulator/carla/blob/master/Docs/python_api.md#getters)

> 您是否尝试过`manual_control.py`在 DReyeVR 模拟器被占用并运行时仅在主机 (A) 上运行？我能够`manual_control.py`在主机 (A) 上毫无问题地运行，并且服务器（DREyeVR 实例）和客户端（pygame 实例）都运行良好。您的错误消息是关于缺少蓝图数据而不是生成位置信息，所以我不确定您在问什么。
> 
> 如果您想知道 DREyeVR 车辆的位置，您应该按照[以下步骤](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#using-our-new-dreyevr-pythonapi)获取对 EgoVehicle 的引用，然后像普通的 carla.LibCarla.vehicle actor（为您跟踪方向）一样使用它。例如，一旦您在变量中有了引用，您就可以按照[此处](https://github.com/carla-simulator/carla/blob/master/Docs/python_api.md#getters)`get_location()`所述使用 Carla 的方法。[](https://github.com/carla-simulator/carla/blob/master/Docs/python_api.md#getters)

Hi! How to run `manual_control.py` and DReyeVR vr mode in one computer? Can you explain it in detail?

wx983, 2023-03-18:
> 我发现在执行 manual_control.py 时无法打开 pygame 窗口，但是我可以毫无问题地执行 generate_traffic.py。我遇到了这个错误，我找不到任何解决方案： 警告！演员生成无效。不会产生任何演员。 追溯（最近调用最后）： 文件“manual_control.py”，第 1325 行，在 main() 文件“manual_control.py”，第 1317 行，在主 game_loop(args) 文件“manual_control.py”，第 1214 行，在 game_loop world = World(sim_world, hud, args) 文件“manual_control.py”，第 217 行，在**init** self.restart() 文件“manual_control.py”，第 246 行，在重启 blueprint = random.choice(get_actor_blueprints(self.world, self._actor_filter, self._actor_generation)) 文件“C:\Users\elnaz\AppData\Local\Programs\Python\Python37\lib\random.py”，第 261 行，在选择中 引发 IndexError('Cannot choose from an empty sequence') from None IndexError: Cannot choose from an empty sequence
> 
> 我在一台有 carla 的电脑上执行了 manual_control.py，但我还没有安装 DREyeVR，我没有遇到这个问题（附图片）。你可以帮帮我吗？ [image]

@GustavoSilvera @FBanani Hi! I also encountered the same problem when running `manual_control.py` and DReyeVR vr mode in one computer. Need I change to the `gamemode` right now, or something else?

wx983, 2023-03-18:
> run `git pull` in the DReyeVR repo, then `git checkout gamemode`, then run `make install` on a clean version of CARLA (with no DReyeVR)

I've already installed DReyeVR, how to uninstall DReyeVR to get a clean version of Carla? Can you guide me?

GustavoSilvera, 2023-03-18:
`gamemode` is now fully merged into `main` (since #74) so as long as you `git pull` from the main branch, you should be good to go. 

You should be able to run `python PythonAPI/examples/manual_control.py` to start the pygame client, which can then interact with DReyeVR as if it were a normal Carla installation. 

For future updates of DReyeVR (in case you are not on the latest version) you should follow the [instructions here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#upgrading-dreyevr) on how to update DReyeVR (clean carla and reinstall DReyeVR)

wx983, 2023-03-19:
> `gamemode` is now fully merged into `main` (since #74) so as long as you `git pull` from the main branch, you should be good to go.
>
> You should be able to run `python [...]
[image]
I have already in `main` branch.
But I still encountered the same question as following when I run `python PythonAPI/examples/manual_control.py` or `python PythonAPI/examples/manual_control.py --host 10.1.44.122 --port 2000`.
[image]

GustavoSilvera, 2023-03-19:
Hm, it says something about `Actor Generation is not valid`. I think DReyeVR is sufficiently installed but there is something else going wrong. Can you try running with all vehicle generations: use flag `--generation "All"` in addition to the host/port args.

wx983, 2023-03-19:
> Hm, it says something about `Actor Generation is not valid`. I think DReyeVR is sufficiently installed but there is something else going wrong. Can you try running with all vehicle generations: use flag `--generation "All"` in addition to the host/port args.

Oh yes, it works! Thank you!
[image]

FBanani, 2023-03-20:
> > Hm, it says something about `Actor Generation is not valid`. I think DReyeVR is sufficiently installed but there is something else going wrong. Can you try running with all vehicle generations: use flag `--generation "All"` in addition to the host/port args.
> 
> Oh yes, it works! Thank you! [image]

> Hm, it says something about `Actor Generation is not valid`. I think DReyeVR is sufficiently installed but there is something else going wrong. Can you try running with all vehicle generations: use flag `--generation "All"` in addition to the host/port args.

Hi. I have another question I want another user who drives with Logitech wheels and pedals as well as a person who drivers with DReyeVR what should I do?

wx983, 2023-03-21:
@GustavoSilvera 
Hi, I want to know how to output the location of `manual_control.py` vehicle in each frame, just like EgoVehicle.
[image]

GustavoSilvera, 2023-03-21:
You'll need to track that separately or through Carla's built-in means. Since it is all obtained through Python this should not be very difficult to do.

wx983, 2023-03-22:
> 

How to do that? Can you give me more guidance?Thank you very much!

wx983, 2023-03-22:
> You'll need to track that separately or through Carla's built-in means. Since it is all obtained through Python this should not be very difficult to do.

How to do that? Can you give me more guidance?Thank you very much!

GustavoSilvera, 2023-03-22:
Well, seeing as the location is obtained [like so](https://github.com/carla-simulator/carla/blob/0.9.13/PythonAPI/examples/manual_control.py#L678) you can probably just `print` this variable and save it to a file.

closed 2023-05-10 by GustavoSilvera