HARPLab/DReyeVR #78: 'make launch' failed after installing SRanipal and Logitech Wheel Plugin: assertion failed

wx983, 2023-02-02:
build carla 0.9.13 from source

I was able to use carla before installing SRanipal and Logitech Wheel Plugin.
I've already 'make pythonAPI' again after installing above Plugins and it reports no errors seemingly from the log information. But when 'make launch', it generates an error as beblow:
[image].

GustavoSilvera, 2023-02-02:
Hi there, we noticed this problem as well recently when we tried rebuilding from scratch to validate build settings. 

I believe the issue is that you are using "too new" of a UE4 commit. The CARLA fork works for DX12 but we prefer DX11 for its improved VR performance and compatibility. So I recommend you `checkout` a UE4 version before the CARLA `0.9.14` work and rebuild the engine and editor. 

We verified that switching the Carla Engine fork to `d40ec35474e8793b4eea60dba6c4f051186e458e` worked for us. Or you can also use this [custom fork](https://github.com/gustavosilvera/unrealengine/tree/fov-render) (on branch [`fov-render`](https://github.com/gustavosilvera/unrealengine/tree/fov-render)) that we've been developing with. 

(NOTE: this is only for Carla/DReyeVR 0.9.13)

We will be working on a patch for Carla/DReyeVR 0.9.14 (and the newest version of CarlaUnreal's UE4)  soon!

wx983, 2023-02-03:
Hmm, I'm not very familiar with git commands. If I want to switch the Carla Engine fork to d40ec35474e8793b4eea60dba6c4f051186e458e, is it as following two commands?
1. git clone https://github.com/CarlaUnreal/UnrealEngine.git
2. git checkout d40ec35474e8793b4eea60dba6c4f051186e458e

wx983, 2023-02-03:
> Hmm, I'm not very familiar with git commands. If I want to switch the Carla Engine fork to d40ec35474e8793b4eea60dba6c4f051186e458e, is it as following two commands?
> 
> 1. git clone https://github.com/CarlaUnreal/UnrealEngine.git
> 2. git checkout d40ec35474e8793b4eea60dba6c4f051186e458e
something goes wrong with me. It seems the whole Carla Engine project is too big in size, so I can't git it through 'git clone https://github.com/CarlaUnreal/UnrealEngine.git', is there any method to obtain 'd40ec35474e8793b4eea60dba6c4f051186e458e' ?

GustavoSilvera, 2023-02-03:
Okay, try instead this command: 
```bash
git clone https://github.com/GustavoSilvera/UnrealEngine -b fov-render --depth 1
```
This should work with 0.9.13 and only clone one commit.

closed 2023-02-13 by wx983