HARPLab/DReyeVR #80: can not find frame capture images in Unreal/CarlaUE4/{FrameDir}/{DateTimeNow}/{FrameName}* executing 'python start_replaying.py -f C:\Users\11\AppData\Local\CarlaUE4\Saved\test1.log'

wx983, 2023-02-12:
Hi~  Long time no see. How are you? I have a new issue.
Executing `CarlaUE4.exe -vr` from x64 cmd on windows and `python start_replaying.py -f /PATH/TO/RECORDING/FILE # windows`(in my computer is `python start_replaying.py -f C:\Users\11\AppData\Local\CarlaUE4\Saved\test1.log),`  I can't find the resulting frame capture images in Unreal/CarlaUE4/{FrameDir}/{DateTimeNow}/{FrameName}*. And I can only see images two days ago, which were generated in Editor mode, I remember. That's a little bit strange.
[image]

By the way, I verify **RunSyncReplay** and **RecordFrames** are enabled in DReyeVRConfig.ini.
[image]

wx983, 2023-02-12:
Sorry, my carelessness.
I have found the resulting images in %CARLA_ROOT%\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\CarlaUE4\FrameCap.
[image]

But I see the offical document points out it is in Unreal/CarlaUE4/{FrameDir}/{DateTimeNow}/{FrameName}*, [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#:~:text=Unreal/CarlaUE4/%7BFrameDir%7D/%7BDateTimeNow%7D/%7BFrameName%7D*), although I do see generated images. Images seem to be also produced in %CARLA_ROOT%\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\CarlaUE4\FrameCap.

wx983, 2023-02-12:
Great! I totally understand why it is so.

GustavoSilvera, 2023-02-12:
Yes, sorry the documentation was not clear enough! We'll update it soon. Glad your issue is resolved.

closed 2023-02-12 by GustavoSilvera