HARPLab/DReyeVR #81: How to acquire eye tracking data?

wx983, 2023-02-18:
Hi~ I have executed `start_recording.py` , `start_replaying.py` and `show_recorder_file_info.py -a -f C:\Users\11\AppData\Local\CarlaUE4\Saved\test1.log > recorder.txt `, which generate a video and a file named `recorder.txt`.
`recorder.txt` includes eye information of each frame like this:
[image]

I also see the eye tracking in the video during the replaying process, but I don't konw if there is any better method to acquire or store eye tracking data?

https://user-images.githubusercontent.com/73030394/219868156-4f94173e-b5e1-45e7-9088-7957abd905c0.mp4

GustavoSilvera, 2023-02-18:
The way you are currently extracting eye gaze data is via the `show_recorder_file_info` script which queries the recorder file (`test1.log`) and writes everything back to a file. This works great for getting per-tick eye gaze data but if you wanted something more real-time, you might be interested in our `DReyeVR_logging.py` script which demonstrates how you can query the DReyeVRSensor from the PythonAPI and print/save it or stream to a ROS instance if you'd like. 

```bash
python PythonAPI/examples/DReyeVR_logging.py
```

Otherwise I'm not sure what you mean by a "better method" to acquire or store the eye tracker data.

GustavoSilvera, 2023-02-18:
Another option is to keep the recorder file as you have it right now (`recorder.txt`) and then use our DReyeVR parser to parse through the file and create (pandas/numpy) dictionaries of each individual data measurement:

https://github.com/harplab/dreyevr-parser

wx983, 2023-02-20:
> 您当前提取眼睛注视数据的方式是通过`show_recorder_file_info`查询记录器文件 ( `test1.log`) 并将所有内容写回文件的脚本。这对于获取每个滴答的眼睛注视数据非常有用，但如果您想要更实时的东西，您可能会对我们的`DReyeVR_logging.py`脚本感兴趣，该脚本演示了如何从 PythonAPI 查询 DReyeVRSensor 并打印/保存它或流式传输到 ROS 实例如果你愿意的话。
> 
> ```shell
> python PythonAPI/examples/DReyeVR_logging.py
> ```
> 
> 否则我不确定你所说的“更好的方法”是什么意思来获取或存储眼动仪数据。

Well, I mean, I don't know what some data structure exactly mean, for instance, GazeDir and GazeOrigin. But now I understand them totally.
GazeDir means GazeDirection and I misunderstood it as GazeDirectory. So that confused me before.
[image]

As for a better way to store eye data, I'm exploring some visual methods to summarize eye data rules or show them. It's just an immature thinking. If I succeed, I will share them here. Thank you for your help!

closed 2023-03-05 by wx983

wx983, 2023-03-18:
@GustavoSilvera 
Hi! How do I know what Actor it refers to in the frame according to ActorName in FocusInfo of Eye tracking data?
[image]
[image]

wx983, 2023-03-18:
@GustavoSilvera 
And I also want to know what it means. Thank you!
[image]

GustavoSilvera, 2023-03-18:
The actor name corresponds to the name given for each of the entities in the world. The easiest way to find these is in the Editor by clicking on the actors and reading their AActor name, for instance I think the "Sculpture" you are seeing is the center fountain statue in Town03. We also track the hit position and direction so you can place some [debugging objects](https://docs.unrealengine.com/4.27/en-US/API/Runtime/Engine/DrawDebugSphere/) at these locations to better visualize which actor is being highlighted. 

The steering input comes from this [line](https://github.com/HARPLab/DReyeVR/blob/984d7560b7a01d34ed97559a0ebe7e7115c80a19/DReyeVR/EgoInputs.cpp#LL105C60-L105C73) which scales the raw steering input (-1..1) by 360 (degrees) as per the [SteeringScale](https://github.com/HARPLab/DReyeVR/blob/main/Config/DReyeVRConfig.ini#L66). So it is designed to measure the degrees of rotation (assuming limited to a single wheel rotation)

wx983, 2023-03-19:
> The actor name corresponds to the name given for each of the entities in the world. The easiest way to find these is in the Editor by clicking on the actors and reading their AActor name, for [...]

OK, I see the Actor name in Editor. 
So `Steering:-0.0021` refers to 0.0021*180  = 37.8 degrees left turn of the wheel?

GustavoSilvera, 2023-03-19:
The steering wheel is scaled relative to the maximum rotation of the hardware wheel, which could be >360 degrees. So unfortunately this scale doesn't really work if you rotate the steering wheel beyond 360, so in reality the scale is something like: percentage of turning (-1 to 1) / 360. 

Importantly, note that 0 is the "resting" state, and -1 is 100% rotated counter-clockwise (depends on how many rotations you can do on your hardware) and likewise +1 is 100% rotated clockwise

wx983, 2023-03-19:
@GustavoSilvera 
Hi! I also want to know what it refers to. I mean, something like 1001.7578 km/h?
[image]

GustavoSilvera, 2023-03-19:
The velocities in UE4 are measured in cm/s, which is what we use as well. Multiply by 0.036 to get cm/s -> km/h

FBanani, 2023-03-22:
Hi @GustavoSilvera I have another question, I want to find the other vehicles around the DReyeVR driving simulator. is there any way to record all vehicles positions.