HARPLab/DReyeVR #83: Do not have the access to Unreal Engine 4.26 (HARPLab fork)

chenruijia120, 2023-03-01:
Hi! I have joined the Epic Games Organization and am able to download UnrealEngine from Epic Games, but I do not have the access to Unreal Engine 4.26 (HARPLab fork) (https://github.com/HARPLab/UnrealEngine). What could have gone wrong? What should I do then? Looking forward to your response and thank you in advance!

GustavoSilvera, 2023-03-01:
Interesting! Can you still download the Carla fork? (https://github.com/CarlaUnreal/UnrealEngine)? If so, the HarpLab fork is essentially the same but fixed at commit d40ec35474e8793b4eea60dba6c4f051186e458e to preserve rendering quality (and performance) with DirectX11 on Windows. So if HARPLab fork is unavailable try doing:

```bash
git clone https://github.com/CarlaUnreal/UnrealEngine
cd UnrealEngine # ensure this is your $UE4_ROOT
git checkout d40ec35474e8793b4eea60dba6c4f051186e458e
# or git reset --hard d40ec35474e8793b4eea60dba6c4f051186e458e
```

We provide the HarpLab fork so you can skip the `git checkout` command and have DReyeVR work at optimal settings. So we will look into this issue! For now I hope this temporary solution resolves the issue for you!

chenruijia120, 2023-03-02:
It works. Thanks a lot!

chenruijia120, 2023-03-04:
Thank you for your quick response! I have met another problem when I tried to install the project. I was on the stage of "building everything", and when I tried to use "make launch" command, there was an error. I got this:

closed 2023-03-04 by chenruijia120

xuisshoe, 2023-11-10:
I am trying to checkout the "d40ec35" commit today but got "The remote server returned an error: (403) Forbidden. (WebException)". Is that commit no longer open to public?