HARPLab/DReyeVR #84: [CarlaUE4Editor] Error 6 when using "make launch" command

chenruijia120, 2023-03-04:
Hi Gustavo! I have met another problem when I tried to install the project. I was on the stage of "building everything", and when I tried to use "make launch" command, there was an error. I got this:
D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/DReyeVRSensor.cpp(13): fatal error C1083: can not open file: “carla/sensor/s11n/DReyeVRSerializer.h”: No such file or directory
Util/BuildTools/Windows.mk:20: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 6

And I cannot find DReyeVRSerializer.h file in carla folder, it seems only to appear in DReyeVR folder. Did I do anything wrong? Thank you in advance!

GustavoSilvera, 2023-03-04:
Hi @chenruijia120, when you ran `make install CARLA=...` there should have been a bunch of output for files moved around. Can you re-run this script and copy the output to verify the location of this file?

GustavoSilvera, 2023-03-04:
For instance, in our CI build we have that the `DReyeVRSerializer.h` file should be installed [here](https://github.com/HARPLab/DReyeVR/actions/runs/4306251448/jobs/7509808854#step:5:103) (`carla/LibCarla/source/carla/sensor/s11n/DReyeVRSerializer.[h|cpp]`)

GustavoSilvera, 2023-03-04:
You can also find any missing or modified files in your Carla repo by running `make check CARLA=...` (where `...` is the path to your Carla repo) which should look like [this](https://github.com/HARPLab/DReyeVR/actions/runs/4306251448/jobs/7509808854#step:6:157).

chenruijia120, 2023-03-06:
Thank you so much!!! I have solved this and met another problem... When I tried to "make package", an error occurred when cooking.
Here is the log:

> 
ProcessResult.StdOut:   LogWindows: Error: begin: stack for UAT
ProcessResult.StdOut:   LogWindows: Error: === Critical error: ===
ProcessResult.StdOut:   LogWindows: Error:
ProcessResult.StdOut:   LogWindows: Error: Fatal error!
ProcessResult.StdOut:   LogWindows: Error:
ProcessResult.StdOut:   LogWindows: Error: Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x0000000000000038
ProcessResult.StdOut:   LogWindows: Error:
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffd81df9885 UE4Editor-Carla.dll!ULoadAssetMaterialsCommandlet::ULoadAssetMaterialsCommandlet() [D:\crj\driving\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Commandlet\LoadAssetMaterialsCommandlet.cpp:37]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffdfd31bd63 UE4Editor-CoreUObject.dll!UClass::CreateDefaultObject() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\CoreUObject\Private\UObject\Class.cpp:3707]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffdfd5ddfae UE4Editor-CoreUObject.dll!UObjectLoadAllCompiledInDefaultProperties() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\CoreUObject\Private\UObject\UObjectBase.cpp:908]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffdfd5c13cf UE4Editor-CoreUObject.dll!ProcessNewlyLoadedUObjects() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\CoreUObject\Private\UObject\UObjectBase.cpp:997]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffdfd4f1625 UE4Editor-CoreUObject.dll!TBaseStaticDelegateInstance<void __cdecl(FName,bool),FDefaultDelegateUserPolicy>::ExecuteIfSafe() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:731]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffdf1ae9baa UE4Editor-Core.dll!TMulticastDelegate<void __cdecl(FName,bool),FDefaultDelegateUserPolicy>::Broadcast() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateSignatureImpl.inl:955]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffdf1b11897 UE4Editor-Core.dll!FModuleManager::LoadModuleWithFailureReason() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Core\Private\Modules\ModuleManager.cpp:519]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffe0e37e6e3 UE4Editor-Projects.dll!FModuleDescriptor::LoadModulesForPhase() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Projects\Private\ModuleDescriptor.cpp:561]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffe0e39321a UE4Editor-Projects.dll!FPluginManager::TryLoadModulesForPlugin() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Projects\Private\PluginManager.cpp:1295]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffe0e37e4e0 UE4Editor-Projects.dll!FPluginManager::LoadModulesForEnabledPlugins() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Projects\Private\PluginManager.cpp:1369]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ff720f052c7 UE4Editor-Cmd.exe!FEngineLoop::LoadStartupModules() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:3808]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ff720f0874d UE4Editor-Cmd.exe!FEngineLoop::PreInitPostStartupScreen() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:3198]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ff720f00e6d UE4Editor-Cmd.exe!GuardedMain() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Launch\Private\Launch.cpp:127]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ff720f011ca UE4Editor-Cmd.exe!GuardedMainWrapper() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:137]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ff720f14e5d UE4Editor-Cmd.exe!WinMain() [D:\crj\driving\UnrealEngine\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:268]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ff720f16f02 UE4Editor-Cmd.exe!__scrt_common_main_seh() [D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffe43027614 KERNEL32.DLL!UnknownFunction []
ProcessResult.StdOut:   LogWindows: Error: [Callstack] 0x00007ffe44de26a1 ntdll.dll!UnknownFunction []
ProcessResult.StdOut:   LogWindows: Error:
ProcessResult.StdOut:   LogWindows: Error: end: stack for UAT
CommandUtils.Run: Took 43.7468365s to run UE4Editor-Cmd.exe, ExitCode=3

Have you ever met this? Sorry for brothering you again. Thank you a lot!!

GustavoSilvera, 2023-03-06:
According to [this file](https://github.com/carla-simulator/carla/blob/f14acb257ebf44c302b225b02080ac5f0eedcf7f/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/LoadAssetMaterialsCommandlet.cpp#L35) it looks like this error might be because you might be missing this file: `"Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'"` (its a `.uasset` file). 

Can you build this in editor with `make launch`? Did you install all the content using `Update.bat`?

chenruijia120, 2023-03-07:
It is right. There is something wrong when I used Update.bat. Thank you very much!