HARPLab/DReyeVR #85: How to use autonomous driving algorithm rather than simple set_autopilot?

wx983, 2023-03-05:
Hi~ Long time no see. I have a new issue.
In DReyeVR_AI.py and start_recording.py, vehicles are all generated by `set_autopilot`. It seems a simple self-driving strategy that carla have implemented.
Are there other self-driving algorithms to use? If so, how should I do that to generate vehicles with autonomous algorithm?

GustavoSilvera, 2023-03-05:
While in "autopilot mode" (press 3 for the EgoVehicle) you can actually use any Carla-valid self-driving scheme. For instance, you can run `manual_control.py` and even control the EgoVehicle from the PythonAPI. This is how all Carla autonomous algorithms work (calling [`apply_control`](https://carla.readthedocs.io/en/latest/python_api/#carla.Vehicle) from the `carla.Vehicle`), you should be able to use any algorithm. The `DReyeVR_AI.py` script just instantiates a `TrafficManager` which is Carla's built-in algorithm for driving the vehicles. 

We don't provide any particular algorithms for DReyeVR since you should be able to use any Carla algorithms as if you are dealing with a normal Carla vehicle.

wx983, 2023-03-06:
Hum, thank you. But what I want to ask is exactly how to use self-driving algorithm in carla. It's a problem about carla rather than DReyeVR. Do you know that? Or maybe you could recommend some good open-resource project in github to me.
I would appreciate it very much!

GustavoSilvera, 2023-03-06:
Yes, this is more of a Carla situation than DReyeVR. Since we fully support the Carla API this enables you to apply any strategy to your autonomous vehicle as if it were a normal `carla.LibCarla.Vehicle`. So any Carla algorithm that uses the vanilla 0.9.13 version should work. 

There are many ways to enable Autonomous driving in Carla, such as creating your own AI agent from the [Leaderboard](https://leaderboard.carla.org/get_started_v1/#3-creating-your-own-autonomous-agent) or downloading an existing vehicle controller that you might find on [GitHub](https://github.com/topics/carla-simulator?o=asc&s=forks)

wx983, 2023-03-07:
@GustavoSilvera OKay, thank you. 
And I also want to know what kind of sensors are used in DReyeVR EgoVehicle?

GustavoSilvera, 2023-03-07:
The DReyeVR EgoVehicle by default only has one Carla sensor and it is a custom "DReyeVR sensor". It is implemented in [EgoSensor.h/cpp](https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/EgoSensor.h) and houses the logic to get the eye tracker data, user inputs, focus data, and vehicle data that we track as part of the overall [DReyeVRData](https://github.com/HARPLab/DReyeVR/blob/main/Carla/Sensor/DReyeVRData.h) abstraction. 

Since the purpose of DReyeVR is to introduce humans into the Carla world, there are no assistive sensors like cameras/vision that you might use on an autonomous vehicle. But you can always add them as you would for a normal Carla vehicle.

closed 2023-03-27 by GustavoSilvera