HARPLab/DReyeVR #86: How to get the user input from the hardware

chenruijia120, 2023-03-06:
Hi! I opened a new issue for a problem I met before. I tried using carla.VehicleControl.throttle and carla.VehicleControl.steer of the EgoVehicle to get user input on the physical steering wheel and throttle, but the output is always 0. Should I use another function to get the raw data from the hardware？Data that is mapped to 0-1 also works.

BTW that is why I reinstall the project and met lots of other issues lol. If I could solve this then maybe I could stop reinstall it. Thank you in advance and sorry for bothering you again!

GustavoSilvera, 2023-03-06:
That is a good point, we should forward our steering wheel data to the Carla API, we will work on this for the next version. 

Currently, the most straightforward way to get the (DReyeVR) user inputs in PythonAPI is to use our DReyeVR sensor API as follows:
```python
from DReyeVR_utils import DReyeVRSensor

 world = self.client.get_world()

 # find ego vehicle
 self.hero_actor = find_ego_vehicle(world)
 self.hero_transform = self.hero_actor.get_transform()

 # find ego sensor
 # DReyeVRSensor implicitly calls find_ego_sensor, then wraps it with a custom class
 self.sensor = DReyeVRSensor(world) 
 self.sensor.ego_sensor.listen(self.sensor.update)  # subscribe to readout

print(self.sensor) # this readout should contain some general string serialization of the sensor
print(self.sensor.data) # this should contain all the DReyeVR data fields (as a dictionary) including user inputs (steer, throttle, buttons, etc.) and eye gaze data.
```

GustavoSilvera, 2023-03-07:
Oh, this sounds similar to #12 and happens when you didn't rebuild the PythonAPI after installing DReyeVR (You might see warnings about mismatching client (PythonAPI) and server (UE4) versions in your Python script). You'll just need to rebuild the PythonAPI with DReyeVR installed by running `make PythonAPI` again. 

And no, you won't need eye tracking for this feature, if you disable it then you should still get user inputs.

chenruijia120, 2023-03-08:
Thank you for the answer. I met a new problem when I tried to run make PythonAPI. Here is the log:

D:\Apps\anaconda\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -ID:\Apps\anaconda\include -ID:\Apps\anaconda\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.9\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
dependencies/include\carla/road/element/RoadInfoMarkRecord.h(1): warning C4819: This file contains characters that cannot be represented in the current code page (936). Please save this file in Unicode format to prevent data loss
D:\crj\driving\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open included files: “OSM2ODR.h”: No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

It seems to be similar to [https://github.com/carla-simulator/carla/issues/4721](https://github.com/carla-simulator/carla/issues/4721). I tried to reinstall xerces-c with the solution in [https://github.com/carla-simulator/carla/issues/3320#issuecomment-735540901](https://github.com/carla-simulator/carla/issues/3320#issuecomment-735540901), but it did not work. Do you have any idea about that? Thank you in advance!!

GustavoSilvera, 2023-03-08:
Hi @chenruijia120, I believe this is similar to the [this sub-issue](https://github.com/HARPLab/DReyeVR/issues/88#issuecomment-1457278482) in #88 where I provide a solution [here](https://github.com/HARPLab/DReyeVR/issues/88#issuecomment-1457290512) which has worked for me, can you try this?

chenruijia120, 2023-03-10:
Thank you for your answer! But it did not work. I changed the XercesTranscoderSelection.cmake file with your instructions and rebuilt xerces, but the results did not change. I guess there are something wrong with proj. Do you have any idea about it? Here is the log:

`-- Setting build type to 'Release' as none was specified.
CMake Warning (dev) at CMakeLists.txt:23 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19044.
-- The C compiler identification is MSVC 19.29.30148.0
-- The CXX compiler identification is MSVC 19.29.30148.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR: D:/crj/driving/carla/Build/osm2odr-visualstudio
-- CMAKE_SOURCE_DIR: D:/crj/driving/carla/Build/om2odr-source
--
-- Platform:
--     Host: Windows-10.0.19044 AMD64
--     Target: Windows-10.0.19044 AMD64
--     CMake: 3.26.0-rc5
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30148.0
--
-- Found XercesC: D:\crj\driving\carla\Build\\xerces-c-3.2.3-install\lib\xerces-c.lib (found version "3.2.3")
CMake Error at CMakeLists.txt:75 (find_package):
  By not providing "FindProj.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Proj", but
  CMake did not find one.

  Could not find a package configuration file provided by "Proj" with any of
  the following names:

    ProjConfig.cmake
    proj-config.cmake

  Add the installation prefix of "Proj" to CMAKE_PREFIX_PATH or set
  "Proj_DIR" to a directory containing one of the above files.  If "Proj"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
用于 .NET Framework 的 Microsoft (R) 生成引擎版本 16.11.2+f32259642
版权所有(C) Microsoft Corporation。保留所有权利。

MSBUILD : error MSB1009: 项目文件不存在。
开关:install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\crj\driving\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
D:\Apps\anaconda\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -ID:\Apps\anaconda\include -ID:\Apps\anaconda\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.9\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true  [... cut ...]
dependencies/include\carla/road/element/RoadInfoMarkRecord.h(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的 字符。请将该文件保存为 Unicode 格式以防止数据丢失
D:\crj\driving\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: 无法打开包括文件: “OSM2ODR.h”: No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\crj\driving\carla\PythonAPI\carla\dist"!`

Some of the logs are presented in Chinese. Thank you in advance!

GustavoSilvera, 2023-03-10:
Hm. This usually works for me. Can you verify the proj dependency has been built? If not, try the `install_proj.bat` in installersWin

chenruijia120, 2023-03-11:
Sorry for being a little late and thanks for your answer! I verified all of the installation and a new kind of log occurred. It seems similar to [the issue you mentioned earlier](https://github.com/HARPLab/DReyeVR/issues/88#issuecomment-1457278482). But I have reinstall xerces as your instructions. What could be wrong then?

-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\crj\driving\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
D:\Apps\anaconda\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -ID:\Apps\anaconda\include -ID:\Apps\anaconda\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.9\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
dependencies/include\carla/road/element/RoadInfoMarkRecord.h(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以 防止数据丢失
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\link.exe" /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:D:\Apps\anaconda\libs /LIBPATH:D:\Apps\anaconda /LIBPATH:D:\Apps\anaconda\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.22000.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.22000.0\um\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.22000.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.22000.0\um\x64" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.9\Release\source/libcarla/libcarla.obj /OUT:build\lib.win-amd64-3.9\carla\libcarla.cp39-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.9\Release\source/libcarla\libcarla.cp39-win_amd64.lib shlwapi.lib Advapi32.lib ole32.lib shell32.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\carla_client.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\Detour.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\DetourCrowd.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\libboost_filesystem-vc142-mt-x64-1_72.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\libboost_python39-vc142-mt-x64-1_72.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\libpng.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\osm2odr.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\proj.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\Recast.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\rpc.lib D:\crj\driving\carla\PythonAPI\carla\dependencies/lib\sqlite3.lib D:\crj\driving\carla\PythonAPI\carl [... cut ...]
xerces-c_3.lib(XMLUniCharacter.obj) : error LNK2001: 无法解析的外部符号 u_charType_58
build\lib.win-amd64-3.9\carla\libcarla.cp39-win_amd64.pyd : fatal error LNK1120: 24 个无法解析的外部命令
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit code 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\crj\driving\carla\PythonAPI\carla\dist"!

Thank you in advance and sorry for bothering you again!

GustavoSilvera, 2023-03-11:
Nope that you've got `proj` you should verify that the xerces transcoder is the windows one. (Make sure to clear xerces build cache by deleting the xerces-source/build and xerces-install folders, then rebuild).

Those ICU linker errors look like transcoder errors which should be resolved with the windows selection.

chenruijia120, 2023-03-12:
Thank you for your answer! But I have rebuilt xerces and changed the transcoder file again. That is kind of wierd... What else may I do?

Another strange thing is that I can only install xercesc successfully with "install_xercesc.bat". But if I use the make PythonAPI command, the installation fails. But I guess these two are essentially one command?

GustavoSilvera, 2023-03-12:
Hmm I'm pretty confident that the linker errors with `u_strlen_58` come from the transcoder compilation which needs to be disabled as described [here](https://github.com/carla-simulator/carla/issues/5632#issuecomment-1304714179). Can you list out the steps you take exactly to do the following:
1. delete  `D:\crj\driving\carla\Build\xerces-c-3.2.3-install` entirely
2. delete  `D:\crj\driving\carla\Build\xerces-c-3.2.3-source/build` (and any other cache)
3. disable the `XERCES_USE_TRANSCODER_ICU` macro and enable the `XERCES_USE_TRANSCODER_WINDOWS` macro in [`D:\crj\driving\carla\Build\xerces-c-3.2.3-source/cmake/XercesTranscoderSelection.cmake`](https://github.com/apache/xerces-c/blob/4dc7c00d192dcf8f89e1f9f9e3bc5f9a2db52535/cmake/XercesTranscoderSelection.cmake#L97-L101) (you should also disable the if-else condition to make sure the ICU transcoder macro isn't re-enabled)
4. rebuild with `make PythonAPI` and post the logs here.

chenruijia120, 2023-03-13:
Hi Gustavo! Sorry for bothering for a long time. I followed your instructions and changed the XercesTranscoderSelection file as follows:

set(XERCES_USE_TRANSCODER_ICU 0)
set(XERCES_USE_TRANSCODER_ICONV 0)
set(XERCES_USE_TRANSCODER_GNUICONV 0)
set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 0)
set(XERCES_USE_TRANSCODER_WINDOWS 1)
if(transcoder STREQUAL "icu")
  set(XERCES_USE_TRANSCODER_ICU 0)
elseif(transcoder STREQUAL "iconv")
  set(XERCES_USE_TRANSCODER_ICONV 1)
elseif(transcoder STREQUAL "gnuiconv")
  set(XERCES_USE_TRANSCODER_GNUICONV 1)
elseif(transcoder STREQUAL "macosunicodeconverter")
  set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 1)
elseif(transcoder STREQUAL "windows")
  set(XERCES_USE_TRANSCODER_WINDOWS 1)
else()
  message(FATAL_ERROR "Invalid transcoder: \"${transcoder}\"")
endif()

Xerces appears to be installed successfully this time, with the bin/cmake/include/lib/share folder appearing, but there are errors in the logs during the installation of Xerces, and the final make PythonAPI does not succeed. I've put the full log (the first small part is not saved due to limited command prompt display) in [this Google Drive file](https://drive.google.com/file/d/1TfN5T6e-lrJ4J1-jvOaUuTolxLF3X4rw/view?usp=sharing). Thank you for answering and it is really kind of you!!

GustavoSilvera, 2023-03-13:
No I don't think xerces is installed correctly (although Carla says it is). Can you delete the `if-else` chain in the cmake file and clear the build cache and try again. You shouldn't be getting any `ICUTransService` related linker errors since this should force the macro `XERCES_USE_TRANSCODER_ICU` to false. Since you still have errors related to this macro I believe this transcoder macro is not being set correctly.

I might be forgetting which build folder is the cache, so a safer bet is to delete all xerces folders (source and install), then change the cmake file in the source folder then rebuild.

chenruijia120, 2023-03-15:
Thank you for your answer! It is weird that it still did not work. I tried another way to get the data from the hardware. Thank you for all the patience!

GustavoSilvera, 2023-03-18:
Hi @chenruijia120 , after looking into this more, a more robust solution is as follows:
1. delete your entire xerces build folder
2. reset all your changes to `Util/InstallersWin/install_xercesc.bat` (same as [vanilla carla](https://github.com/carla-simulator/carla/blob/0.9.13/Util/InstallersWin/install_xercesc.bat)) (Make sure Xercesc version is 3.2.3!)
3. Change the url for `XERCESC_REPO` to `https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-%XERCESC_VERSION%.zip` (uses the archive link)
4. Add the flag `-Dtranscoder=windows^` to the `cmake` command on line ~120 so it looks like:
```
cmake .. -G "Visual Studio 16 2019" -A x64^
  -DCMAKE_INSTALL_PREFIX="%XERCESC_INSTALL_DIR:\=/%"^
  -DBUILD_SHARED_LIBS=OFF^
  -Dtranscoder=windows^
  "%BUILD_DIR%%XERCESC_BASENAME%-%XERCESC_VERSION%-source"
```
5. run `make PythonAPI` again (might have to run twice)

We'll be updating the DReyeVR repo with this fix soon.

closed 2023-05-10 by GustavoSilvera