HARPLab/DReyeVR #88: Error while executing make PythonAPI

Dreyevr, 2023-03-06:
When executing make PythonAPI I get the following error:

C:\Users\Administrator\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  24
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Users\Administrator\carla\Build\"
-[Setup]: Creating "C:\Users\Administrator\carla\Build\" folder...
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\Administrator\carla\Build\"
    -[install_zlib]: Retrieving zlib.
    -[install_zlib]: Extracting zlib from "zlib-1.2.13.zip".
    -[install_zlib]: Removing "zlib-1.2.13.zip"
    -[install_zlib]: Creating "C:\Users\Administrator\carla\Build\zlib-source\build"
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.29.30148.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - works
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
CMake Error at C:/CMake/share/cmake-3.25/Modules/CheckTypeSize.cmake:148 (try_compile):
  Cannot copy output executable

    ''

  to destination specified by COPY_FILE:

    'C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CheckTypeSize/OFF64_T.bin'

  Recorded try_compile output location doesn't exist:

    C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeScratch/TryCompile-x7t6g5/cmTC_6b0b2.exe

Call Stack (most recent call first):
  C:/CMake/share/cmake-3.25/Modules/CheckTypeSize.cmake:278 (__check_type_size_impl)
  CMakeLists.txt:40 (check_type_size)

-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Renaming
--     C:/Users/Administrator/carla/Build/zlib-source/zconf.h
-- to 'zconf.h.included' because this file is included with zlib
-- but CMake generates it automatically in the build directory.
-- Configuring incomplete, errors occurred!
See also "C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeError.log".

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

I tried reinstalling CMake, as well as trying different versions of it. I am using Carla 0.9.13 as required by DreyeVR.

Dreyevr, 2023-03-06:
Yes, I've tried it with CMake 3.26, 3.25.2 and 3.24.3 with the same results.

GustavoSilvera, 2023-03-06:
Can you clean your build directory by deleting the `zlib-source` (and `zlib-install`) folder?

Dreyevr, 2023-03-06:
You mean the Build folder? I always delete the whole build directory in carla before executing make PythonAPI.

Also, i changed the install_zlib.bat file to download the 1.2.13 version of zlib instead of 1.2.11 because it couldn't find the file on the server.(In case it's relevant)

Dreyevr, 2023-03-06:
I deleted the zlib folder. What should I do next?

Dreyevr, 2023-03-06:
It gives me the same output as the one i posted in my issue:

C:\Users\Administrator\carla>make setup
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  24
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Users\Administrator\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\Administrator\carla\Build\"
    -[install_zlib]: Retrieving zlib.
    -[install_zlib]: Extracting zlib from "zlib-1.2.13.zip".
    -[install_zlib]: Removing "zlib-1.2.13.zip"
    -[install_zlib]: Creating "C:\Users\Administrator\carla\Build\zlib-source\build"
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.29.30148.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - works
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
CMake Error at C:/CMake/share/cmake-3.26/Modules/CheckTypeSize.cmake:148 (try_compile):
  Cannot copy output executable

    ''

  to destination specified by COPY_FILE:

    'C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CheckTypeSize/OFF64_T.bin'

  because:

    Invalid argument (input)

  Recorded try_compile output location doesn't exist:

    C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeScratch/TryCompile-fhd679/cmTC_1149e.exe

Call Stack (most recent call first):
  C:/CMake/share/cmake-3.26/Modules/CheckTypeSize.cmake:273 (__check_type_size_impl)
  CMakeLists.txt:40 (check_type_size)

-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Renaming
--     C:/Users/Administrator/carla/Build/zlib-source/zconf.h
-- to 'zconf.h.included' because this file is included with zlib
-- but CMake generates it automatically in the build directory.
-- Configuring incomplete, errors occurred!

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

GustavoSilvera, 2023-03-06:
I just double checked, we in fact are using CMake 3.20. So you might want to try this version.

Dreyevr, 2023-03-06:
Now i get a slightly different(but mostly the same) output:

`C:\Users\Administrator\carla>make setup
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  24
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Users\Administrator\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\Administrator\carla\Build\"
    -[install_zlib]: Retrieving zlib.
    -[install_zlib]: Extracting zlib from "zlib-1.2.13.zip".
    -[install_zlib]: Removing "zlib-1.2.13.zip"
    -[install_zlib]: Creating "C:\Users\Administrator\carla\Build\zlib-source\build"
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.29.30148.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - works
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
CMake Error at C:/CMake/share/cmake-3.20/Modules/CheckTypeSize.cmake:139 (try_compile):
  Cannot copy output executable

    ''

  to destination specified by COPY_FILE:

    'C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CheckTypeSize/OFF64_T.bin'

  Unable to find the executable at any of:

    C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeTmp/cmTC_867aa.exe
    C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeTmp/Debug/cmTC_867aa.exe
    C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeTmp/Development/cmTC_867aa.exe

Call Stack (most recent call first):
  C:/CMake/share/cmake-3.20/Modules/CheckTypeSize.cmake:270 (__check_type_size_impl)
  CMakeLists.txt:40 (check_type_size)

-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Renaming
--     C:/Users/Administrator/carla/Build/zlib-source/zconf.h
-- to 'zconf.h.included' because this file is included with zlib
-- but CMake generates it automatically in the build directory.
-- Configuring incomplete, errors occurred!
See also "C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Administrator/carla/Build/zlib-source/build/CMakeFiles/CMakeError.log".

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1`

GustavoSilvera, 2023-03-06:
okay weird, can you try following the workaround suggested [here](https://gitlab.kitware.com/cmake/cmake/-/issues/23115)

Dreyevr, 2023-03-06:
Hi, I tried follow the link you sent, but I am confused as to what I am supposed to do.

Am I supposed to comment 'set(CMAKE_EXECUTABLE_SUFFIX ".x")' ? And if so, in which file exactly? 

I am not very knowledgeable with CMake.

GustavoSilvera, 2023-03-06:
Hmm okay yeah I realize this is for their specific use case. So not super applicable here. 

Can you try using our prebuilt zlib? 
- Delete all the `zlib-*` folders in your `Build`. Then download this [pre-built Carla 0.9.13 Build](https://drive.google.com/file/d/1tOcxvBGx9CWRMbUwpQ96X3qFno3bNw_s/view?usp=share_link) folder that we've built (and extract with 7zip into somewhere else) and copy the `zlib-source` and `zlib-install` from there folders into your `Build` folder, then try re-running `make setup`.

Dreyevr, 2023-03-06:
Hi, so for some unknown reason the error disappeared. Instead now I get the following error:

common.copy C:\Users\Administrator\carla\Build\boost-1.72.0-install\lib\cmake\boost_python-1.72.0\boost_python-config-version.cmake
build\boost\bin.v2\libs\python\build\install\boost_python-config-version.cmake
        1 file(s) copied.
...failed updating 5 targets...
...skipped 4 targets...
...updated 15297 targets...

    -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

Could this have something to do with the python version? I am using the latest one 3.11.2.

GustavoSilvera, 2023-03-06:
Can you instead use Python 3.7.x, we've noticed similar problems on later versions. We have instructions on our Install page on how to set up an Anaconda environment for this as well which should make the overall PythonAPI experience better (and is our recommended strategy).

Dreyevr, 2023-03-07:
I applied your advice and used the conda environment. It fixed the error I was getting, but further down the line I got a new error:

CMake Error at C:/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/CMake/share/cmake-3.20/Modules/FindXercesC.cmake:110 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Administrator/carla/Build/osm2odr-visualstudio/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\Users\Administrator\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
creating source\carla.egg-info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
C:\Users\Administrator\.conda\envs\carla13\lib\site-packages\setuptools\command\install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
running install_lib
running build_py
creating build
creating build\lib.win-amd64-cpython-37
creating build\lib.win-amd64-cpython-37\carla
copying source\carla\command.py -> build\lib.win-amd64-cpython-37\carla
copying source\carla\__init__.py -> build\lib.win-amd64-cpython-37\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-cpython-37
creating build\temp.win-amd64-cpython-37\Release
creating build\temp.win-amd64-cpython-37\Release\source
creating build\temp.win-amd64-cpython-37\Release\source\libcarla
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\Administrator\.conda\envs\carla13\include -IC:\Users\Administrator\.conda\envs\carla13\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-cpython-37\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
C:\Users\Administrator\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: 'OSM2ODR.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\Administrator\carla\PythonAPI\carla\dist"!

(carla13) C:\Users\Administrator\carla>

Not sure where the problem lies for this one.

GustavoSilvera, 2023-03-07:
Okay now this is an error I've seen before, can you do the following:
- Delete your `xerces-source/build`and `xerces-install` folder in your build directory
- In `xerces-source/cmake/XercesTranscoderSelection.cmake` and replace the bottom of the file (after line 100) like this:
```
(This is the original, which you should update)

# line ~100 of XercesTranscoderSelection.cmake
set(XERCES_USE_TRANSCODER_ICU 0)
set(XERCES_USE_TRANSCODER_ICONV 0)
set(XERCES_USE_TRANSCODER_GNUICONV 0)
set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 0)
set(XERCES_USE_TRANSCODER_WINDOWS 0)
if(transcoder STREQUAL "icu")
  set(XERCES_USE_TRANSCODER_ICU 1)
elseif(transcoder STREQUAL "iconv")
  set(XERCES_USE_TRANSCODER_ICONV 1)
elseif(transcoder STREQUAL "gnuiconv")
  set(XERCES_USE_TRANSCODER_GNUICONV 1)
elseif(transcoder STREQUAL "macosunicodeconverter")
  set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 1)
elseif(transcoder STREQUAL "windows")
  set(XERCES_USE_TRANSCODER_WINDOWS 1)
else()
  message(FATAL_ERROR "Invalid transcoder: \"${transcoder}\"")
endif()
```

With this (Notice it just forces the use of the Windows transcoder):
```
set(XERCES_USE_TRANSCODER_ICU 0)
set(XERCES_USE_TRANSCODER_ICONV 0)
set(XERCES_USE_TRANSCODER_GNUICONV 0)
set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 0)
set(XERCES_USE_TRANSCODER_WINDOWS 1)
```
 

I've described this workaround [here](https://github.com/carla-simulator/carla/issues/5632#issuecomment-1304714179) as well

GustavoSilvera, 2023-03-07:
Also, in preparation for using Conda for your Python, you'll need to follow [the steps here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#building-dreyevr-pythonapi) to call `python setup.py` rather than the (built in Windows) `py -3 setup.py`.  

(specifically the fix for `BuildPythonAPI.bat`)

Dreyevr, 2023-03-09:
Hi, I implemented your solution, but I'm still getting the error. Like you said, I removed 

set(XERCES_USE_TRANSCODER_ICU 0)
set(XERCES_USE_TRANSCODER_ICONV 0)
set(XERCES_USE_TRANSCODER_GNUICONV 0)
set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 0)
set(XERCES_USE_TRANSCODER_WINDOWS 0)
if(transcoder STREQUAL "icu")
  set(XERCES_USE_TRANSCODER_ICU 1)
elseif(transcoder STREQUAL "iconv")
  set(XERCES_USE_TRANSCODER_ICONV 1)
elseif(transcoder STREQUAL "gnuiconv")
  set(XERCES_USE_TRANSCODER_GNUICONV 1)
elseif(transcoder STREQUAL "macosunicodeconverter")
  set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 1)
elseif(transcoder STREQUAL "windows")
  set(XERCES_USE_TRANSCODER_WINDOWS 1)
else()
  message(FATAL_ERROR "Invalid transcoder: \"${transcoder}\"")
endif()

with 

set(XERCES_USE_TRANSCODER_ICU 0)
set(XERCES_USE_TRANSCODER_ICONV 0)
set(XERCES_USE_TRANSCODER_GNUICONV 0)
set(XERCES_USE_TRANSCODER_MACOSUNICODECONVERTER 0)
set(XERCES_USE_TRANSCODER_WINDOWS 1).

and this is the error I get:
CMake Error at C:/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/CMake/share/cmake-3.20/Modules/FindXercesC.cmake:110 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

I did however have to change the xercesc version to 3.2.4

Dreyevr, 2023-03-09:
Here is the configuration summary:

-- Xerces-C++ configuration summary
-- --------------------------------
--
--   Version:                   3.2.4
--   Library interface version: 3.2
--
--   Installation directory:    C:/Users/Administrator/carla/Build/xerces-c-3.2.4-install
--   C compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
--   C++ compiler:              C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
--
--   Build shared libraries:    OFF
--   Path delimiters:           "/\"
--   File Manager:              windows
--   Mutex Manager:             standard
--   Transcoder:                windows
--   NetAccessor:               winsock
--   Message Loader:            inmemory
--   XMLCh type:                char16_t
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Administrator/carla/Build/xerces-c-3.2.4-source/build
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

The macro seems to have been applied, but the error still persists.

GustavoSilvera, 2023-03-09:
Did you delete the build cache of xerces? (Remove Build/xerces-c-...-source/build)

Also why did you need to change to 3.2.4? I think 3.2.3 should still work fine with the current version of Carla.

Dreyevr, 2023-03-09:
Yes, I did delete that folder.

I changed it because the url for 3.2.3 wasn't working. It was the same with zlib

Dreyevr, 2023-03-09:
this was also part of the output, don't know if this is relevant;

-- Looking for C++ include arpa/inet.h
-- Looking for C++ include arpa/inet.h - not found
-- Looking for C++ include dlfcn.h
-- Looking for C++ include dlfcn.h - not found
-- Looking for C++ include fcntl.h
-- Looking for C++ include fcntl.h - found
-- Looking for C++ include float.h
-- Looking for C++ include float.h - found
-- Looking for C++ include langinfo.h
-- Looking for C++ include langinfo.h - not found
-- Looking for C++ include limits.h
-- Looking for C++ include limits.h - found
-- Looking for C++ include locale.h
-- Looking for C++ include locale.h - found
-- Looking for C++ include memory.h
-- Looking for C++ include memory.h - found
-- Looking for C++ include netdb.h
-- Looking for C++ include netdb.h - not found
-- Looking for C++ include netinet/in.h
-- Looking for C++ include netinet/in.h - not found
-- Looking for C++ include nl_types.h
-- Looking for C++ include nl_types.h - not found
-- Looking for C++ include stdbool.h
-- Looking for C++ include stdbool.h - found
-- Looking for C++ include stddef.h
-- Looking for C++ include stddef.h - found
-- Looking for C++ include stdlib.h
-- Looking for C++ include stdlib.h - found
-- Looking for C++ include string.h
-- Looking for C++ include string.h - found
-- Looking for C++ include strings.h
-- Looking for C++ include strings.h - not found
-- Looking for C++ include sys/param.h
-- Looking for C++ include sys/param.h - not found
-- Looking for C++ include sys/socket.h
-- Looking for C++ include sys/socket.h - not found
-- Looking for C++ include sys/stat.h
-- Looking for C++ include sys/stat.h - found
-- Looking for C++ include sys/time.h
-- Looking for C++ include sys/time.h - not found
-- Looking for C++ include sys/timeb.h
-- Looking for C++ include sys/timeb.h - found
-- Looking for C++ include sys/types.h
-- Looking for C++ include sys/types.h - found
-- Looking for C++ include unistd.h
-- Looking for C++ include unistd.h - not found
-- Looking for C++ include wchar.h
-- Looking for C++ include wchar.h - found
-- Looking for C++ include wctype.h
-- Looking for C++ include wctype.h - found
-- Looking for C++ include endian.h
-- Looking for C++ include endian.h - not found
-- Looking for C++ include machine/endian.h
-- Looking for C++ include machine/endian.h - not found
-- Looking for C++ include arpa/nameser_compat.h
-- Looking for C++ include arpa/nameser_compat.h - not found

Dreyevr, 2023-03-10:
[output.txt](https://github.com/HARPLab/DReyeVR/files/10941696/output.txt)

This is the entire output from the cmd after deleting the whole build folder and doing make PythonAPI from scratch.

I'll see if I can download the 3.2.3 version of Xercesc again, but if you see that the problem lies somewhere else, I'd really appreciate the help.

GustavoSilvera, 2023-03-10:
Looking at your `output2.txt` I don't see any errors? It looks like everything built successfully. You can verify that a .egg and a .whl were built by looking for the files in `PythonAPI/carla/dist/`. 

Then try running the sanity check:
```bash
python -c "import carla; print(dir(carla))"
```
Should give output like:
```bash
['Actor', 'ActorAttribute', 'ActorAttributeType', 'ActorBlueprint', 'ActorList', 'ActorSnapshot', 'ActorState', 'AttachmentType', 'BlueprintLibrary', 'BoundingBox', 'CityObjectLabel', 'Client', 'ClientSideSensor', 'CollisionEvent', 'Color', 'ColorConverter', 'DReyeVREvent', 'DVSEvent', 'DVSEventArray', 'DebugHelper', 'EnvironmentObject', 'FakeImage', 'FloatColor', 'GearPhysicsControl', 'GeoLocation', 'GnssMeasurement', 'IMUMeasurement', 'Image', 'Junction', 'LabelledPoint', 'Landmark', 'LandmarkOrientation', 'LandmarkType', 'LaneChange', 'LaneInvasionEvent', 'LaneInvasionSensor', 'LaneMarking', 'LaneMarkingColor', 'LaneMarkingType', 'LaneType', 'LidarDetection', 'LidarMeasurement', 'Light', 'LightGroup', 'LightManager', 'LightState', 'Location', 'Map', 'MapLayer', 'MaterialParameter', 'ObstacleDetectionEvent', 'OpendriveGenerationParameters', 'OpticalFlowImage', 'OpticalFlowPixel', 'Osm2Odr', 'Osm2OdrSettings', 'RadarDetection', 'RadarMeasurement', 'Rotation', 'SemanticLidarDetection', 'SemanticLidarMeasurement', 'Sensor', 'SensorData', 'ServerSideSensor', 'TextureColor', 'TextureFloatColor', 'Timestamp', 'TrafficLight', 'TrafficLightState', 'TrafficManager', 'TrafficSign', 'Transform', 'Vector2D', 'Vector3D', 'Vehicle', 'VehicleControl', 'VehicleDoor', 'VehicleLightState', 'VehiclePhysicsControl', 'VehicleWheelLocation', 'Walker', 'WalkerAIController', 'WalkerBoneControlIn', 'WalkerBoneControlOut', 'WalkerControl', 'Waypoint', 'WeatherParameters', 'WheelPhysicsControl', 'World', 'WorldSettings', 'WorldSnapshot', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'bone_transform', 'bone_transform_out', 'command', 'libcarla', 'vector_of_bones', 'vector_of_bones_out', 'vector_of_gears', 'vector_of_ints', 'vector_of_transform', 'vector_of_vector2D', 'vector_of_wheels']
```

To build the next steps (editor and UE4 package) run `make launch` and `make package`

Dreyevr, 2023-03-10:
The issue was resolved by allowing powershell scripts to run

closed 2023-03-10 by Dreyevr

GustavoSilvera, 2023-03-22:
Hi @gfresdoc, we recently pushed an update to main (#92) that should significantly improve the build situation on Windows. Can you try cleaning everything (can delete DReyeVR entirely and re-clone) pull from `main` and run a `make install CARLA=..\PATH\TO\CARLA` and one single `make PythonAPI`. 

There were several bugs in Carla's original Windows build pipeline involving xerces, zlib, and osm2odr which we believe are all addressed. You should only need to run a single `make PythonAPI` from the Carla 0.9.13 directory (once DReyeVR is installed)

GustavoSilvera, 2023-03-22:
(And by "clean everything" I also mean delete whatever `Build/` directory you have in your `carla` repository)