HARPLab/DReyeVR #89: Installing DReyeVR Core

ande269f, 2023-03-10:
hello @GustavoSilvera 
i'm having major issue installing DReyeVR Core at this step:

[image]

my "make install" is not found: 

[image]

make has worked uptill now, and i have been trying to research it, and found that the cmake --install might be the same, though it does not work as well: 

[image]

but it exits with another issue: 

[image]

i hope you can help me, im desperate

ande269f, 2023-03-10:
okay, i found the answer in "frequently asked questions: 

[image]

new problem arised: 

[image]

translation: "python blev ikke fundet" -> "python was not found" and then something about deactivating it to download it from windows store

GustavoSilvera, 2023-03-10:
First, you should only need to run `make install CARLA=...` from within the DReyeVR directory, and install to your carla directory. 

Then you'll want to be in the `Carla` directory for the remainder of the build process. In a `conda` activated environment using Python 3.7 on Windows, run `make PythonAPI`. You might run into errors similar to #88 but we have notes for fixing issues in case they arise in your build process. 

Once `make PythonAPI` is completed, you should have a `.whl` and `.egg` file in your `PythonAPI/carla/dist/` directory, which then allows you to use the PythonAPI with Carla. 

The PythonAPI also builds `LibCarla` which is required to build the server (Unreal Engine editor) as part of Carla's build process. 

You should then run `make launch` and `make package` in the Carla directory to see the visual engine.

ande269f, 2023-03-13:
hello again

thank you for your comment, i see now that i messed up the order of the steps.

i have successfully set up a conda enviorenment inside VS2019, but when i'm running "make PythonAPI", it ends with the same issue as issue #2339 in the carla issue line on Github. 

[image]

Luckily, there was linked to the issue on the build guide, but i have followed the instructions on to swap "py -3" to "python":

[image]

but the build fails the same way. I make sure to use Python 3.7, make 3.81, Cmake 3.25.2

[image]

i found an error codes that might be relevant, or might not: 

[image]

note: removed my other python enviorenment paths, as comment https://github.com/carla-simulator/carla/issues/2881#issuecomment-657084327 points, which looks like this now:
[image]
but its found when im using the carla13 conda enviorenment: 
[image]

there's some other related issue fixes posted: https://github.com/carla-simulator/carla/pull/2348, but i feel unsure is wheather there has been updates since then.
im not sure what im doing wrong. Can you help?
I really apreciate your patience!

ande269f, 2023-03-13:
and as again, i find the solution right after posting a comment haha. 

i messed up the carla.tbh file, i fixed it and now it works

GustavoSilvera, 2023-03-13:
Awesome! Looks like everything is good!

No, those warnings can be safely ignored (they are from Carla's backend using PhysX which is deprecated in newer versions of UE4). 

Sounds like you are good to go!

ande269f, 2023-03-14:
Awesome! :D thanks for eveything

there were some issues when downloading it on windows, but its hard to say if whether they came from my own actions or from the software. 

What helped a lot was:
(1) to use 7Zip to download files that might have been missing for some reason (in my case mainly xerces-source in the carla\Build directory)
(2) make sure make (called minGW32-make in my case on windows) and Cmake is the right version, and described in the guide on installing DReyeVR 
(3) and the path envireonment variables are set and double checked in the used console to see if they are available for the commands used (i just put a path on every single used program into both of the path variable envirenment options on windows).
(4) double-check that you dont just copy paste the /path/to/carla/... exambles into the prescribed places, such as the carla.pth file, but actually set your own instance of the path to it. This happened alot for me when my brain got tired
(5) follow the order of things in the guide. When issues come up, they cannot be ignored, and needs to be fixes before you continue.

some maybe helpful notes: 
* make on windows is the same as minGW32-make, there are youtube videoes on how to download it
* most of the problems that occour have an explaination on the net if you search 
* sometimes, it helps to simply just restart the computer and try again
* its very possible to set up the program whitout knowing anything about any of the added programs (conda, 7Lib, cmake, vs2019), but in my case it took about 25 hours to debug issues. It helps to understand what these programs are doing. 
* issues in VS2019 are sometimes not visable to begin with, and you need to check the console output to spot if any ERRORS are showing up.

GustavoSilvera, 2023-03-14:
These are great notes for the initial installation process! Sorry it was such a headache but I'm glad it finally works for you. 

Happy to help debug the VR issues if they come up!

ande269f, 2023-03-15:
Its worth the work though! 

i have some issues with start-up now, but i think it deserves a new issue
hope to see you there:)

closed 2023-03-27 by GustavoSilvera