HARPLab/DReyeVR #91: EXCEPTION_ACCESS_VIOLATION reading address 0xffffffffffffffff

jemishGhoghari, 2023-03-16:
Hello,

After successfully installing Library, I am facing some problems while launching Carla from Unreal Engine. You can see the below Picture! Can you give me a possible solution? 
The error looks like 
[image]

GustavoSilvera, 2023-03-16:
Hi, can you forward the output log as a text file. 

Did you change anything in the source code to enable the DReyeVRCustomActors in any way? There seems to be a segfault in the destructor during garbage collection, but this is not enabled by default when you first install DReyeVR.

jemishGhoghari, 2023-03-16:
Hi,

I installed the DReyeVR again to revert my changes! But, It is still throwing this Error!

Here is the log file,

[CarlaUE4.log](https://github.com/HARPLab/DReyeVR/files/10996040/CarlaUE4.log)

One more thing,
When I installed the Repo for Carla, UE Editor was crashing a lot! This was probably because of the Direct X version! My PC has DX12, but UE4 uses DX11, So in the DefaultEngine.ini config, I changed DefaultGraphicsRHI=DefaultGraphicsRHI_DX12. Could this create an error?

GustavoSilvera, 2023-03-16:
I've never seen this CustomActor destructor error before, you should be able to work around it by deleting the custom destructor (cpp impl) and setting the destructor (in the header file) to `default` (like this: `~ADReyeVRCustomActor() = default;)

But I've also never seen this build issue before, are you building/cooking the game in the editor after it is installed? 

Also, I'd recommend getting DX11 working (You seem to have a compatible GPU) for best performance and stability, DX12 is less tested and slower for VR. However, if you must use DX12 then I'd make sure you are on the UnrealEngine Carla fork commit (`d40ec35474e8793b4eea60dba6c4f051186e458e`) mentioned [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#getting-started). 

Also can you run `make check` (From the DReyeVR repo) to see if any files are missing/modified in the Carla repo?

jemishGhoghari, 2023-03-19:
Hello @GustavoSilvera 

I did the same as you recommended, and Carla is launching successfully from the UE editor!

I changed the destructor to default by deleting the custom destructor!

Also, I have managed to work with DX11! So, I do not need to change DefaultGraphicsRHI!
But, There are some warnings! Please look at the Log file (The last part of the text file).

[CarlaUE4.log](https://github.com/HARPLab/DReyeVR/files/11011083/CarlaUE4.log)

Thank you very much for your Support @GustavoSilvera

GustavoSilvera, 2023-03-19:
All of these warnings (and the `LogStreaming` errors too) are benign and do not need to be worried about :) Looks like you're good to go!

closed 2023-03-27 by GustavoSilvera